Smoothieware/Smoothieware #1115: delta-grid.size is incorrect

pflugj, 2017-02-09:
### What should I do before submitting an issue?
#### What were you doing?
DELTA GRID leveling on a cartesian
1. leveling-strategy.delta-grid.size    3
2. leveling-strategy.delta-grid.size    5
3. leveling-strategy.delta-grid.size    7

#### What did you expect to happen?
1. Grid 3x3
2. Grid 5x5
3. Grid 7x7

#### What happened instead?
1. grid.size  3   probed only the center
2. grid.size  5   probed a 3x3 grid
3. grid.size  7   probed a 5x5 grid

#### Branch & Commit or Version of Smoothieware
Edge from 7 Feb 2017

#### Hardware model you are using (Make, and version)
Smoothieboard

#### Host and Operating System running the Host
Octoprint

#### Link to and console logs
I do not have a log is it really needed / how to do?
   
#### Link to running config and M503 results
##################  GRID Leveling using Delta or cartesian
leveling-strategy.delta-grid.enable               true     # The strategy must be enabled in the config, as well as the zprobe module.
leveling-strategy.delta-grid.is_square           true     # Set this to true if using on a cartesian machine

leveling-strategy.delta-grid.do_home            true
mm_per_line_segment                               1        # necessary for cartesians using delta-grid

# If the bed is rectangular, use the following options to set the maximum absolute distance from 0,0 in X and Y.
leveling-strategy.delta-grid.max_x                150      # Maximum distance in the X axis from center
leveling-strategy.delta-grid.max_y                150      # Maximum distance in the Y axis from center
leveling-strategy.delta-grid.size                     3        # The size of the grid, for example, 7 causes a 7x7 grid with 49 points. 
                                                           # Must be an odd number.
leveling-strategy.delta-grid.probe_offsets        0,28,0   # Optional probe offsets from the nozzle or tool head
leveling-strategy.delta-grid.save                 false    # If the saved grid is to be loaded on boot then this must be set to true
leveling-strategy.delta-grid.initial_height       5        # Optionally an initial_height can be set that tell the intial probe 
                                                           # where to stop the fast decent before it probes, this should be 
                                                           # around 5-10mm above the bed

wolfmanjm, 2017-02-09:
Please copy the log from the G31 and provide the full unedited config as instructed. I am unable to work from what you have provided.

wolfmanjm, 2017-02-09:
It says provide a LINK that means paste you full config and M503 to pastebin.com or equivalent and paste the link to it here.

wolfmanjm, 2017-02-09:
You are not providing the requested information, when you do so then please reopen this.
It asks for version of board ot asks for operating system etc etc please be more careful to follow the guidelines.

closed 2017-02-09 by wolfmanjm

wolfmanjm, 2017-02-09:
FWIW don't bother to comply with the requests. We are going to remove the is_square support from delta grid entirely. It will be replaced by a proper rectangle grid implementation.

qharley, 2017-02-09:
I am rewriting the Zgrid code at the moment. Any requests?

wolfmanjm, 2017-02-09:
@qharley I'd wait until you see what we come up with. unless you really want to do your own :)

qharley, 2017-02-09:
@wolfmanjm I don't know it I can wait that long.  I will give it a go to
replace my working ZGrid and you can take a look at it, no pressure and no
pull request.
I would love to get a grid leveling back into Smoothieware that works well
on my machines.

wolfmanjm, 2017-02-09:
@qharley we should have it done by tomorrow :) it is basically the original that Johann hacked to get the delta grid. The original was rectangular, he hacked it for a round bed, LoH will hack it back again to get the original rectangular grid.

lordofhyphens, 2017-02-10:
#hackhackhack

pflugj, 2017-02-10:
I tried to be as brief as possible and forgot to mention that I used a custom menu entry
to issue the Delta Grid using a BLTouch sensor like so:
custom_menu.delta_grid.command        M280_S3.0|G31

This has worked until I installed the latest edge from the 7th of Feb 2017.
So it did not make sense to me to include for ex. the OS 
and I didn't even try to use OctoPrint to issue the command ...stupid me.

Anyway, thanks for the info about the new coming grid leveling ;-)
For now I am back to the master release.

Thanks and keep up the great work
Best regards from Namibia
Jurgen

wolfmanjm, 2017-02-10:
try the current edge you will need to build it.