Smoothieware/Smoothieware #1119: DHCP fails due to discarded ethernet response frame

reedb, 2017-02-10:
## How to file an issue

If you encounter an issue with SmoothieWare, you are welcome to
[submit an issue](https://github.com/Smoothieware/Smoothieware/issues)

**Please note** while other projects will use github issues as a general discussion forum, we try to restrict the issues to **actual** issues only. If you want to discuss new features, ask questions, or anything else that is not an issue, please go to the Smoothie forums, mailing lists, IRC, social media etc.

**DO NOT ASK QUESTIONS HERE** they will not be answered and the issue will be summarily closed.

Before you do that for the first time though please take a moment to read the
following section *completely*. Thank you! :)

### What should I do before submitting an issue?

1. **this is not the right issue tracker if you are running a
   forked version of Smoothieware**. Seek help for such unofficial versions from
   the people maintaining them instead.

2. Please make sure to **test out the current edge version** of Smoothieware to see
   whether the problem you are encountering still exists, and **test with a clean config**.

3. The problem still exists? Then please look through the
   [existing tickets](https://github.com/Smoothieware/Smoothieware/issues?state=open)
   to check if there already exists a report of the issue you are encountering.

   **Very important:** Please make absolutely sure that if you find a bug that looks like
   it is the same as your's, it actually behaves the same as your's. E.g. if someone gives steps
   to reproduce his bug that looks like your's, reproduce the bug like that if possible,
   and only add a "me too" if you actually can reproduce the same
   issue. Also **provide all information** as [described below](#what-should-i-include-in-a-bug-report)
   and whatever was additionally requested over the course of the ticket
   even if you "only" add to an existing ticket. The more information available regarding a bug, the higher
   the chances of reproducing and solving it. But "me too" on an actually unrelated ticket
   makes it more difficult due to on top of having to figure out the original problem
   there's now also a [red herring](https://en.wikipedia.org/wiki/Red_herring) interfering - so please be
   very diligent here!

### What should I include in a bug report?

Always use the following template (you can remove what's within `[...]`, that's
only provided here as some additional information for you), **even if only adding a
"me too" to an existing ticket**:

#### What were you doing?

    [Please be as specific as possible here. The maintainers will need to reproduce
    your issue in order to fix it and that is not possible if they don't know
    what you did to get it to happen in the first place. If you encountered
    a problem with specific files of any sorts, make sure to also include a link to a file
    with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of Smoothieware

    [Can be found with the version command. (@version in pronterface)]

#### Hardware model you are using (Make, and version)

    [eg Smoothieboard 5X V1b, Azteeg X5 Mini V2, ... etc]

#### Host and Operating System running the Host

    [eg Pronterface, Octoprint, linux x64]

#### Link to and console logs

    [On gist.github.com or pastebin.com. Always include and never truncate.]
    
#### Link to running config and M503 results

    [On gist.github.com or pastebin.com. Always include and never truncate.]

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Copy-paste this template **completely**. Do not skip any lines!

It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)

arthurwolf, 2017-02-10:
I believe you made a mistake somewhere, there is no content.

reedb, 2017-02-10:
Working on that now. I just inserted the blank template apparently.

-Reed

reedb, 2017-02-10:
Updated now. Do you want to have a short discussion on IRC? -Reed

wolfmanjm, 2017-02-10:
I believe this is an issue with your DHCP server sending too big packets. UDP packets are not supposed to exceed 512 bytes.
Increasing this wastes memory IMO and is not an option for the general users. Memory is extremely tight in smoothie, generally increasing the UDP packets size just for one DHCP server that seems to be sending illegal packets is not an option I am afraid.

closed 2017-02-10 by wolfmanjm

wolfmanjm, 2017-02-10:
Also setting the UIP buffer to 600 affects all network not just UDP, and will significantly slow down TCP/IP as well, the 400 was chosen to be optimum speed without using too much memory.
I suggest you reconfigure your DHCP server to send legal sized UDP packets (perhaps turn off a few options in it). Or use static IP instead, or of course you can run your own build with UIP packets set larger but look out for out of memory crashes and slower network performance.

reedb, 2017-02-10:
My DHCP server is a WRT160N V3. A pretty popular router. I see no options which would allow a decrease in the DCHP options size. I was looking for some field that could be set in the discover packet that might limit the size of the response. I tried using UIP_CONF_DHCP_LIGHT but that didn't help. I'll also note that 400 < 500 so there may be other UDP packets that have issues.

wolfmanjm, 2017-02-10:
well sorry about that. However there is a fairly strict spec on UDP packet size which says < 512 byes if you want it to traverse all known routers. As we only use UDP for DHCP, and DHCP packets are generally well under 400 bytes it is not usually an issue. Maybe you have a  particularly long host name? either way 600 bytes exceeds the standard recommended length for UDP so the router is at fault here I am afraid.
So you have the options I outlined earlier. I suspect one of those will work for you.

wolfmanjm, 2017-02-13:
I researched this a little more, and found this in RFC1533.

```
9.8. Maximum DHCP Message Size

   This option specifies the maximum length DHCP message that it is
   willing to accept.  The length is specified as an unsigned 16-bit
   integer.  A client may use the maximum DHCP message size option in
   DHCPDISCOVER or DHCPREQUEST messages, but should not use the option
   in DHCPDECLINE messages.

   The code for this option is 57, and its length is 2.  The minimum
   legal value is 576 octets.

    Code   Len     Length
   +-----+-----+-----+-----+
   |  57 |  2  |  l1 |  l2 |
   +-----+-----+-----+-----+
```

Unfortunately we cannot set the buffer size to 576, so in the event DHCP fails to work the only alternative is to use static ip addresses, I'll add this to the wiki.

FWIW it seems many DHCP servers send less than 400 bytes in the reply packets.

reedb, 2017-02-14:
I did add a one line comment to that effect:

"Due to buffer size limitations, some DHCP servers are not supported."

Thanks, -Reed