Smoothieware/Smoothieware #1126: Actions in outside workspace

userosos, 2017-02-16:
Firmware can drive an hot-end or hot-bed outside the workspace. It very dangerous. Can you add in to config file ban the action (Hot-end outside from workspace).

arthurwolf, 2017-02-16:
Hey.

First off, note that you are not respecting the rules for posting issues, and that issues are for bugs, and should not be used to request features ( there are a lot of other ways to do that ).

Second : this is a feature called "soft endstops", and it is a planned feature, but it is very difficult to implement correctly, so it is taking time.

Cheers.

closed 2017-02-16 by arthurwolf

userosos, 2017-02-16:
I'm sorry. I understand. How soon it do?

arthurwolf, 2017-02-16:
It is a very big project, I do not know when it will be done, anyone who can code C++ is welcome to try to help.
There is some information here : https://docs.google.com/document/d/1U6nzx1boqF-J2GGPWF4yIaaVib0JNodVWSKBfwiyp_M/edit#heading=h.79oxultkpcgc

userosos, 2017-02-16:
I Am an engineer. I am not know code C++ but i can think about how realize scheme on principal level. I need for this?

arthurwolf, 2017-02-16:
At this point we already have a principal level scheme, we mostly only need the actual code to be written.
But thanks for offering.

qharley, 2017-02-16:
I have a functional soft endstop solution running on the Morgans.
Works well.

I will give it another go with the specification in hand.   lets see

userosos, 2017-02-16:
Very well.