Smoothieware/Smoothieware #1128: Support for analog z-probe processing

jonkean, 2017-02-20:
I'm using JohnSL's FSR controller on my delta printer (https://github.com/JohnSL/FSR_Endstop),
Since slow 8-bit processor are not a limitation, it would be better to integrate this logic into the printer software.

I propose adding analog z-probe support to Smoothieware, and let it handle the z-probe triggering logic.  FSR's would be connected in series to a single analog sensor input (or if available, each FSR could have an analog input of their own)

The benefits would be:
- configure z-probe using config file
- auto tuning of z-probe ambient and trigger values
- pausing print on extreme z-probe values (ie: effector crashes into build plate)

Since FSR's detect pressure on the bed (including the printed object), this opens the door to some interesting ideas:
- adjusting z-height in real time - is the print head pushing on previously built layers?  Should it move up until pressure is relieved?  Or should it reduce layer extruder flow for that area?
- verify print progress, by pressing down on stiff infill structure and confirm z-probe detects pressure.  If no pressure, then pause print as something went wrong.
- per layer z-level adjusting, by periodically z-probing the print itself and adjust z-probe grid values

wolfmanjm, 2017-02-20:
You are welcome to try to do this and submit a PR. but the core team will not be doing this.

closed 2017-02-20 by wolfmanjm

wolfmanjm, 2017-02-20:
IMO FSRs need to be handled by an external board not by smoothie, It does add a significant load to smoothie and would be very hard to implement in a way that did not interfere with other operations.
But having said that it should be possible in theory to do. I wish you luck in trying it :) If you need help please use the IRC #smoothiedev channel to ask questions.

jonkean, 2017-03-13:
Alrighty.  I might take a shot at hacking JohnSL's external controller code into Smoothie firmware.  The code is pretty simple and only needs to run during z-probing.