Smoothieware/Smoothieware #1135: Unnecessary alarm (and blocking) after ctrl-x

cprezzi, 2017-02-23:
#### What were you doing?

    Connect via USB to smoothieware (firmware-cnc).
    Query version.
    Send ?.
    If state is idle, send ctrl-x.

#### What did you expect to happen?

    Nothing. Smoothie should stay in idle state.

#### What happened instead?

    Smoothie raised an alarm and is blocked.

#### Branch & Commit or Version of Smoothieware

    edge-9399ed7, Build date: Feb  7 2017 22:14:13, MCU (firmware-cnc.bin)

#### Hardware model you are using (Make, and version)

    Cohesion3D Remix

#### Host and Operating System running the Host

    LaserWeb4, Windows 10

#### Link to and console logs

    
    
#### Link to running config and M503 results

    ; No config override
    ;Steps per unit:
    M92 X53.33000 Y53.33000 Z80.00000 
    ;Acceleration mm/sec^2:
    M204 S1500.00000 Z300.00000 
    ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
    M205 X0.02000 Z-1.00000 S0.00000
    ;Max cartesian feedrates in mm/sec:
    M203 X400.00000 Y400.00000 Z5.00000
    ;Max actuator feedrates in mm/sec:
    M203.1 X400.00000 Y400.00000 Z5.00000 
    ;WCS settings
    G54
    ;Digipot Motor currents:
    M907 X0.70000 Y0.25000 Z0.70000 A0.90000]

arthurwolf, 2017-02-23:
What did you expect Control-X to do ?

cprezzi, 2017-02-23:
If moves are currently executing, they should be stopped and an alarm should raise, like it does now.
But when no move is executiong and the planer is empty, nothing should happen (like on Grbl).

arthurwolf, 2017-02-23:
Well this is the way C-X is currently being defined in Smoothie, I don't
think it's "wrong" per-se, but I guess it makes sense the way you are
describing too. If this makes us closer to GRBL in terms of behavior, I
think if somebody wants to modify this it's acceptable.

cprezzi, 2017-02-23:
Right, it's not wrong like it works now, it's just a bit inconveient for the user.

cprezzi, 2017-02-23:
Let's see if somebody finds the time to implement it.

cprezzi, 2017-02-23:
@tbfleming Do you think you could look into this sometime?

cprezzi, 2017-02-23:
Am I correct that this behaviour could probably be solved by changing https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/Kernel.cpp#L231 like this:

    void Kernel::call_event(_EVENT_ENUM id_event, void * argument){
        bool was_idle= true;
        if(id_event == ON_HALT) {
            was_idle= conveyor->is_idle(); // see if we were doing anything like printing
            if (!was_idle) this->halted= (argument == nullptr);
        }
    
        if(id_event != ON_HALT || !was_idle) {
            // send to all registered modules
            for (auto m : hooks[id_event]) {
                (m->*kernel_callback_functions[id_event])(argument);
            }
        }
    
        if(id_event == ON_HALT && this->halted && !was_idle) {
            // we need to try to correct current positions if we were running
            this->robot->reset_position_from_current_actuator_position();
        }
    }

Not absolutely sure if `conveyor->is_idle()` allway is false if something is running or planner not empty.

wolfmanjm, 2017-02-23:
No sorry this is how it is implemented and for various reasons I do not want to change it. - Sorry about that.

wolfmanjm, 2017-02-23:
One reason it aborts regardless of state is for the heaters, and other potential modules that may need an abort whether it is moving or not. In the heater case it turns off all heaters. GRBL does not have the heaters or other modules so does not need this behavior.

closed 2017-02-23 by wolfmanjm

cprezzi, 2017-02-23:
I understand that. Probably somebody with deeper knowledge of smoothieware can solve it correctly sometime.

wolfmanjm, 2017-02-23:
It is solved correctly for Smoothie as I explain above. The need to abort in Smoothie goes beyond whether the motors are moving or not IMO.

cprezzi, 2017-02-23:
The main thing is, you know what you want.

openhardwarecoza, 2017-02-23:
Cant we have that behavior in firmware-cnc.bin then?  Ie no heaters.?  We get a lot of user complaints about this issue.  Just trying to solve it at the root

openhardwarecoza, 2017-02-23:
(; i mean you do pitch smoothieware as a 'not only 3d printing'  firmware

qharley, 2017-02-24:
Is easier to just roll your own firmware and have your customers use that.
It is not possible to please everyone, and especially if smoothie
developers see no need or use for your code for their machines there is no
chance for it to be included, even if it is a very useful feature.

Just fix this and use it.  My 2c

openhardwarecoza, 2017-02-24:
> 282113614>,

Good point @qharley -  are you maintaining your own fork for Morgan?

qharley, 2017-02-24:
Yes. Works well for me.

openhardwarecoza, 2017-02-24:
> issues/1135#issuecomment-
> 282198293
> tOgjcMkxkLyTQ7VWe2yAIks5rflZMgaJpZM4MKCgg
> 282205255>,

Good to know.

If we do end up on that route at least good to have someone we already
know/trust to get into the team when we need to do maintenance (;