Smoothieware/Smoothieware #1136: Config File Handling Highly Unstable

lgTethers, 2017-02-24:
I believe there's a serious bug in the "FileConfigSource.cpp" file.   In the "transfer_values_to_cache()" function, it creates a "cv" object, which could very well be "NULL" from the return of the "process_line_from_ascii_config()" call (if the line in the config file is an empty/commented/whitespace line).   Without checking for this NULL case, the program proceeds to de-reference the pointer "if(cv->check_sums...)" many times.   

I believe there may be other worrisome code in the config file parsing/handling as well, as I can crash the program with "config-get" requests over the serial port, which cause a re-read of the config-cache.   

Here is the build I'm running, should be the very latest:
Build version: edge-9399ed7, Build date: Feb  7 2017 22:13:16, MCU: LPC1769, System Clock: 120MHz

I'll attach a sample config file which crashes immediately on the first "config-get" on the serial port.   You can make your own by just copying the last 30--40 lines of the file a couple times to the end of it, basically increasing the file size and number of config parameters the parser attempts to create.   

Here's the output from my serial port (it just hangs) :
Smoothie

ok

ok

config-get beta_step_pin

And here's the output from the debug console:
Build version: edge-9399ed7, Build date: Feb  7 2017 22:13:16, MCU: LPC1769, System Clock: 120MHz
5 axis
NOTE: One extruder configured and enabled
NOTE: 1 extruders enabled out of 1
Watchdog enabled for 10.000000 seconds
File not found: /sd/on_boot.gcode
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
WARNING: duplicate config line replaced
$T050c:cc730010;0d:a87d0010;0e:0d250400;0f:6edb0300;#a8

Notice the re-parsing of the config cache, with multiple warnings of duplicates (should not crash, even with duplicates).

The config file to repro this is just the standard config with tens of duplicate lines copied to the end of it (to simulate a longer config file).

#### Branch & Commit or Version of Smoothieware

Edge as of 2/24/17

#### Hardware model you are using (Make, and version)

   Smoothieboard v1.0b

#### Host and Operating System running the Host

   Windows7, terminal program is PuTTY

#### Link to and console logs

    Pasted above
    
#### Link to running config and M503 results
M503 results:
M503
; No config override
;Steps per unit:
M92 X80.00000 Y80.00000 Z1600.00000
;Acceleration mm/sec^2:
M204 S3000.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X500.00000 Y500.00000 Z5.00000
;Max actuator feedrates in mm/sec:
M203.1 X500.00000 Y500.00000 Z5.00000
;WCS settings
G54
;Digipot Motor currents:
M907 X1.50000 Y1.50000 Z1.50000 A1.50000
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00
;E Steps per mm:
M92 E140.0000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/sec²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E50.0000 P57988
;PID settings:
M301 S0 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P300.0000
;PID settings:
M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000

ok

[bugConfig.txt](https://github.com/Smoothieware/Smoothieware/files/800536/bugConfig.txt)

arthurwolf, 2017-02-24:
Aren't you just running out of RAM ? Smoothie's RAM is pretty tight at this
point, increasing the file size too much is expected to crash.

wolfmanjm, 2017-02-24:
Yea all lines that are not required need to be removed from the config to keep th ememory use low. You can check the memory use with the mem command.

This is not an issue.

wolfmanjm, 2017-02-24:
I will check for the null pointer dereferencing though.

closed 2017-02-24 by wolfmanjm

wolfmanjm, 2017-02-24:
I fixed the null ptr dereference but it really was not causing an issue as it was a read.
Running out of memory is nothing that can be fixed you just need to keep the config file shorter or comment out unused lines

lgTethers, 2017-02-25:
The config file that I attached that crashes the program is just full of duplicates, which are not stored twice, if I understand the code correctly.    Also, I believe you can crash it with extra lines that are all comments, which shouldn't be the case if comments are ignored.