Smoothieware/Smoothieware #1138: Still problems with config file handling/parsing (see #1136)

lgTethers, 2017-02-28:
Please see bug 1136 for full details and a sample of a config file which should be just fine, but crashes the system on the first "config-get".

Bug 1136 was rapidly closed, with lack of memory blamed.   I've tested and memory appears fine.

I'm on the edge build, here is the output of the "mem" command--plenty of memory; we are up and running, parsing a ~400 line ascii config file/cache should not crash the system with ~7.5kB of ram available, no?   I'm not sure I completely follow the output of the "mem" command, though.    The "used heap size" is larger than what's allocated (18832 > 14828), with the "Total Free RAM" being less than both of them...10k

Here's the output:

Smoothie

ok

mem
Unused Heap: 7520 bytes

Used Heap Size: 18832
Allocated: 14828, Free: 2508

Total Free RAM: 10028 bytes

Free AHB0: 13160, AHB1: 10440

Block size: 100 bytes

ok

CRASHES UPON: config-get panel.enable

Output from debug console:
$T050c:cc730010;0d:a87d0010;0e:0d250400;0f:6edb0300;#a8

Commented lines, whitespace, empty lines, etc, should be parsed and ignored, correct?   Should not be crashing...

If I am mis-interpreting the output of the "mem" command, please let me know what to look for, and I apologize for the confusion.

Thank you in advance for your help.

wolfmanjm, 2017-02-28:
you need to debug this, run gdb and supply a crash report. This is explained on the wiki. I'm closing this again because to be honest the config system has been working fine for years so whatever your problem is is not common.
Once you perform a crash debug and attach it to this issue I'll re open it.

closed 2017-02-28 by wolfmanjm

wolfmanjm, 2017-02-28:
The issue is 99% certainly out of memory. You cannot really load the coinfig cache once the system is started unless you have a lot of modules disabled. The cache can take upto 10k of memory depending on the size of the config file.

wolfmanjm, 2017-02-28:
a 400 line line config file is ridiculously large. cut it down to half that size. Why do you need so much in your config file?

wolfmanjm, 2017-02-28:
I added this to the wiki under the config-get command. This is really used for debugging and should NOT be used in everyday use.

WARNING it is HIGHLY recommended to NOT use this command as it uses a huge amount of memory to reload the config file. use cat /sd/config instead

lgTethers, 2017-02-28:
Thanks.   It would be extremely helpful if we could chat for 10min about this, and my particular use-case.   If I emailed you my phone number would you have time this afternoon?

wolfmanjm, 2017-02-28:
I do not believe you really understand what open source code is....

Basically we are volunteers, our time is valuable. I believe if you feel this issue is so important I recommend you try to fix it yourself. If this were a bug that affected many users I'd be more willing to look into it in more depth. This is an edge case caused by abusing a system that is already known to be extremely memory sensitive and short of memory. Using the supplied config files it works adequately.

If you want to talk to someone who is not a volunteer may I suggest you contact wolf.arthur@gmail.com. He sells smoothieboards.

I do not wish to be rude but maybe this will explain the situation more clearly :)

https://hueniverse.com/2016/01/26/how-to-use-open-source-and-shut-the-fuck-up-at-the-same-time/

Please do not be offended.

lgTethers, 2017-02-28:
Ok, I won't ask to speak with you.   Perhaps you could at least explain the output of the "mem" command that you pointed me to?   The labels for the numbers it returns do not make sense.  

The unused heap is 7.5kB?
The used heap is ~19kB?
The "allocated" is ~15kB 
The "free" is ~2.5kB
The "Total Free RAM" is 10kB?

The only math that makes sense with these labels/numbers is the Total Free RAM minus the "free" seems to equal the "Unused heap."      How is the "used heap" so much larger than the "allocated" and the "total free RAM?"

Lastly, my original config file which turned me on to this vulnerability is full of comments and whitespace for readability (I don't consider adding comments and spacing to be "abusing a system"), which, if my reading of the parser is correct, should not create a larger cache as a config object is not created when it encounters such a line.   Anyways, that file likely only had about 150 actual parameters that were uncommented.   

You can start with the original config file you provide, and just copy the last 30--50 lines of the file to the end, creating a list of duplicate parameters, extra whitespace, and extra comments (this is the file I attached to #1136)  and get the same behavior.   It shouldn't crash if duplicates don't actually create a larger config-cache, correct?   

I feel I found some concerning behavior that should not be happening (adding comments/whitespace/duplicate parameters increasing the config-cache size/behavior).   It is easy to reproduce with the edge version files, but if it won't be examined any further on this forum, it would at least ease my mind that it only shows up when the cache is examined through the serial port.   I hope to use this in a real-world environment with a program like Pronterface, and worry that it may send a query or something similar that causes the config-cache to be reloaded and I end up with a hung system.   

If what I'm seeing with the serial console "config-get" command is just an edge case that causes conditions that don't reproduce with a program like Pronterface, then I guess I can just let this go...

Is that the case?

Thanks

wolfmanjm, 2017-02-28:
it is a malloc memory chain. Highly fragmented. Free does not mean it can be allocated as it maybe 1000's of tiny fragments, and not one contiguous space. The Unused Heap: 7520 bytes means sbrk has not been called on that and is still available if the malloc cannot find a free chain.
Every malloc system is different I believe ours is based on the nano-newlib from gcc, there is documentation in the gnu toolchain that explains some of it.

FWIW it is correct to assume that once boot has finished and the config cache has been cleared it will not be loaded again unless you call config-get which is not used under normal circumstances.
If it runs out of memory during boot it will never start and the leds will not start flashing.

I did look briefly at the code and it does look like blank lines and comments are not added to the config cache, so should not cause an out of memory condition.

lgTethers, 2017-02-28:
Thanks for the info.   For now I'll rest easy that this shouldn't occur in real-world situations.   It is weird that adding duplicates from the base, working config file, could cause the crash.

wolfmanjm, 2017-03-01:
duplicate config entries cause a memory leak as they are replaced before being deleted. that is a bug, but as there are rarely any duplicate config entries, and the lost memory is recovered when the cache is discarded it is not worth fixing right now, although I will put it on the todo list.

lgTethers, 2017-03-01:
Thanks.   I'll pull the update once you've fixed it.

wolfmanjm, 2017-03-01:
it is fixed #1139