Smoothieware/Smoothieware #1141: AB axis does not work as expected

cangelis, 2017-03-01:
#### What did you expect to happen?

Trying to make 6 axis work with the 3D Printer (not `CNC` mode).

#### What happened instead?

Steppers just vibrating instead of moving. See the video: https://goo.gl/photos/ctcRwbHv7oY9933w7. XYZ and C axis works as expected

#### Branch & Commit or Version of Smoothieware

`edge` branch.

Compiled using the commands:

`make AXIS=6 PAXIS=6` -> AB axis does not work as expected

`make AXIS=6 PAXIS=4` -> only A axis does not work as expected

`make AXIS=6 PAXIS=3` -> ABC axis works as expected

Looks like there is an issue with `N_PRIMARY_AXIS` but could not find out anything.

#### Hardware model you are using (Make, and version)

Smoothieboard 5X + 1 external stepper driver (total 6 axis)

#### Host and Operating System running the Host

Mac OS X - Repetier Host
    
#### Link to running config and M503 results

https://gist.github.com/cangelis/7417c95ab337943685ff97c1c3e45a17

wolfmanjm, 2017-03-01:
Your config is non standard I suggest you try with a standard config first and do not set these...

minimum_steps_per_minute 1200
acceleration_ticks_per_second 1000
base_stepping_frequency 50000

These are not supported values.

I also suggest you NOT use the PAXIS setting unless you understand why you need it. If you do need it please explain.

wolfmanjm, 2017-03-01:
You have no current set for the B axis.

wolfmanjm, 2017-03-01:
FWIW 6-axis is NOT supported for 3d printers, so you may have to debug this yourself. We can answer any questions you have but not in this forum. I suggest you use the IRC channel #smoothiedev

wolfmanjm, 2017-03-02:
I am closing this as running Extruders on ABC will require make AXIS=6 PAXIS=3 which you say works as expected so is therefore NOT an issue. (Extruders are NOT primary axis and should never be treated as such).

If you are not running extruders and truly think you need to use PAXIS=6 then explain why you think so, and exactly what behavior you expect and what actually happens and I'll reopen this issue.

Thank you

closed 2017-03-02 by wolfmanjm

cangelis, 2017-03-02:
Thank you @wolfmanjm I could not find any information on the documentation or on the internet about what the primary axis is.

Btw, I removed `endstops_enable` config and endstops with new syntax is also working.