Smoothieware/Smoothieware #1146: Reset positions of ABC axis

cangelis, 2017-03-14:
`G92` does not set the positions of ABC axis so `G92 B0` is not effective. Reviewed the code a little bit and saw that current implementation only handles XYZ axis. I could not find any way to set the positions of ABC axis. 

Is there anyway to reset the positions of ABC axises currently or any plans to implement it?

Thanks.

wolfmanjm, 2017-03-14:
despite this not being the correct use of an issue, and ignoring the how to file an issue docs...

The only way is to use manual homing G28.3 and define the extra homing axis for ABC, which is defined in the wiki.

In future please do not use issue to ask questions as there are plenty of other forums to do this in.

closed 2017-03-14 by wolfmanjm

wolfmanjm, 2017-03-14:
FWIW G92 does not set position anyway it sets the global work space coordinate offset. (See WCS in linxcnc) and generally WCS is not needed for rotary axis like ABC, so it is unlikely support for ABC axis will be added for G92 as it would suck up a lot of RAM for little reason.

What we may do is allow the G28.3 to reset the ABC axis if they exist, and even if there are no endstops defined for them. G28.3 or homing is the only way to set the axis position for any axis as this sets the machine coordinates.