Smoothieware/Smoothieware #1150: M84 or M18 after print often (~30% cases) closes serial connection

arhi, 2017-03-24:
#### What were you doing?

printing. nothing "specific" really. at the end of the print I send M84 to free motors.

#### What did you expect to happen?

motor drivers enable signal to release them and nothing else

#### What happened instead?

motor drivers did got released *but* in addition to that serial communication is often dropped.

#### Branch & Commit or Version of Smoothieware
Send: version
Recv: Build version: edge-c4790b1, Build date: Oct 31 2016 21:37:27, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)

replaced cpu board on teartime up plus 2 with nxp mcu (1768) so I can run smoothieware

#### Host and Operating System running the Host

Octoprint from rpi
Octoprint from orangepi
Screen from orangepi
Screen from linux (fedora 64bit)
RealTerm from Windows (7 64bit)

#### Link to and console logs

http://pastebin.com/WfUdL8Mi
    
#### Link to running config and M503 results

http://pastebin.com/153aRVNN

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

1.     Make sure you use a short good quality shielded USB cable with ferrite beads on it.

- done

2.     Make sure the Smoothie PSU is plugged into the same outlet as the PC you are connected to, otherwise you can get ground loops.

- done

3.     If your printer is unusually large, add ferrite to everything coming from or going to your board ( motors, endstops, thermistors, mosfets, etc )

- done :D

4.     Try using a shorter USBÂ cable

- done

5.     Make sure your power supply is clean ( ideally with an oscilloscope )

- done 

6.     Route wires for motors and mosfets away from signal cables ( endstop, thermistors, USB, etc )

- as much as possible

now, **this is not smoothieboard!** so it's entirely possible that somehow the EMI that happens after enable is released connection is somehow lost as on this printer enable pin apart from being tied to the motor drivers it's also tied to 2 small relays (properly protected with diode) and few fets so the whole printer goes into "sleep" mode when enable is off (the extruder fan goes off, part fan goes off, both PTC amplifiers are turned off and disconnected from the cpu etc etc..).. what is weird is that the board does not reset when this happens, it just disconnects... it's not a big deal, this only happens on M84/M18 and it is probably hardware issue, but since the mcu don't restart and usb2serial is part of firmware I still believe there's a chance that the problem is in the firmware so creating an issue.

I can "fairly easy" reproduce the problem (M17, M18, M17, M18. and it will disconnect after few iterations) so if there's a manual on how to debug the darn thing please point me to it :)

wolfmanjm, 2017-03-24:
Not a firmware issue, so closing. I suggest you check the EMI, and check with whoever you bought the hardware from.

wolfmanjm, 2017-03-24:
FWIW the routing if the USB signals is very very critical. They are high speed and ANY misrouting of those traces WILL cause it to disconnect.
That is why smoothieboards are carefully routed and are 4 layer boards.

closed 2017-03-24 by wolfmanjm

arhi, 2017-03-24:
thanks, important for me to know it's for sure not the fw issue... 

I know about usb signals, have some experience with it, unfortunately this motherboard is PRC design (fairly weird I have to say) and cpu replacement board is "eagle/kicad throw it in, make it somehow work" so it's probably EMI but had to rule the fw first (as original firmware never had the issue with losing connection so..)

triffid, 2017-03-24:
I've regularly had serial connection uptimes of multiple months on genuine
smoothieboards, even with soft power (smoothie itself turns motors and
power off after print finishes) and me using power tools right next to the
printer on the same power strip!

PCB design is critical for reliable performance when moderate current high
∆v/∆t signals, analog, and high speed digital all have to coexist.

Something I noticed on a certain PRC clone is that they routed the USB
traces *underneath* one of the stepper drivers! If you have this board,
that design choice would definitely be prime suspect for causing USB
disconnects on M84...

arhi, 2017-03-24:
@triffid that's great to know! ... well I did tested this with a **very** dirty (electrically) drill and was not able to break the serial connection, tried 15kV zapper around the box, tried many things - can't reproduce the problem... I send M84 - puff, the serial conn is gone :( .. so I wrote an issue here just to check if there's a possibility of the bug nothing else... it's clear that it's not a fw issue (as there's no way "it only crashes for me" :D so ..)

this is not a prc clone, I would never actually purchase one, I'd either get original or I'd made it myself, but this is some "upgrade" of a teartime printer that comes with good working but completely closed source electronic. they divided the electronic to "motherboard" with all the protection, amps, drivers etc and a cpu board that's just mcu.. someone ( tinyFab ) created a replacement cpu board with LPC1768 and it kinda works (well I had to make that pt100 sensor type as there's some weird pt100 amp on the motherboard and this serial dying on M84).. 

the cpu board schematic: https://drive.google.com/file/d/0BxnykIRpb264WDBLdTdFVXpKMG8/view
as you can see there's really nothing going on there, few decoupling caps, few jumpers and going straight into the 2 dual inline connectors.. don't have the pcb routing layout :( to look how the usb is routed, maybe I take some images later and try to debug it... but should be straight forward and since this is usb1.2 or maybe usb2 there should not be some super nasty routing problem... apart from crossing over high power signals and ground/power planes it should be ok, and there are no high power signals on the board :D

arhi, 2017-03-24:
there actually are some layout files on thingiverse :D .. if anyone want to take a look and comment (not that it's anything easy fixable since the hw is already done ages ago)
http://www.thingiverse.com/thing:1655144

arhi, 2017-03-24:
hm, traces look ok to me, I don't see a problem there... I do see that a pin
P2[9]/USB_CONNECT/RXD2 is driving a 1.5k pnp pullup to 3V3, not sure what's that doing and could it be related?

triffid, 2017-03-24:
That pullup tells the host that this device is ready to speak USB full
speed, it's part of the USB spec and quite important.. Without it, your
computer will never realise that anything is connected at all!

arhi, 2017-03-24:
ah cool, I usually have just the pullup, without the transistor

triffid, 2017-03-24:
That won't work with smoothie, the host will assume the USB device is
broken due to timeout while smoothie is still checking the SD card for
configuration. Very early smoothieboard prototypes were wired this way and
that's exactly what happened. It would also render DFU unusable since
smoothie couldn't issue bus resets

arhi, 2017-03-24:
makes sense :D thanks for the clear-up!! 
will try to see what's going on with that signal and with the rest of the usb, don't get why the .. this board loses connection .. traces to usb look to me totally ok on the daugterboard, and the motherboard works for sure as same thing worked for years with original cpu board