Smoothieware/Smoothieware #1154: Overheating issue when Probing the bed

BS3DP, 2017-04-06:
#### What were you doing?

 Regular print though octoprint, usb and sd card all run into the same error.

#### What did you expect to happen?

Regular Print to start auto probe the bed with a BL Touch and proceed to print as normal.

#### What happened instead?

The printer heats up the bed as it should and starts to heat up the hotend as it should. Once the hotend is at its set temperature the printer runs through my startup gcode for probing the bed with a BLTouch and then as it probes in a 4x4 grid no matter what is sending the gcode, it gets stuck at a temperature and doesn't read a new temperature until the probing is done in which the nozzle has been left on causing it to drastically overheat and disconnect once a new temperature is measured because its typically well over 20c higher from what's set. Sometimes it doesn't even finish probing before it overheats, My best guess is when it probes it polls the thermistor of the bed and hotend a lot slower when probing the bed.

#### Branch & Commit or Version of Smoothieware

   Send: N49 M1000 version*117
Recv: Build version: edge-7f416bb, Build date: Apr  4 2017 04:05:26, MCU: LPC1768, System Clock: 100MHz
Recv: 5 axis
Recv: ok

#### Hardware model you are using (Make, and version)

    MKS Sbase v1.3

#### Host and Operating System running the Host

    Pronter face/windows for USB, Octoprint/raspberry pi for USB/Network Print and MKS TFT and/or REPRAP discount smart controller for SD

#### Link to and console logs

Recv: ok
Send: N907 M400*41
Recv: ok
Send: N908 M114*38
Recv: ok C: X:121.7890 Y:33.9170 Z:0.1300 E:3.5081
Send: N909 M84*31
Recv: ok
Send: N910 M104 T0 S0*41
Recv: ok
Send: N911 M140 S0*108
Recv: ok
Send: N912 M106 S0*109
Recv: ok
[...]
Send: N915 M280 S10.6*78
Recv: ok
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 G90*17
Recv: ok
Send: N2 M82*27
Recv: ok
Send: N3 M106 S0*100
Recv: ok
Send: N4 M140 S70*86
Recv: ok
Send: N5 M190 S70*90
Recv: ok
Send: N6 M104 S215 T0*33
Recv: ok
Send: N7 M109 S215 T0*45
Recv: T:210.0 /215.0 @222
Recv: T:209.4 /215.0 @255
Recv: T:208.9 /215.0 @255
Recv: T:208.7 /215.0 @255
Recv: T:208.6 /215.0 @255
Recv: T:209.0 /215.0 @203
Recv: T:209.7 /215.0 @190
Recv: T:210.5 /215.0 @165
Recv: T:211.6 /215.0 @127
Recv: T:212.7 /215.0 @85
Recv: T:213.8 /215.0 @37
Recv: T:214.9 /215.0 @0
Recv: ok
Send: N8 M280 S10.6*75
Recv: ok
Send: N9 M280 S3.0*126
Recv: ok
Send: N10 G28 X0 Y0*35
Recv: ok
Send: N11 M280 S3.0*71
Recv: ok
Send: N12 G32*43
Recv: Rectangular Grid Probe...
Recv: Probe start ht is 14.931250 mm, rectangular bed width 180.000000mm, height 190.000000mm, grid size is 4x4
Recv: probe at 0,0 is 0.011250 mm
Recv: DEBUG: X0.0000, Y0.0000, Z-0.0100
Recv: DEBUG: X60.0000, Y0.0000, Z0.2438
Recv: DEBUG: X120.0000, Y0.0000, Z0.4625
Recv: DEBUG: X180.0000, Y0.0000, Z0.4500
Recv: DEBUG: X180.0000, Y63.3333, Z0.7725
Recv: DEBUG: X120.0000, Y63.3333, Z0.5325
Recv: ERROR: Temperature runaway on T (delta temp 25.820465), HALT asserted, TURN POWER OFF IMMEDIATELY - reset or M999 required
Changing monitoring state from 'Printing' to 'Error:  Temperature runaway on T (delta temp 25.820465), HALT asserted, TURN POWER OFF IMMEDIATELY - reset or M999 required\x0a'

    
#### Link to running config and M503 results

  Config: https://www.dropbox.com/s/yclvl893vgfxc1n/config.txt?dl=0

M503:
Send: N56 M503*22
Recv: ; config override present: /sd/config-override
Recv: ;Steps per unit:
Recv: M92 X160.00000 Y160.00000 Z800.00000
Recv: ;Acceleration mm/sec^2:
Recv: M204 S600.00000 Z100.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.02000 Z-1.00000 S0.00000
Recv: ;Max cartesian feedrates in mm/sec:
Recv: M203 X300.00000 Y300.00000 Z10.00000
Recv: ;Max actuator feedrates in mm/sec:
Recv: M203.1 X300.00000 Y300.00000 Z2.00000
Recv: ;WCS settings
Recv: G54
Recv: ;Digipot Motor currents:
Recv: M907 X0.80000 Y0.80000 Z1.20000 A1.50000
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00
Recv: ;E Steps per mm:
Recv: M92 E873.2000 P57988
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P57988
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F4.0000 P57988
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P57988
Recv: ;E max feed rate mm/sec:
Recv: M203 E100.0000 P57988
Recv: ;PID settings:
Recv: M301 S0 P42.9560 I2.9172 D158.1319 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S0 P300.0000
Recv: ;PID settings:
Recv: M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S1 P300.0000
Recv: ;Probe feedrates Slow/fast(K)/Return (mm/sec) max_z (mm) height (mm):
Recv: M670 S5.00 K100.00 R0.00 Z100.00 H5.00
Recv: ;Probe offsets:
Recv: M565 X0.00000 Y0.00000 Z0.00000
Recv: 

Copy-paste this template **completely**. Do not skip any lines!

wolfmanjm, 2017-04-06:
Sorry we do not support MKS SBase I suggest you contact them for help. Sounds like a H/W issue.

This is not reproducible.

wolfmanjm, 2017-04-06:
Also G32 is NOT designed to be run from a gcode file and probably will not work like that it is supposed to be run manually.

closed 2017-04-06 by wolfmanjm

wolfmanjm, 2017-04-06:
FYI I would also check for loose wires this means the temp varied drastically, probably a loose thermistor wire that was disloged when it probes.

BS3DP, 2017-04-06:
Thanks for the reply, iv checked all cables, and all seem to be intact and connected correctly but will go over them again. Regarding G32, i guess you're saying it should be run manualy before I print outside a gcode file? Iv been running the probing in the file for a while and have little to no issues. Again ill go back over it all again but I'm really stumped about the matter and thought it was a bug in the firmware as probing makes this happen all the time, if it doesn't overheat and stop once probing is finished it typically just cools down and prints but that's normally out of 5 cycles of heating up and down.

BS3DP, 2017-04-06:
Also, I would like to add, in Australia its quite hard to get a SmoothieBoard from anywhere and to buy it directly from overseas is a bit pricey. I do plan on upgrading to a better board like a genuine SmoothieBoard.

127, 2017-04-07:
I am @127 and I am here!