Smoothieware/Smoothieware #1155: System freezes when calibrating Delta using G32

nebbian, 2017-04-06:
#### What were you doing?

Using the latest Edge firmware, powered on the machine, homed, and sent G32.

#### What did you expect to happen?

I expected the calibration to complete.

#### What happened instead?

The system froze up halfway through the calibration, the menu was unresponsive, and wasn't able to be talked to using Pronterface.  All motors turned off and the print head dropped to the build plate.

This was quite repeatable, and happened on a couple of different versions of edge, some that I had compiled from source.  The one I'm reporting was the one that was downloaded from Github.
Using the master version as reported below was repeatedly able to complete.

It didn't always happen at exactly the same part of the calibration, which leads me to think that it might have something to do with the watchdog.

#### Branch & Commit or Version of Smoothieware

The issue was present in: edge-97b30ad, Build date: Mar 23 2017 22:12:46, MCU: LPC1768, System Clock: 100MHz

The issue was not present (the calibration completed successfully) in master-40f595b, Build date: Jan  2 2017 14:43:03, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)

Panucatt Re-Arm using RAMPS 1.4

#### Host and Operating System running the Host

Pronterface, MacOS X 10.10.5, although this happens even when choosing to calibrate using a custom menu command, so not talking to the computer at all.
    
#### Link to running config and M503 results

https://gist.github.com/nebbian/c28abcd39053342671e9e1030bece50c
https://gist.github.com/nebbian/117290e38e00b1db99a58b74ad2cde5f

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2017-04-06:
This cannot be duplicated so i suspect there is a problem with your setup or it is electrical interference.
please issue the mem command and report what it says this will eliminate potential out of memory issues.

Your config is also bogus you have duplicate entries that are different I suggest you clean it up.

Also the tower angle stuff is not supported so please don't use it, it is never needed. especially if you also have grid levelling enabled.

As this is not a firmware issue as it cannot be reproduced i am closing this for now. But please give the requested information. AS you will still be able to comment on this issue.

wolfmanjm, 2017-04-06:
Also you need to hook up an ftdi to the uart so we can get a crash dump (that is if it is crashing) so we can see what happened, and then fix it if it is a bug. Without that I cannot really help.
Not knowing if it is a crash or just stopping for some other reason is not enough to go on as I cannot reproduce this and I use G32 everyday.
http://smoothieware.org/mri-debugging

wolfmanjm, 2017-04-06:
Also do not run G32 from the menu, this is likely to run out of stack space. When testing only issue G32 from pronterface.

wolfmanjm, 2017-04-06:
Also you need to submit the full console log. There simply is not enough information here to proceed.

Sungod3000, 2017-04-07:
I have the problem as well. I will post log and config as soon as im at my printer tmr. 
But its pretty much as described above. Happens on G31 and G32 and it on G31 it only manages to go to 1 or two points before halting.

wolfmanjm, 2017-04-07:
one thing to check is if you have an aluminum bed then static can build up and discharge into the head which will cause the system to freeze and potential fry the board.

wolfmanjm, 2017-04-07:
config and logs is not enough. If this is to be solved we need a crash dump.. 
http://smoothieware.org/mri-debugging

nebbian, 2017-04-07:
OK I'll try to connect via the UART on the board and see what gets spat out.

If it helps, the problem appears to be introduced between:
edge-9399ed7, Build date: Feb  7 2017 22:13:16
and
edge-97b30ad, Build date: Mar 23 2017 22:12:46

There are no firmware binaries between these dates, so I'll have to manually compile some git commits between them.  Are all commits to edge between these dates valid?  ie if I were to compile them, would they run well enough to verify the issue?

wolfmanjm, 2017-04-07:
well there have been zero changes to the delta calibration strategy between those two dates. So I doubt git bisect will do anything for you.

wolfmanjm, 2017-04-07:
In fact there were no changes between those two commits that could have any effect on G32.

git diff 9399ed7..97b30ad

wolfmanjm, 2017-04-07:
did you try fixing your config? it is a mess has duplicate conflicting settings and is using unsupported features like the angles which need to be removed.
I suspect your messed up config may have something to so with it.

wolfmanjm, 2017-04-07:
Also disable the external sdcard that is also not supported and can cause electrical issues. In fact you should really not connect the external sdcard cables at all as extending them will cause corruption on the internal sdcard.
Your current control is enabled which is an issue as rearm has none.
and countless other config errors.

nebbian, 2017-04-07:
I've done a git bisect on the code, and have verified that the first bad commit is this one:

```
567279443e9630f2f2a957b8166ec2a194f6b252 is the first bad commit
commit 567279443e9630f2f2a957b8166ec2a194f6b252
Author: Jim Morris <morris@wolfman.com>
Date:   Mon Feb 20 14:31:25 2017 -0800

    Allow larger strings for M114 and get pos
    check for sane values of number of axis

:040000 040000 83dfde2c29406b641cd1ceed568e8237cc1434b6 4c2977dcec5d964e2f98c01f58efbf76cd3c4416 M	src
```

After doing the bisect, I have verified that the previous commit (432e0566) produces a workable file, and have tried exactly the same procedure 6 times, 3 times on the previous commit, and 3 times on the 'bad' commit.  Every time, the bad commit froze my board, and the good commit let the G32 procedure finish.

Looking at the commit, it appears that my board might be running out of memory while doing the calibration.  

Doing a mem command shows the following when the bad commit is loaded:
```
>>> @mem
SENDING:mem
Unused Heap: 6080 bytes
Used Heap Size: 20272
Allocated: 14768, Free: 4048
Total Free RAM: 10128 bytes
Free AHB0: 11612, AHB1: 10440
Block size: 100 bytes
>>> @version
SENDING:version
Build version: -5672794, Build date: Apr  7 2017 22:21:45, MCU: LPC1768, System Clock: 100MHz
5 axis
```

And it shows the following when the good commit is loaded:
```
>>> @mem
SENDING:mem
Unused Heap: 6080 bytes
Used Heap Size: 20272
Allocated: 14768, Free: 4048
Total Free RAM: 10128 bytes
Free AHB0: 11612, AHB1: 10440
Block size: 100 bytes
>>> @version
SENDING:version
Build version: -432e056, Build date: Apr  7 2017 21:50:54, MCU: LPC1768, System Clock: 100MHz
5 axis
```

ie, no difference.  

I don't understand how that commit can change the behaviour, unless perhaps some memory hasn't been allocated properly.

I'm going to try connecting my FTDI connector next and seeing what happens when the system freezes.

wolfmanjm, 2017-04-07:
did you fix your config yet? all bets are off with a bad invalid config like you have.

wolfmanjm, 2017-04-07:
The difference between those two commits would not affect anything unless you were issuing M114, as that uses more stack.

You have plenty of memory so yo are not running out of memory.

Without a crash dump It is impossible to say what is happening. It is possible the stack is getting corrupted after a M114, but unlikely.

wolfmanjm, 2017-04-07:
I have a theory that the issue is stack overrun, to test please edit this line
https://github.com/Smoothieware/Smoothieware/blob/edge/src/makefile#L22
and change to..
STACK_SIZE=4096

recompile with make clean followed by make
flash the resulting binary and test.
If it works then the issue is stack overrun, and I will have to think of a way to improve it without losing too much heap space.

wolfmanjm, 2017-04-07:
As I cannot reproduce any of these crashes, I have blindly optimized the stack usage for the probe commands. I checked that into edge, so if you do a git pull, compile, flash and test let me know if it improves anything (make sure you reverse any changes you made to the makefile above first)

nebbian, 2017-04-08:
@wolfmanjm, thanks for the optimization.  I downloaded the very latest edge commit (3ec874df9023b05333bf8931e8c08816091333b1) and compiled it.  
I had also updated my config to remove the current control at the bottom of the file.  Careful examination of my file showed no other duplications of settings. 

I repeated the test ten times with the new firmware, and it completed the test ten times.

I repeated the test three times again with the 'bad' firmware above, and it failed every single time.

So it seems that your suspicion of this problem being related to the stack overflowing might be correct.  The optimisations you did certainly seem to have improved things greatly.  

Thanks again for taking this report seriously, feel free to close this issue.

wolfmanjm, 2017-04-08:
You should fix these other settings too..

```
switch.servo1.input_on_command			M280		# G code to set PWM period.  M280 S5 is full one direction, S7.5 would be midway, S10 is the other extreme.
switch.servo1.input_off_command			M280		# G code to set PWM period
```
You cannot have the same m code for on and off

```
delta_tower1_angle							 -0.23				# Angle offset for tower 1 (alpha)
delta_tower2_angle							 -0.25				# Angle offset for tower 2 (beta)
delta_tower3_angle							 0.0				# Angle offset for tower 3 (gamma)
```

These settings are in compatible with delta grid and need to be removed.

closed 2017-04-08 by wolfmanjm

Sungod3000, 2017-04-18:
Not to rain on the parade but I still have some minor issues with that. 

The optimization definitely fixed it, but yesterday I ran G32/G31 combo with 9 )points and it dropped the connection like before. When I changed to 7 points it worked. I have to repeat that some more times but it seems like the same issue.

and secondly something where I see the same symptom but dont know if its related. When I just turn on the printer for the first time after a while, select a file from the micro sd, the bed starts heating up and after a few seconds the display and the whole machine freezes. When I reset the board and do the same thing again the heat up goes normal and the prints starts just fine.

These are not the biggest issues and i can print like that just an FYI.

wolfmanjm, 2017-04-18:
@Sungod3000  Doesn't sound like firmware issues I am afraid. Without a crash dump then there is no way to debug or confirm either way.

This actually sounds more like H/W issues, like noise on the USB, ground loops, PSU overload etc, check the trouble shooting guide in the wiki for answers.

If you have a chinese clone then it is almost certainly H/W issues.

Sungod3000, 2017-04-18:
unfortunately I cant produce the crashdump with the equipment I have. 

with clones you mean the ramps itself? because my re-arm is official :)

wolfmanjm, 2017-04-18:
re-arm should be ok, but I have been getting a lot of reports of odd issues with re-arm that we don't get with smoothoieboards.

As I said without a crash dump there is nothing I can do further. Crash dumps can be produced with no equipment just a ftdi adapter is required.

try using a shorter USB cable. Or print from sdcard with no USB attached. See if that helps.

Sungod3000, 2017-04-18:
ah i thought ftdi was some sort of programming header. ill check that out.

"I have been getting a lot of reports of odd issues with re-arm that we don't get with smoothoieboards." thats interesting and I totally believe that smoothieware for smoothie boards works flawlessly, since your optimization for the original G31/G32 issues worked so well, so Im kinda hoping that this trend continues :)

wolfmanjm, 2017-04-18:
they are identical in the firmwares view, so it will run identically on both. The issue is noise and how the boards are built and respnond to EMF and noise.

Firmware will not run different on one vs the other so I do not think this is a firmware issue.