Smoothieware/Smoothieware #1157: Webui/Opera interaction causing lock up during G32 Delta Calbration

AxMod3DPrint, 2017-04-07:
#### What were you doing?
Running G32 to Re-Calibrate the printer. This is the 2nd Azteeg X5 Mini which is a replacement for the original board I bought, which did the same thing. I have been through the hardware with a fine tooth comb and this is the only constant with the same issue happening over 2 boards. 

#### What did you expect to happen?
G32 to complete

#### What happened instead?
Having previously uploaded a gcode file via the built in webui over network it completed without issue. Opera was restarted a few hours later with the webui page open and pinned. When the tab was re-opened it seems the file was cached somewhere and it tried to re-upload the file which killed a print in progress and locked the printer up. Everything was fine until I came to re-calibrate the printer via G32 about a day and a half later where it locks up mid process just before the delta radius calibration. Which is what was happening on the first board. I had remembered the same issue was happening with a file re-appearing on the SD even after it was deleted but still selected in the upload box of the web ui. Disabling the network module rectified this issue and G32 completed without issue

#### Branch & Commit or Version of Smoothieware
Send: version
Recv: Build version: edge-75df940, Build date: Jan 18 2017 20:23:52, MCU: LPC1769, System Clock: 120MHz
Recv: 5 axis

However, this has happened on multiple versions up from release to current edge. 

#### Hardware model you are using (Make, and version)
Azteeg X5 Mini v3

#### Host and Operating System running the Host
Desktop used to upload files - Linux Mint 18 Cinnamon - Opera browser, happens with all versions - opera turbo enabled.

Host Control - This has happened both when running from custom commands from the menu, on an  Ubuntu Linux server running Octoprint and a Raspi3 Running Octoprint.

#### Link to and console logs
Nothing shows, printer locks up and loses connection to host
    
#### Link to running config and M503 results
M503 - https://pastebin.com/tzTBPnyG
Config- https://pastebin.com/UEZ0iKb3

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Copy-paste this template **completely**. Do not skip any lines!

It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)

wolfmanjm, 2017-04-07:
You cannot run G32 from the web i/f it is not supported - sorry.

closed 2017-04-07 by wolfmanjm

wolfmanjm, 2017-04-07:
This appears to be a web issue and nothign to do with G32. the issue is mislabeled.

What seems to be happening is refershing the browser is issuing the last command which is a browser caching issue. RE uploading a file is probably corrupting the sdcard, and that causes issues with G32.

The network stuff is not well supported, so this is unlikely to be fixed. Maybe use telnet instead of the web interface if you cannot use USB.

AxMod3DPrint, 2017-04-08:
The SD Card does not get corrupted as it is fine on a reset of the board. Apologies for not labelling correctly. However, the caching issue causes memory issues on the board which in turn cause high memory operations, such as but not limited to, G32. I generally print via USB however on occasion uploading to SD is necessary for high detail/long prints. G32 was ran from USB, screen custom commands and the Webui GCode box.