Smoothieware/Smoothieware #1159: Webpage is useless on mobile

justgook, 2017-04-07:
When you enter documentation webpage from mobile on all screen there is menu... looks like classic bootstrap problem .. I cannot find source code for that.. if u tell me where is source I could try make pull request for fix.. for now it is totally unusable ( tried horezontal and vertical position, menu just left opened)

arthurwolf, 2017-04-07:
Several contributors have offered to fix it, but I have been crazy busy. I'll get to them very soon. Thanks for offering to help.

closed 2017-04-07 by arthurwolf

justgook, 2017-04-10:
You need just add 2 classNames to correct place - and that will fix all (main issues)
here is screens - how it works now, and how after adding `collapse navbar-collapse` to correct place
[image]
[image]
[image]

arthurwolf, 2017-04-10:
Thanks a lot for the instructions.
I just made the changes and it looks like here it works, can you confirm ?

justgook, 2017-04-10:
yup; - screenshot from mobile - finally - ill be able read it from mobile (btw - there still lot small bugs  / visual glitches - but at least it finally possible to use on mobile..)
[image]