Smoothieware/Smoothieware #1170: offset probe points

DeathmannX, 2017-04-17:
#### What were you doing?

    Tryed to move/offset the Probe Points in thx X and Y cause it starts to probe in the air.

#### What did you expect to happen?
    Probe the Bed without crashing the Nozzle in to the Bed?

#### What happened instead?
    Crashing the Nozzle in to the Bed?

#### Branch & Commit or Version of Smoothieware
    Build version: edge-97b30ad, Build date: Mar 23 2017 22:12:46, MCU: LPC1768, System Clock: 100MHz
5 axis

#### Hardware model you are using (Make, and version)
     Capacitive Probe and the Re-Arm Board + Ramps 1.4

#### Host and Operating System running the Host
     Pronterface

#### Link to and console logs
    got no console log. (turn the printer direct off if i see that he tryes to probe in the air)
    
#### Link to running config and M503 results
    Config: https://pastebin.com/xkYaw6Bv
    M503: https://pastebin.com/qtEHnAGM

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2017-04-17:
This works perfectly well for most people, and you do not provide enough information to debug so i am closing this issue. I do not believe this is a bug it is just a users misunderstanding of how things are supposed to work or a configuration issue.

closed 2017-04-17 by wolfmanjm

DeathmannX, 2017-04-17:
It was more a Question than a issue/or bug.
How can i do what i want?

wolfmanjm, 2017-04-17:
Issues are NOT for asking questions as explained in detail in the how to submit an issue. DO NOT use issues to ask questions please. Use one of the many forums to ask questions.

DeathmannX, 2017-04-17:
Did that already and no one can help and the Team don't care about that!

What is with that:
`if(only_by_two_corners){
                if(gcode->has_letter('X') && gcode->has_letter('X') && gcode->has_letter('A') && gcode->has_letter('B')){
                    _x_start = gcode->get_value('X'); // override default probe start point
                    _y_start = gcode->get_value('Y'); // override default probe start point
                    _x_size = gcode->get_value('A'); // override default probe width
                    _y_size = gcode->get_value('B'); // override default probe length
                }`

found that in **src/modules/tools/zprobe/CartGridStrategy.cpp**

should be:
`if(gcode->has_letter('X') && gcode->has_letter('Y') && gcode->has_letter('A') && gcode->has_letter('B'))`