Smoothieware/Smoothieware #1171: CartGridStrategy.cpp "if" is wrong

DeathmannX, 2017-04-17:
#### What were you doing?
    Inspect the files why the hell the: "leveling-strategy.rectangular-grid" not work for me.

#### What did you expect to happen?
    Level the bed with: leveling-strategy.rectangular-grid

#### What happened instead?
    ignored the offset of the Probe and the homed or manual set x0 y0 position.

#### Branch & Commit or Version of Smoothieware
    Build version: edge-97b30ad, Build date: Mar 23 2017 22:12:46, MCU: LPC1768, System Clock: 100MHz
    5 axis

#### Hardware model you are using (Make, and version)
     Capacitive Probe and the Re-Arm Board + Ramps 1.4

#### Host and Operating System running the Host
     Pronterface

#### Link to and console logs
    got no console log. (turn the printer direct off if i see that he tryes to probe in the air)
    
#### Link to running config and M503 results
    Config: https://pastebin.com/xkYaw6Bv
    M503: https://pastebin.com/qtEHnAGM

THIS Line here looks wrong:
`if(gcode->has_letter('X') && gcode->has_letter('X') && gcode->has_letter('A') && gcode->has_letter('B'))`

can be found here: https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/zprobe/CartGridStrategy.cpp#L352

should be:
`if(gcode->has_letter('X') && gcode->has_letter('Y') && gcode->has_letter('A') && gcode->has_letter('B'))`

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2017-04-17:
fixed

closed 2017-04-17 by wolfmanjm

DeathmannX, 2017-04-17:
well no Problem...
Looks like a thank you hurts you. @wolfmanjm

arthurwolf, 2017-04-17:
@DeathmannX : for future reference, this sort of adversorial behavior isn't
ok in the issues ( or elsewhere in the smoothie community )k
This is a community of volunteers spending a lot of their free time helping
others, we'd have to end every sentence with a "thank you" if we were to
thank everyone who deserves it. That's not how it works.

triffid, 2017-04-18:
> 294546350>,

As stated in http://www.catb.org/esr/faqs/smart-questions.html#keepcool ,
careful attention and a prompt fix is vastly more valuable than typing
"thank you", give it a read sometime :P