Smoothieware/Smoothieware #1180: Slow performance with Laser raster

mayhem2408, 2017-04-30:
I have been using GRBL-MEGA 1.1 and Marlin for my laser engravers and decided to upgrade to a Smoothie as I was under the impression it was faster and capable of processing gcode very fast. So far I have seen it is significatly slower. I have been able to push Marlin at a rate of 80mm/s and GRBL at a rate of 60mm/s engraving at a 0.1mm resolution. I run a 40W CO2 laser and have had to run the power on the tube below 20% due to the top speed I was getting out of GRBL and Marlin. I was hoping to up the speed to 200+mm/s and bump up the power on the laser tube. It turns out I can't even get 50mm/s to run smooth on the smoothie. I know that smoothie was originally developed as a 3D printer controller but with laser capabilities. I have been reading for a while that raster support is comming, but it appears no one can get it to work and it apprears dead in the water. Are there any sugestions on how to improve the raster performance of smoothie? I'd like to hear other peoples experiences with rastering.

arthurwolf, 2017-04-30:
> I have been using GRBL-MEGA 1.1 and Marlin for my laser engravers and
> decided to upgrade to a Smoothie as I was under the impression it was
> faster and capable of processing gcode very fast. So far I [...]
No actually Smoothie is faster than 8-bit firmwares for raster engraving,
there is possibly a problem with data transfer or the Gcode you generate
here.

What program do you use to generate the Gcode, and what program do you use
to send the Gcode to the board ?
If you don't already, can you ( for testing purposes ) play an engraving
file from the SD card ?

> Are there any sugestions on how to improve the raster performance of
> smoothie? I'd like to hear other peoples experiences with rastering.
>
> <https://github.com/Smoothieware/Smoothieware/issues/1180>, or mute the
> thread

arthurwolf, 2017-04-30:
Oh I didn't see this was posted on Github.
Github isn't the right place for this sort of question, can you please post
on the forums, or mailing list, or google+ ?

closed 2017-04-30 by arthurwolf

wolfmanjm, 2017-04-30:
use the fast-stream.py to stream the raster to smoothie, it is in the root directory on github.
The issue is not smoothie but your host cannot send the gcode fast enough, it is a driver issue on the host computer. There is a long discussion about that in an issue that was closed many months ago here.

mayhem2408, 2017-04-30:
@arthurwolf I am using laserweb4 to generate the gcode file, and saving onto an SD card to playing from there. I appear to be getting about 160 lines of gcode per second processed from the card. The exact same file to GRBL over USB/Serial is topping out at 350 lines per second. I tried the file to smoothie over serial on still got 160 lines per second, so the bottleneck is not USB or SD Card. I was hoping via the SD card to be pulling 1000+ lines per second, but I am not seeing it.

The Gcode has been optimised with no repeated G0/G1 commands when not needed and not repeating Y axis movement on each line when not needed. The lines are small and tight but smoothie just isn't processing it quick enough, even of the SD card. I will try over the network next.

My intention is not to use USB and the pixel rate I am after isn't practical over serial. Any other suggestions.

arthurwolf, 2017-04-30:
Just out of curiosity can you try Visicut-generated Gcode ?

wolfmanjm, 2017-04-30:
I get over 1,000 pixels /sec using fast-stream.py I suggest you use that, sdcard can be much slower. You probably have an sdcard that is very slow on smoothie.

Proof is in this thread #1096

wolfmanjm, 2017-04-30:
USB is not serial, it is ACM/CDC and is much faster than serial, if you use the fast-stream.py script.

wolfmanjm, 2017-04-30:
You will NOT get good performance over network that will be slower than sdcard and USB.