Smoothieware/Smoothieware #1181: Arc code 'G2' executed in the XZ plane is not performed in the clockwise direction.

xcs400, 2017-05-06:
Hello,
I've founded an issue while running a 'Fusion360' generated gcode:  suddenly my cnc machine move in unexpected direction (and crash the tool ...).

After investigating the Gcode and do many simulation, I 've founded that the Arc code 'G2' executed in the  XZ plane is not performed in the correct direction (clockwise)  by the Smoothie software.

 
#### What were you doing?

i'm trying to generate and arc segment in the XZ plane (G18 cmd + G2).
this simple code shows the problem.
 
G90 G94
G17
G21
G1 X0 Z0 X0 F3009.6
G18 G2 X-2 Z0  I-1  K-1

   
#### What did you expect to happen?
#### What happened instead?

The first picture shows what I expected,  the second picture shows what really append.
https://drive.google.com/open?id=0BykMgE6n44YMYTNiNXdUZjBpaDA

My analyse:
The G2 'clockwise'  in the XZ plane should be normaly be interpreted from the Y axis positive side (looking to 0).  It looks like Smoothie does not do that way ( and perform like an G3 anti-clockwise move )    

note: that if a similare move is done in XY plane,  the G2 move is correct 👍
 ie :  G12 G2 X-2 Z0  I-1  J-1     (the move is done clockwise,  view from the top (positive Z))

 

#### Branch & Commit or Version of Smoothieware

   custom branch.  based on Edge 2016 april.

#### Hardware model you are using (Make, and version)

   custom hardware
#### Host and Operating System running the Host
  windows 
#### Link to and console logs

no log available on my hardware
    
#### Link to running config and M503 results

https://pastebin.com/raw/mH31WLnd

M503 result:

; No config override
;Steps per unit:
M92 X1638.00000 Y1638.00000 Z1638.00000
;Acceleration mm/sec^2:
M204 S1000.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X100.00000 Y100.00000 Z33.33333
;Max actuator feedrates in mm/sec:
M203.1 X61.00000 Y61.00000 Z16.66667
;WCS settings
G54
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00

[image]

wolfmanjm, 2017-05-06:
"custom branch. based on Edge 2016 april". is 1 year old. Please try this on current edge and report back. As that is such an old version it is not really relevant anymore, and the G2/G3 has changed since then.

xcs400, 2017-05-06:
Ok. I will do the port of the current  'Edge' to my hardware and let you know.
Thank you

xcs400, 2017-05-06:
I've re-builded the current Smoothieware Edge and load it on my hardware.
It does not change anything,  The direction probleme is still here.

with:
G90 G94
G17
G21
G1 X0 Z0 X0 F3009.6
G18 G2 X-20 Z0 I-10 K-10

On simulator, the movement starts with    :   Z ; Up direction    X : negative direction           (as expected)
On my Smoothie hard,  the movement starts with:    Z ; Down direction    X :  positive direction

------------------------------
Build version: ~11myvar2, Build date: May 6 2017 23:58:51, MCU: LPC1769, System Clock: 120MHz 
(not on git rep)
------------------------------
config file:
https://pastebin.com/raw/G3UUFXz4

-----------------------------
M503:

; No config override
;Steps per unit:
M92 X1638.00000 Y1638.00000 Z1638.00000
;Acceleration mm/sec^2:
M204 S1000.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X100.00000 Y100.00000 Z33.33333
;Max actuator feedrates in mm/sec:
M203.1 X61.00000 Y61.00000 Z16.66667
;WCS settings
G54
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00
;E Steps per mm:
M92 E140.0000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/sec²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E50.0000 P57988
;PID settings:
M301 S0 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P300.0000
;PID settings:
M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000

xcs400, 2017-06-07:
hello,
I've provided the missing information since a while now but the status has not changed...
Do i need to provide extra information ?
thank you

wolfmanjm, 2017-06-07:
I think the information is ok, but who do you expect to fix this? I do not think this will get fixed unless someone volunteers to fix it. For me this is very low priority and I will not be fixing it.
This is an open source project manned by volunteers, so do not expect things to get fixed just because there is an open issue on it.

wolfmanjm, 2017-09-29:
fixed in #1226

closed 2017-09-29 by wolfmanjm