Smoothieware/Smoothieware #1185: extruder.modulename.z_offset does not work

well-engineered, 2017-05-13:
Setup:
- 3d-printer with multiple individual toolheads, all with a little different z-heights
- Z-axis is moving down, away from XY-stage
- current smoothie-edge firmware

Issue: 
While fiddling around with multiple Hotends and discovered, that the setting "extruder z-offset" doesn't do anything in response on my printer.

For example, neither:
extruder.EXT1.z_offset 1
nor:
extruder.EXT1.z_offset 50

does anything to my printjob, it just prints as if the setting isn't even there.

Things I tried to get z_offset to work like expected:
- I renamed the extruder modules to T0/T1 etc to maybe match the tool identifier. Didn't help.
- Though I am not that good at coding, I took a look into smoothie-source and it seems, that the z_offset is at least programmed in at every place where x&y offsets are mentioned. But I am not even close to sure, that I checked this correctly.
- tried using the gCode-command "G10" (tool offset), because I saw it mentioned in the source. Didn't work either
changed X&Y offsets of each of mine extruders like I did with the Z-offset before. They all work. I tried both, positive and negative values. Every time I get a different position, according to my set offsets. Only Z-offset doesn't work.

wolfmanjm, 2017-05-13:
can you set the Z offset to say 20 on the second extruder then do T1 then in a command window send $G and report the results here?
Please include both your current running config and the results from M503.
Thank you

wolfmanjm, 2017-05-13:
$# is the command to send

well-engineered, 2017-05-13:
Set z_offset to 20:
`Send: config-set sd extruder.T1.z_offset 20
Recv: sd: extruder.T1.z_offset has been set to 20
[...]
Send: reset
Recv: Smoothie out. Peace. Rebooting in 5 seconds...`

Picking T1:
`Send: N5 T1*62
Recv: ok`

sending $#:
`
Send: $#
Recv: [G54:0.0000,0.0000,0.0000]
Recv: [G55:0.0000,0.0000,0.0000]
Recv: [G56:0.0000,0.0000,0.0000]
Recv: [G57:0.0000,0.0000,0.0000]
Recv: [G58:0.0000,0.0000,0.0000]
Recv: [G59:0.0000,0.0000,0.0000]
Recv: [G59.1:0.0000,0.0000,0.0000]
Recv: [G59.2:0.0000,0.0000,0.0000]
Recv: [G59.3:0.0000,0.0000,0.0000]
Recv: [G28:0.0000,0.0000,0.0000]
Recv: [G30:0.0000,0.0000,0.0000]
Recv: [G92:0.0000,0.0000,0.0000]
Recv: [TL0:20.0000]
Recv: [PRB:0.0000,0.0000,0.0000:0]
Recv: ok
`

Sending M503:
`
Send: N69 M503*26
Recv: ; No config override
Recv: ;Steps per unit:
Recv: M92 X80.00000 Y80.00000 Z800.00000
Recv: ;Acceleration mm/sec^2:
Recv: M204 S3000.00000 Z2000.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.30000 Z0.00000 S20.00000
Recv: ;Max feedrates in mm/sec, XYZ cartesian, ABC actuator:
Recv: M203 X1000.00000 Y1000.00000 Z30.00000 A1250.00000 B1250.00000 C125.00000
Recv: ;WCS settings
Recv: G54
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00
Recv: ;E Steps per mm:
Recv: M92 E150.0000 P55428
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P55428
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P55428
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P55428
Recv: ;E acceleration mm/sec��:
Recv: M204 E1000.0000 P55428
Recv: ;E max feed rate mm/sec:
Recv: M203 E120.0000 P55428
Recv: ;E Steps per mm:
Recv: M92 E150.0000 P55685
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P55685
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P55685
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P55685
Recv: ;E acceleration mm/sec��:
Recv: M204 E1000.0000 P55685
Recv: ;E max feed rate mm/sec:
Recv: M203 E120.0000 P55685
Recv: ;E Steps per mm:
Recv: M92 E150.0000 P55942
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P55942
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P55942
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P55942
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P55942
Recv: ;E max feed rate mm/sec:
Recv: M203 E120.0000 P55942
Recv: ;PID settings:
Recv: M301 S0 P75.3000 I10.3140 D137.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S0 P300.0000
Recv: ;PID settings:
Recv: M301 S1 P75.3000 I10.3140 D137.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S1 P300.0000
Recv: ;PID settings:
Recv: M301 S2 P75.3000 I10.3140 D137.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S2 P300.0000
Recv: ;PID settings:
Recv: M301 S3 P80.6000 I10.6740 D152.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S3 P120.0000
Recv: 
Recv: ok
`

config-file in use:
[config.txt](https://github.com/Smoothieware/Smoothieware/files/998846/config.txt)

wolfmanjm, 2017-05-13:
Well according to this...   [TL0:20.0000] the z offset is set to 20.

So on the next move Z will be offset by 20 mm.

How are you testing this? Note relative moves WILL NOT change the offset, only absolute moves, so jogging Z will not change the offset but G1 Z0 will, in that latter case Z will be 20mm above the 0

wolfmanjm, 2017-05-13:
So please try this...

After setting tool offset Z to 20 and selecting the extruder with T1...
issue command `get pos`
then move Z.. G1 Z10
the issue `get pos` 

and paste results.

Thank you

wolfmanjm, 2017-05-13:
Ok I think I see the issue here.
In CNC world tool offsets are NOT applied until M6 is issued. reprap does not use M6.
So in our case what happens is the offset is not applied until that axis changes, so if you change the Z tool offset then do a XY only move Z will not change until there is a change in Z (ie G1 Z10).

I think this is a bug and I will fix it, but theoretically implementing M6 after a tool change should be required. I don't see the reprap community doing this though.

This bug would also affect changing WCS though so it is broader than just tool change.

qharley, 2017-05-13:
So, if the user adds M6 to the tool change Gcode, will it work correctly?

wolfmanjm, 2017-05-13:
correction... this is not a bug in WCS just the way tool offsets are handled.
You need to move in the Z before the tool offset is changed for Z this would be equivalent to issuing M6.

If you move in Z the tool offset you set would be honored, but not until you move in Z. 
This is undefined in NIST GCode spec, but in effect M6 would move the tools, but we do not implement M6 at the moment.

qharley, 2017-05-13:
So the tool change code should include an absolute G0 Z<current>

wolfmanjm, 2017-05-13:
The problem is that neither GRBL/LINUXCNC nor the NIST gcode standard deal with things like extruder offset. The closest is tool offset which is only in Z and handled completely differently.

We are the only reprap firmware that implements WCS correctly if at all, so it is also unclear how the extruder offsets affect the MCS vs the WCS.
This is not going to be an easy one to solve at all.

wolfmanjm, 2017-05-13:
Yes the workaround for now is to do a G1 Zxxx after the T1. Where xxx is wherever you want Z to be after an extruder change. At that point any Z offset in Extruder offset will be applied to the new Z position.
The same is true for X and Y it just happens that X and Y are usually issued after a T1.

I'll continue to examine alternative solutions.

well-engineered, 2017-05-13:
atm I am unable to test this further (it is late at night), but if a z_offset is only applied if the z-axis moves again, this might be the problem in my case.
My slicer of choice is S3D and it does change to the next layer **BEFORE** executing the tool change script. And **AFTER** finishing the tool change script it sends the T0/T1/T2 command, followed by XY-movement because the Z-change already happened. I think, that forcing an output at the end of the tool change script like "[active tool]" and "G1 [current layerhight]" should result in an output like:

;firstlayer z-change 
**G1 Z0.4;**          _; <-  generated normal gcode_
;tool change script: pick tool0    
**[do stuff]** 
**T0;** <- manually written in toolchange script for picking T0
**G1 Z0.4;** <- generated by S3D tool change script via command: "_G1 [current_position_z];_"

And thus forcing the Z-axis to move again. I'll give feedback tomorrow.

well-engineered, 2017-05-14:
What I described in my last post does work as a workaround. Forcing a "**G1 Z[current_position_z]**" after manually writing "**[active_tool]**" results in an additional z-move to adjust the z-position to the z_offset of the corresponding tool.
Should issuing an "**M6**"-command do the same?

wolfmanjm, 2017-05-14:
In theory M6 would do the same thing but it is used for tool change not for extruder changes which are specific to 3d printing and not covered by the NIST standard. It is unclear to me that extruder offsets are the same thing as tool offsets which are usually just in Z.
We do not currently implement M6 as we do not implement tool offsets either. When we do then M6 will work as specified in the NIST standard.

closed 2017-07-10 by wolfmanjm