Smoothieware/Smoothieware #1190: Setting hotend temp (M104), then bed temp (M140), then waiting for bed temp to hit a certain value (M190) can cause thermal runaway in hotend

TheAppleFreak, 2017-05-28:
#### What were you doing?

I was attempting to print out a file when a small part of it lost adhesion while printing the first layer. I cancelled the print in Octoprint, resliced the file in Cura with some different settings, and restarted the print. During this time, both the hotend heater and the bed heaters were turned off and had been cooling down, but had not yet reached room temperature. 

#### What did you expect to happen?

In Octoprint's temperature graph, I expected to see both the hotend temperature and the bed temperature change to reflect their current temperatures in real time.

#### What happened instead?

[image]

As seen in that screenshot, both heaters began heating up as normal at the same time, but only the bed thermistor appears to have been polled. The hotend temperature stalled at 116.3 C for several ticks, and when the bed temp hit its target of 50 C the hotend temp suddenly updated to its then-current temperature of 139.8 C. Both thermistors then worked as normal for the remainder of the print, updating as needed. 

I went through the file that Cura generated and found the offending Gcode. I was able to reproduce the issue by issuing the same commands via the terminal.

```
M104 S200 ; set extruder temp
M140 S50 ; set bed temp
M190 S50 ; wait for bed temp
M109 S200 ; wait for extruder temp
```

It appears that Smoothie does not poll the hotend thermistor while it's waiting on `M190`, even though the board is actively heating it. Depending on how long it takes for `M190` to resolve, the extruder could experience thermal runaway and start a printer fire. Pardon my French, but that's a Serious Fucking Issue™ right there.

#### Branch & Commit or Version of Smoothieware

The firmware build I'm running is dated March 25 2017, from master. 

#### Hardware model you are using (Make, and version)

Azteeg X5 Mini v3

#### Host and Operating System running the Host

Octoprint, running on a Raspberry Pi 3

#### Link to and console logs

```
Send: N3835 M104 S200*90
Recv: ok
Send: N3836 M105*25
Recv: ok T:73.7 /200.0 @255 B:39.2 /0.0 @0
Send: N3837 M140 S50*111
Recv: ok
Send: N3838 M105*23
Recv: ok T:73.7 /200.0 @255 B:39.2 /50.0 @255
Send: N3839 M105*22
Recv: ok T:74.7 /200.0 @255 B:39.4 /50.0 @255
Send: N3840 M190 S50*98
Recv: B:39.9 /50.0 @255
Recv: B:40.1 /50.0 @255
Recv: B:40.4 /50.0 @255
Recv: B:40.8 /50.0 @255
Recv: B:41.1 /50.0 @255
Recv: B:41.4 /50.0 @255
Recv: B:41.7 /50.0 @255
Recv: B:42.1 /50.0 @255
Recv: B:42.5 /50.0 @255
Recv: B:42.7 /50.0 @255
Recv: B:43.2 /50.0 @255
Recv: B:43.5 /50.0 @255
Recv: B:43.9 /50.0 @255
Recv: B:44.3 /50.0 @255
Recv: B:44.7 /50.0 @255
Recv: B:45.1 /50.0 @255
Recv: B:45.4 /50.0 @255
Recv: B:45.7 /50.0 @255
Recv: B:46.0 /50.0 @255
Recv: B:46.4 /50.0 @255
Recv: B:46.7 /50.0 @255
Recv: B:47.0 /50.0 @255
Recv: B:47.3 /50.0 @255
Recv: B:47.7 /50.0 @255
Recv: B:48.0 /50.0 @255
Recv: B:48.3 /50.0 @255
Recv: B:48.6 /50.0 @77
Recv: B:48.8 /50.0 @255
Recv: B:49.1 /50.0 @186
Recv: B:49.3 /50.0 @0
Recv: B:49.6 /50.0 @255
Recv: B:49.8 /50.0 @112
Recv: B:49.9 /50.0 @255
Recv: ok
Send: N3841 M105*25
Recv: ok T:132.9 /200.0 @255 B:50.2 /50.0 @224
Send: N3842 M105*26
Recv: ok T:135.9 /200.0 @255 B:50.7 /50.0 @255
Send: N3843 M105*27
Recv: ok T:138.8 /200.0 @255 B:50.8 /50.0 @0
Send: N3844 M105*28
Recv: ok T:141.7 /200.0 @255 B:51.1 /50.0 @0
Send: N3845 M104 S0.000000*113
Recv: ok
Send: N3846 M105*30
Recv: ok T:144.7 /0.0 @0 B:51.2 /50.0 @226
Send: N3847 M140 S0.000000*115
Recv: ok
Send: N3848 M105*16
Recv: ok T:147.5 /0.0 @0 B:51.4 /0.0 @0
```
    
#### Link to running config and M503 results

* [M503](https://pastebin.com/PEUHcS9s)
* [Config](https://pastebin.com/Y7qLKnqz)

TheAppleFreak, 2017-05-28:
I'm not sure if there's a Gcode specification that defines how this behavior is supposed to work, so I'm not sure how feasible this would be to fix this. Perhaps Smoothie should temporarily set the target temperature of any other heaters on the system to their current temperatures, monitor and adjust the current temps while it also waits on `M190` (similar to how it normally behaves without `M190`), and when `M190` resolves it sets them back to their actual target temperatures. It wouldn't be a breaking change, but would hopefully prevent potential printer fires.

With that said, I'm also creating an issue over in the Cura repository to generate the heating commands in a different order, so hopefully it will no longer generate this race condition in the final 2.6 release.

wolfmanjm, 2017-05-28:
This is in fact a mixture of octoprint and smoothie, it is not a bug. While waiting for one temp it cannot poll the other temp. The temps however are both being maintained and controlled correctly.
This is simply that you have specifically used a wait for temp which means that NO OTHER gcodes will be executed while it waits. If you do not like this behavior then simply do not use the wait for temp.

closed 2017-05-28 by wolfmanjm

TheAppleFreak, 2017-05-28:
Fair enough. Hopefully the Cura team will fix this, then.

wolfmanjm, 2017-05-28:
Well I doubt it... it isn't really a bug in Cura either, it is perfectly correct, if you want that behavior, to set the temps then use the wait for temp before continuing
The issue is really that the graph cannot show both temperatures so one appears frozen, when in fact it is doing exactly what it has been told to do.
The thing is that the 'set temp and wait' has to ignore all further gcodes because that is what it is being asked to do. It has to wait for the temp to be reached before it can read anymore gcodes. At that point the regular M105 will return the current temps and the graph will then reflect both temps correctly.
It is imperfect but IMO is the only wait handle the wait for temp mcode. Personally I never use them, I heat up first then when I see it has reached temp I play the gcode file. It is also safer IMO to do that..

TheAppleFreak, 2017-05-28:
Ah, so it is keeping track of the temperature but just not reporting it to Octoprint? Alright, that makes a lot more sense. I suspected that might have been the case, but wasn't sure so I took the pessimistic approach to be safe.

Thank you for telling me this! I'll sleep easier tonight knowing what's actually happening :)