Smoothieware/Smoothieware #1197: Thermal runaway false alarm when changing bed to substantially lower temperature mid print

Zedsquared, 2017-06-13:
#### What were you doing?
Printing an object with an initial bed temperature of 120C that is then set to 85C after the first layer, this is an attempt to improve bed adhesion while reducing heat distortion on the lower layers during the print.

The Gcode was generated by slic3r, version 1.35.3-prusa3d-win64 and uses M190 S120 to set the initial bed temperature then M140 S85 after the end of layer 1 to set the new lower temperature

#### What did you expect to happen?

The print should have carried on to completion

#### What happened instead?

Shortly after starting layer 2 a temperature runaway event was flagged and smoothieware halted.
Error was:
       ERROR: Temperature runaway on B (delta temp 24.500885), HALT asserted, 

If using the temperature graphing on the new web interface it can be seen that the halt happened while the bed was still cooling to the new temperature. During the first layer there was only 0.4 degree variation, so PID looks good.

Changing only the initial M190 S120 to M190 S85 and retrying the print resulted in a completed print.
  
#### Branch & Commit or Version of Smoothieware

   Build version: upstreamedge-01cacfc, Build date: May 31 2017 16:35:23, MCU: LPC1768, System Clock: 100MHz
5 axis

#### Hardware model you are using (Make, and version)

    Smoothieboard 5X V1.0

#### Host and Operating System running the Host

   printing from internal  SD card via web interface

#### Link to and console logs

    
#### Link to running config and M503 results

  https://pastebin.com/hYfrWhHc

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

Zedsquared, 2017-06-13:
I realise that switching off thermal runaway would fix this and it's no showstopper for me.  But I thought I ought to point out what looks a lot like a misfeature in the runaway detection logic. Thanks!

wolfmanjm, 2017-06-13:
You have an old config and you need to set...

`runaway_cooling_timeout 60`

or to whatever time it takes to cool down from 120 to 85 (plus some leeway) in seconds.

closed 2017-06-13 by wolfmanjm

qharley, 2017-06-14:
In my opinion cool down should be treated differently.  If a heated bed can
reach its temperature in 60 seconds it must have considerable power. Making
the timeout 120 or 240 to compensate for cooling could cause real damage if
a real thermal runaway should occur.

This should be applied to hotends as well. I have 120s configured on my
machine hotends to compensate for Cooling but in a recent controlled safety
test one of my hotends reached almost 600deg C in that time. That is way
past the safe temperature for the PTFE lining

wolfmanjm, 2017-06-14:
@qharley If you looked at the code you would see that heating up and cooling down ARE treated differently! and they DO have different settings.

qharley, 2017-06-14:
It was my intention to do that when I found the time to do it...

Thanks for pointing out my ignorance in capital letters... ;-)

Zedsquared, 2017-06-14:
Ah, ok, thanks ... though, in my defense, the docs at http://forum.smoothieware.org/temperaturecontrol#toc24 have this setting grouped under "initial heatup runaway detection" rather than Out of range runaway detection and also imply that the timeout is disabled by default. So shouldn't my current config result in this timeout being disabled?

Zedsquared, 2017-06-14:
Perhaps "temperature set runaway detection" might be a better heading in the docs then "initial heatup runaway detection" as the latter had me thinking the timeout only applied to heating up initially. 

Wouldn't a check of delta t in a fixed time interval (say, one or two seconds) be a more robust check in these situations? Then it wouldn't matter what the difference between set temperatures was. The PID tuning parameters might even be useful as the source of a sensible delta t to test for too.