Smoothieware/Smoothieware #1198: temperature drops after print start with volcano e3d 40w resistor

se3d, 2017-06-15:
If you set bang bang or not, with a lot of configurations, including temperature_control.hotend.readings_per_second   20 or not, the hotend can't goes to 250º for example and printing pla with 210ºC, after print start, the pid initial configuration is P:15 I0.97 D100.

wolfmanjm, 2017-06-15:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2017-06-15:
This has nothing to do with smoothie issues. If you have a problem with your printer or with configuration check the wiki or get help on IRC or in the forums.

closed 2017-06-15 by wolfmanjm