Smoothieware/Smoothieware #1202: CNC=1 build needs updating

PeeJay, 2017-07-13:
Running make AXIS=6 CNC=1 results in a bunch of undefined references.

```
Linking ../LPC1768/main.elf
../LPC1768/./main.o: In function `init()':
C:\I\Smoothieware-edge\src/main.cpp:147: undefined reference to `ExtruderMaker::load_tools()'
C:\I\Smoothieware-edge\src/main.cpp:153: undefined reference to `TemperatureControlPool::load_tools()'
C:\I\Smoothieware-edge\src/main.cpp:182: undefined reference to `TemperatureSwitch::TemperatureSwitch()'
C:\I\Smoothieware-edge\src/main.cpp:188: undefined reference to `FilamentDetector::FilamentDetector()'
C:\I\Smoothieware-edge\src/main.cpp:232: undefined reference to `vtable for SCARAcal'
../LPC1768/./modules/utils/simpleshell/SimpleShell.o: In function `SimpleShell::print_thermistors_command(std::string, StreamOutput*)':
C:\I\Smoothieware-edge\src/modules/utils/simpleshell/SimpleShell.cpp:875: undefined reference to `Thermistor::print_predefined_thermistors(StreamOutput*)'
../LPC1768/./modules/utils/simpleshell/SimpleShell.o: In function `SimpleShell::calc_thermistor_command(std::string, StreamOutput*)':
C:\I\Smoothieware-edge\src/modules/utils/simpleshell/SimpleShell.cpp:896: undefined reference to `Thermistor::calculate_steinhart_hart_coefficients(float, float, float, float, float, float)'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [../LPC1768/main.elf] Error 1
make[2]: Leaving directory `C:/I/Smoothieware-edge/src'
make[1]: *** [lpc1768] Error 2
make[1]: Leaving directory `C:/I/Smoothieware-edge/src'
make: *** [all] Error 2
```

wolfmanjm, 2017-07-13:
It works perfectly well. Try doing a `make clean` first

closed 2017-07-13 by wolfmanjm

PeeJay, 2017-07-13:
I see. I had assumed a fresh install wouldn't need a make clean, but I guess it does! Thanks.