Smoothieware/Smoothieware #1203: Pause before probing?

DjDemonD, 2017-07-13:
Hi all,

I now market and sell an open-source piezo Z-probe. More details can be found online at reprap forum, just see the general forum main page, I won't leave any spam here.

What we are interested in is having a configurable pause before a probing dive takes place. RRF already has this as DC42 and T3P3 are making a nozzle contact probe based on piezo like we do, and the M558 command now has an R (recovery) parameter which is pause (in seconds) after positioning over a probing point before diving to probe. A value of 0.2sec is usually enough. It allows mechanical noise to settle before probing an reduces/eliminates false triggers from this very sensitive probe.

These things are becoming quite popular now as they are by far the most convenient and accurate z-probe available.

Would you consider implementing it for the benefit of the now hundreds of people using piezo z probes? I am not certain if your firmware already has such a switch that can be used or if the code already builds in a recovery before probing, but would be very grateful if you would consider it.

Kind regards
DjDemonD/Simon
Precisionpiezo.co.uk

wolfmanjm, 2017-07-13:
I have no plans to add it, but you are welcome to add it and submit a pull request.

DjDemonD, 2017-07-13:
Many thanks I'll put that put to some of the users. I am willing to send a free module as a thank you to anyone who can make a fork and add this feature.

wolfmanjm, 2017-07-17:
#1204 fixes this

closed 2017-07-17 by wolfmanjm