Smoothieware/Smoothieware #1207: Firmware display on LCD missing?

KirAsh4, 2017-08-13:
With the previous firmware, when I recompiled for a 4-axis system, the firmware is displayed on the LCD when it's turned on. With the latest one, after compiling I get a string that reads '>11myvar2' displayed. How/Where do I change that so it correctly displays the firmware version?

Compile platform, Win10 Pro, 64bit.
Grabbed latest github source about 30 minutes ago.

[image]

KirAsh4, 2017-08-13:
Looks like the git calls in 'generate-version.bat' file are failing. The machine I'm working on does have git installed, however I downloaded the source as a zip file and compiled from there. So I'm thinking it's failing due to it not being a valid git repository locally. Looking at the previous source I downloaded, it came with a .git folder, whereas this new one did not.

wolfmanjm, 2017-08-13:
That is correct you need a real github repo that has been pulled.

closed 2017-08-13 by wolfmanjm

KirAsh4, 2017-08-13:
This would be a new development then. As I stated, previously when I downloaded a Zip of the source, I was able to do a custom compile (for 4 axis) and get the correct information displayed. Now it doesn't work.