Smoothieware/Smoothieware #1212: Kill button does not reliably respond to very brief button presses

DavidJRobertson, 2017-09-01:
#### What were you doing?
Pressed the kill button very briefly.

#### What did you expect to happen?
The machine should enter the HALTED state immediately.

#### What happened instead?
The machine continued normal operation. It appears that the kill button input is only polled at 5Hz. Therefore, button presses shorter than 200ms are liable to be missed.

#### Branch & Commit or Version of Smoothieware
Edge, 8cb5324a598877904a3b49b4562a2fd5737c6cea

#### Hardware model you are using (Make, and version)
Smoothieboard 5x v1.0a, with kill switch wired to stop button on RRD GLCD.

#### Host and Operating System running the Host
Not applicable.

#### Link to and console logs
Not applicable.
    
#### Link to running config and M503 results
https://gist.github.com/DavidJRobertson/40e71db1843d67974d2d4c375d059a72

wolfmanjm, 2017-09-01:
Yes it is polled slowly to avoid bounces, it is a form of debounce.
I don't see this this as a bug or an issue.
You can increase the poll time in your own branch if you feel you need to change it.

closed 2017-09-01 by wolfmanjm

wolfmanjm, 2017-09-01:
Even better if you think it is important enough you can make the poll rate configurable, defaulting to 5.

DavidJRobertson, 2017-09-01:
I am happy with that approach, see pull request #1214.