Smoothieware/Smoothieware #1217: Switch module: failsafe_set_to option does not match documented fail_safe_set_to

DavidJRobertson, 2017-09-15:
#### What were you doing?
 Following the documentation at http://smoothieware.org/switch, I set the following config option:
```switch.hotendfan.fail_safe_set_to 1```

#### What did you expect to happen?
The switch module should turn on during the halt condition.

#### What happened instead?
The switch module turned off when the machine was halted. It appears as if the configuration option is completely ignored- which it is, as the actual config option has one fewer underscores than the documentation specifies.

https://github.com/Smoothieware/Smoothieware/blob/140a7bb517289dfdbbb1490d112e41c8dfdae7da/src/modules/tools/switch/Switch.cpp#L46

#### Branch & Commit or Version of Smoothieware
edge

wolfmanjm, 2017-09-16:
Then fix the wiki, it is user editable you know.

closed 2017-09-16 by wolfmanjm

arthurwolf, 2017-09-16:
Thanks for fixing the wiki !