Smoothieware/Smoothieware #1228: M110 does not follow expected behavior

matthova, 2017-10-14:
The only way to (re)set the line number is by sending the arguments backwards as follows:
```
N{New Line Position} M110
```

Relevant M110 spec: http://reprap.org/wiki/G-code#M110:_Set_Current_Line_Number
Note, Smoothieware is marked as supporting M110.
Appropriate examples of M110 are
```
M110; Implicit set current line to 0
N{Some Line Number} M110*{checksum value}; Implicit set current line to 0
M110 N{New Line Position}
N{Some Line Number} M110 N{New Line Position}*{checksum value}
```

Relevant code section:
https://github.com/Smoothieware/Smoothieware/blob/d79254323f4bb951426c6add29a4451130eaa018/src/modules/communication/GcodeDispatch.cpp#L91

I may be misunderstanding the implementation. Let me know if I'm off here. Would be open to helping to write/review the new implementation if a fix is deemed appropriate :)

wolfmanjm, 2017-10-14:
It does it this way as that is what Marlin and others do and it works with all known hosts. changing it will break hosts.

Note the reprap docs are notoriously inaccurate

closed 2017-10-14 by wolfmanjm

matthova, 2017-10-16:
Thanks for the explanation :)