Smoothieware/Smoothieware #1229: A Bug in the "MorganSCARASolution.cpp"

jtuomola, 2017-10-17:
**Cartesian_to_actuator function gives incorrect values, if arm1_length is not equal with arm2_length**

void MorganSCARASolution::cartesian_to_actuator(const float cartesian_mm[], ActuatorCoordinates &actuator_mm ) const
{

....
    if (this->arm1_length == this->arm2_length)
        SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - 2.0f * SQ(this->arm1_length)) / (2.0f * SQ(this->arm1_length));
    else
**# incorrect**

        SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - SQ(this->arm1_length) - SQ(this->arm2_length)) / (2.0f * this->SQ(arm1_length));

**# should be instead**
`       SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - SQ(this->arm1_length) - SQ(this->arm2_length)) / (2.0f * this->arm1_length * this->arm2_length);`
....

wolfmanjm, 2017-10-17:
comments @qharley ?

qharley, 2017-10-18:
Projects interfering.  Will take a look at it in a minute.

wolfmanjm, 2018-04-18:
As no more information has been provided I am closing this issue.

closed 2018-04-18 by wolfmanjm