Smoothieware/Smoothieware #1238: dual Z stepper + dual Z endstop

vladimirRUS007, 2017-12-01:
Hello, how to implement dual Z stepper + dual Z endstop on smoothieware

wolfmanjm, 2017-12-01:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2017-12-01:
FYI this is NOT the place to ask questions.

closed 2017-12-01 by wolfmanjm