Smoothieware/Smoothieware #1249: SD file system corruption fix

jasoroony, 2018-01-01:
diskio.cpp: writes the same sector over and over because it passes "sector" and not "s" that actually increments...  (I see quite a few closed issues about this, oh well.)

`DRESULT disk_write (
	BYTE drv,			/* Physical drive nmuber (0..) */
	const BYTE *buff,	/* Data to be written */
	DWORD sector,		/* Sector address (LBA) */
	BYTE count			/* Number of sectors to write (1..255) */
)
{
	FFSDEBUG("disk_write(sector %d, count %d) on drv [%d]\n", sector, count, drv);
	for(unsigned int s=sector; s<sector+count; s++) {
		FFSDEBUG(" disk_write(sector %d)\n", s);
		int res = FATFileSystem::_ffs[drv]->disk_write((char*)buff, sector); // <- this is wrong, use "s"
		if(res) {
			return RES_PARERR;
		}
		buff += 512;
	}
	return RES_OK;
}`

wolfmanjm, 2018-01-01:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

closed 2018-01-01 by wolfmanjm

wolfmanjm, 2018-01-01:
Thanks for catching this. I don't think it is the cause of corruption reported earlier, but more likely the cause of not being able to write blocks more than 512 bytes. We rarely write to the sdcard from smoothie, and when we do we currently hack it to write no more than 400 bytes at a time, so this bug does not currently show up. However fixing this will allow me to remove the 400 byte write limit that is scattered everywhere.

wolfmanjm, 2018-01-01:
Additionally there is only one place that it is called with the sector count > 1 which is why this is only seen on occasional long writes to disk, which very rarely happens anyway. So this is NOT the cause of disk corruption, we already know what that is and why, it is caused by the host writing via MSD which writes directly to the disk bypassing the fatlib, this causes the entire FAT to get corrupted.

jasoroony, 2018-01-02:
With all due respect.. And I appreciated what you were doing.. Well, sure, it doesn't show up until someone comes along and gets a corrupted file when they add some new feature, like I attempted, without an open issue to warn me about any required HACKS!  The closed issues talked about this particular problem (as well as the MSD one which I'm not concerned about).  I had also "upgraded" to the latest ChaNFS that does use >1 sector writes in other routines (directory clearing).  Mostly due to wondering why my files were corrupted.

I also didn't appreciate the automated email message I received about my original post (non-conformance & automatically closed issue).  That's just poor when you consider the time I spent on this.  I expect though, like some other issues that have been locked to contributors, you're not actually interested in (at least some of) these finer details.

As I wrote though, the work is still appreciated - and I'm happy to contribute when and if I see something glaringly obvious like this.  (It was 6 hours BTW - not really that long to work out a 5 year old problem.)

wolfmanjm, 2018-01-02:
If you spent so much time on it then how about a Pull request rather than a very vague pointer that took me ages to track down and actually find. The automatic message is because your Issue did NOT comply with the very specific requirements, I don't see what your problem is with not following guidelines. Those requirements are there to save us time, but it took a large chunk of my time to find this line in the code and fix it. Please do not complain about volunteers donating time to an open source project. it is just bad form.