Smoothieware/Smoothieware #1252: Firmware halts with 32 microsteps set

zen, 2018-01-05:
## How to file an issue

#### What were you doing?

Printing from SD card through webif. Benchy sliced with S3D or slic3r.
Printing from Proterface, both SD and local upload.
Printed with "dry run" scenario: no heat bed, no hotend enabled, extruder motor disconnected

#### What did you expect to happen?

Print to finish.

#### What happened instead?

Print halted, fans stopped, USB disconnected
LEDs: [1] [1] [0] [1] [0]

#### Branch & Commit or Version of Smoothieware

Build version: edge-e9942d1, Build date: Jan  1 2018 22:24:20, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)

Unfortunately MKS Sbase 1.3

#### Host and Operating System running the Host

Pronterface, webif, Mac OS X

#### Link to and console logs

What do you mean by console logs? Serial output from printer? 
    
#### Link to running config and M503 results

16 micro steps:
M503
https://pastebin.com/BQz1pwVR
Config:
https://github.com/zen/zHevo/blob/16us/config.txt

32 microsteps:
Config:
https://github.com/zen/zHevo/blob/master/config.txt

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2018-01-05:
This is almost certainly a H/W issue, contact MKS for support.

closed 2018-01-05 by wolfmanjm

zen, 2018-01-11:
Just a comment here: works fine with Marlin 2 branch. Many different speeds, both 16 and 32microsteps. I also sent request to MKS

OneDayFlie, 2018-01-14:
I have the same problems... 
Marlin works... So i don't think it's Hardware related...