Smoothieware/Smoothieware #1253: Cartesian G32 Grid Compensation hangs smoothie if head park position is outside bed

markshur, 2018-01-06:
Firmware: Edge Jan 1st 2018
Controller: Panucatt reArm with Biqu RAMPS 1.5
Hardware: custom all-aluminum i3-style, bed size 210x210
Endstops: X-min, Y-min, Z-min, ZProbe on Z-max

My X/Y endstops puts the extruder carriage off both edges of the printbed so I can home without worrying about accidently crashing the nozzle into the printbed.

alpha_min = -21
beta_min = -37.5
gamma_min = 0.2
move_to_origin_after_home = true
homing_order = XYZ
mechanical endstop switches
FSR z-probe (temporary / press fit over nozzle)
no limit switches

If I set leveling-strategy.rectangular-grid.do_home to true, sending a G32 command causes smoothie to physically home XYZ to [-21.0,-37.5,0.2] before immediately hanging while trying to move to origin before the grid probing starts.  Once it hangs a power cycle is required to restore operation.

If I set leveling-strategy.rectangular-grid.do_home to false, send a G28 command, send a G0 Z10 command to lift the printhead so I can attach the z-probe, then send the G32, the grid probing works normally.

wolfmanjm, 2018-01-06:
not enough information. if it crashes we need a crash dump. please read the guidelines. if you add the required i fo to this issue it will be reopened

closed 2018-01-06 by wolfmanjm

markshur, 2018-01-08:
I didn't say it crashed, I stated that it hanged - i.e. became unresponsive to user input.  The two are different.

Also, I certainly gave enough information for you to easily attempt to duplicate the issue - which would prove either that you could or could not reproduce the same behavior.  "Can't reproduce" is a totally different answer than "not enough information", especially when the asked-for crash dump requires hardware that a normal user doesn't have.  I'm not a novice user, but even so I had to order a USB/UART breakout cable *which has yet to arrive) because it isn't a part of my normal toolbox.

In any case, the reported behavior implies a race condition during the homing operation, since when the steps are done at human speeds the sequence operates normally, but when done at program speeds it fails.  Looking at zprobe::home() shows that the command is performed asynchronously. Adding a wait_for_idle() after the call_event() fixes the issue (verified in system), and does so with very little risk of "unintended consequences". This is a perfect example of a "low hanging fruit" bug fix.

```
// issue home command
void ZProbe::home()
{
    Gcode gc(THEKERNEL->is_grbl_mode() ? "G28.2" : "G28", &(StreamOutput::NullStream));
    THEKERNEL->call_event(ON_GCODE_RECEIVED, &gc);
    THECONVEYOR->wait_for_idle();
}
```

wolfmanjm, 2018-01-08:
Homing in Smoothie is not done asynchronously, it does not return until the homing cycle is complete.
There are plenty of wait_for_idle in the homing code if you look, so adding one more is not doing anything as the queue is already empty. This is NOT a bug fix. it simply hides whatever is going on in your setup. (If you were to look further in the homing code you would see the last thing it does is a wait_for_idle()). 

How do you know it is not crashing? smoothie rarely hangs unless it actually crashes. Hang is very generic and offers no clue as to what is actually happening. I suspect it is not completing the homing cycle, which would make it appear as if it "had hung" as it won't accept any more commands until it is complete. It may be to do with the got to origin after home. Maybe it can't get to the origin?

So I reiterate there is not enough information here to debug, if indeed there is anything to debug.