Smoothieware/Smoothieware #1255: Bug in the progress command

obujor, 2018-01-10:
Hi,

I've noticed a bug in the _progress_ command when playing large gcode files (more than 2 millions of lines) from the sd on a Smoothieboard 4XC v1.1 controlling my laser diode engraver.

From 0% to 70% the progress was working great, after 70% instead of 71% I'm getting 1%, 2% and so on.
This is the output of _version_ command:

> Build version: edge-4f622ed, Build date: Dec 22 2017 21:38:05, MCU: LPC1769, System Clock: 120MHz  CNC Build 3 axis

I'm sending commands to the board from the _imrahil-ui_.

This's a list of progress outputs (Notice that _progress -b_ command is working properly):

> file: /sd/memi-15-4000.gcode, 0 % complete, elapsed time: 00:02:27, est time: 04:41:53
> file: /sd/memi-15-4000.gcode, 2 % complete, elapsed time: 00:04:42, est time: 03:29:09
> SD printing byte 1363816/58961451
> file: /sd/memi-15-4000.gcode, 3 % complete, elapsed time: 00:06:08, est time: 03:13:19
> SD printing byte 2287582/58961451
> file: /sd/memi-15-4000.gcode, 6 % complete, elapsed time: 00:10:55, est time: 02:49:21
> file: /sd/memi-15-4000.gcode, 6 % complete, elapsed time: 00:11:51, est time: 02:47:48
> file: /sd/memi-15-4000.gcode, 9 % complete, elapsed time: 00:16:08, est time: 02:37:58
> file: /sd/memi-15-4000.gcode, 12 % complete, elapsed time: 00:20:41, est time: 02:28:46
> SD printing byte 7313401/58961451
> file: /sd/memi-15-4000.gcode, 12 % complete, elapsed time: 00:21:51, est time: 02:26:21
> file: /sd/memi-15-4000.gcode, 13 % complete, elapsed time: 00:23:09, est time: 02:24:05
> file: /sd/memi-15-4000.gcode, 18 % complete, elapsed time: 00:30:06, est time: 02:13:35
> file: /sd/memi-15-4000.gcode, 21 % complete, elapsed time: 00:34:11, est time: 02:08:01
> file: /sd/memi-15-4000.gcode, 22 % complete, elapsed time: 00:36:00, est time: 02:05:51
> file: /sd/memi-15-4000.gcode, 24 % complete, elapsed time: 00:38:50, est time: 02:02:39
> file: /sd/memi-15-4000.gcode, 25 % complete, elapsed time: 00:41:07, est time: 02:00:00
> SD printing byte 15085174/58961451
> file: /sd/memi-15-4000.gcode, 32 % complete, elapsed time: 00:51:30, est time: 01:46:45
> SD printing byte 19224035/58961451
> file: /sd/memi-15-4000.gcode, 33 % complete, elapsed time: 00:53:46, est time: 01:44:23
> file: /sd/memi-15-4000.gcode, 34 % complete, elapsed time: 00:54:44, est time: 01:43:17
> file: /sd/memi-15-4000.gcode, 36 % complete, elapsed time: 00:57:46, est time: 01:40:11
> file: /sd/memi-15-4000.gcode, 43 % complete, elapsed time: 01:08:26, est time: 01:29:23
> SD printing byte 25594472/58961451
> file: /sd/memi-15-4000.gcode, 46 % complete, elapsed time: 01:12:45, est time: 01:24:39
> file: /sd/memi-15-4000.gcode, 49 % complete, elapsed time: 01:17:15, est time: 01:19:57
> SD printing byte 28989646/58961451
> file: /sd/memi-15-4000.gcode, 53 % complete, elapsed time: 01:23:25, est time: 01:13:15
> SD printing byte 31422505/58961451
> file: /sd/memi-15-4000.gcode, 59 % complete, elapsed time: 01:32:58, est time: 01:02:36
> SD printing byte 35259058/58961451
> file: /sd/memi-15-4000.gcode, 62 % complete, elapsed time: 01:36:39, est time: 00:58:31
> SD printing byte 36742716/58961451
> file: /sd/memi-15-4000.gcode, 66 % complete, elapsed time: 01:42:42, est time: 00:51:52
> file: /sd/memi-15-4000.gcode, 67 % complete, elapsed time: 01:44:42, est time: 00:49:39
> SD printing byte 40017899/58961451
> file: /sd/memi-15-4000.gcode, 70 % complete, elapsed time: 01:48:50, est time: 00:44:56
> file: /sd/memi-15-4000.gcode, 1 % complete, elapsed time: 01:54:03, est time: 00:39:08
> SD printing byte 43919074/58961451
> SD printing byte 43944035/58961451
> SD printing byte 44042744/58961451
> file: /sd/memi-15-4000.gcode, 2 % complete, elapsed time: 01:54:42, est time: 00:38:26
> SD printing byte 48295898/58961451
> file: /sd/memi-15-4000.gcode, 16 % complete, elapsed time: 02:15:27, est time: 00:15:41
> file: /sd/memi-15-4000.gcode, 21 % complete, elapsed time: 02:22:22, est time: 00:08:02

Thanks!

wolfmanjm, 2018-01-10:
Yes this is a known bug, it is due to overflow of the 32bit integers. I am not quite sure the best way to fix it or when. But at least there is an issue on it now :)

wolfmanjm, 2018-01-11:
Can you test pull request #1256

obujor, 2018-01-11:
Thanks, I'll test it later and I'll let you know.

obujor, 2018-01-11:
It works! Thank you!
This is the tested version:
`Build version: upstreamedge-80b0798, Build date: Jan 11 2018 16:16:26, MCU: LPC1769, System Clock: 120MHz CNC Build 3 axis
`
Here is a log of progress outputs:
```
file: /sd/memi-15-6000.gcode, 1 % complete, elapsed time: 00:02:22, est time: 04:08:53
SD printing byte 693703/58961451
file: /sd/memi-15-6000.gcode, 10 % complete, elapsed time: 00:14:10, est time: 02:07:08
SD printing byte 5934283/58961451
file: /sd/memi-15-6000.gcode, 22 % complete, elapsed time: 00:29:22, est time: 01:42:24
SD printing byte 13163261/58961451
file: /sd/memi-15-6000.gcode, 28 % complete, elapsed time: 00:36:25, est time: 01:34:01
SD printing byte 16483734/58961451
file: /sd/memi-15-6000.gcode, 49 % complete, elapsed time: 01:02:42, est time: 01:06:28
SD printing byte 28678755/58961451
file: /sd/memi-15-6000.gcode, 54 % complete, elapsed time: 01:09:18, est time: 00:59:42
SD printing byte 31702776/58961451
file: /sd/memi-15-6000.gcode, 58 % complete, elapsed time: 01:14:21, est time: 00:54:15
SD printing byte 34154111/58961451
file: /sd/memi-15-6000.gcode, 68 % complete, elapsed time: 01:26:46, est time: 00:40:34
SD printing byte 40204334/58961451
file: /sd/memi-15-6000.gcode, 79 % complete, elapsed time: 01:40:01, est time: 00:25:58
SD printing byte 46827547/58961451
file: /sd/memi-15-6000.gcode, 97 % complete, elapsed time: 01:59:47, est time: 00:04:16
SD printing byte 56963053/58961451
```

@wolfmanjm can I keep this firmware or it's to unstable?

closed 2018-01-11 by obujor

wolfmanjm, 2018-01-11:
It is now merged in edge