Smoothieware/Smoothieware #1257: Endstop `retract` not happening sporadically after home

andySigler, 2018-01-11:
I'm not finding any currently open or closed issue related, so I apologize in advance if this is a duplicate.

I am experiencing what seems to be a bug, about every 1 out of 20 homing sequences, but I can't predict when it will happen.

The unwanted behavior is that the axis which is homed does **not** retract the distance specified in the config. While 19/20 times it will retract 3mm, but then 1/20 times it will not retract, and stay pressing the home switch. If I leave the `limit_enable` set to `true`, then this will cause an error when running any following move.

To test, I update the firmware to latest `edge` commit (`0acf2ad`), with my config file ([sharing my config here on gist](https://gist.github.com/andySigler/395225fd1f9f81964075f25739c5f2ad)) which sets the `X` axis to retract `3` millimeters.

I run the following:
```
--> G28.2X
<-- ok
<-- ok
--> M114.2
<-- ok MCS: X:392.0000 Y:0.0000 Z:0.0000 A:0.0000 B:0.0000 C:0.0000
<-- ok
```

and I get `392` for `X` axis, which is correct (the homing position is `395`, and then retracting `3` millimeters puts it at `392`.

**However**, 1 out of every 20 times I get this:
```
--> G28.2X
<-- ok
<-- ok
--> M114.2
<-- ok MCS: X:395.0000 Y:0.0000 Z:0.0000 A:0.0000 B:0.0000 C:0.0000
<-- ok
```
and `X` is at `395`, because it did not retract, and I can see that the switch is still being pressed.

If I then try to run a move command, I get the expected limit-switch error:
```
--> G91 G0X-20
<-- ok
<-- ok
<-- ALARM: Hard limit -X
```

wolfmanjm, 2018-01-11:
This is usually caused by bounce on the endstop and/or noise on the endstops. It is NOT reproducible, and is NOT a bug.

I suggest you increase the debounce, and/or isolate your endstops from noise. Using normally closed usually reduces noise vs normally open.

closed 2018-01-11 by wolfmanjm

andySigler, 2018-01-11:
Fixed! My intuition told me that a noisy endstop would only cause early triggers, not prevent retractions, but I also didn't carefully read through what was happening in the code, so thank you very much for your fast answer