Smoothieware/Smoothieware #1258: "Home All Axes" should home ($H) when GRBL Mode is on

nuess0r, 2018-01-12:
#### What were you doing?
I'm upgrading a K40 laser cutter to smoothie. I also installed a reprap discount style panel.

To be able to use bCNC I set "grbl_mode true". 

From the panel menu I started a homing cycle ("Prepare -> Home All Axes") but the machine didn't move.

After longer confusion I read about the difference in interpreting the G28 command when GRBL mode is enabled.

#### What did you expect to happen?

That Smoothie starts a homing cycle when I select "Home All Axes"

This is command "G28" in RepRap mode (but "$H" is understood also in this mode)
When GRBL mode is on, the according command is "$H".

#### What happened instead?

Nothing.

The machine sinks it's at position 0,0 so it doesn't need to move to position 0,0.

When I jogged first and then selected "Home All Axes" then it moved back to the original position and was not homing.

Makes total sense now I know the difference of G28 and $H in GRBL mode but was totally confusing until then.

#### Branch & Commit or Version of Smoothieware
Build version: edge-e9942d1, Build date: Jan  1 2018 22:24:20, MCU: LPC1768, System Clock: 100MHz 5 axis

#### Hardware model you are using (Make, and version)

#### Host and Operating System running the Host

bCNC, Pronterface, Debian SID AMD64

#### Link to and console logs
    
#### Link to running config and M503 results

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Yes.

wolfmanjm, 2018-01-12:
This is not reproducible. $H will home all axis on all machines I have tested. I suspect you have an error in your config or H/W.
Also as it appears you have an MKS SBase I suggest you contact MKS for support.

closed 2018-01-12 by wolfmanjm

nuess0r, 2018-01-12:
Yes, $H homes on all machines.

But the menu entry "Home All Axes" on a RepRap style LCD executes a G28. Instead it should do a normal homing cycle and therefor the panel should execute a $H.

Btw: Yes I have a MKS board. The first thing I did was donating 20 US$ to your great project as a little compensation.

I understand your opinion about this and the uncountable hours of customer support this gives you. 
I built open hardware projects my self and tried to introduce them to the community (which needs always luck). In my opinion Smoothie has won because of the clone boards. No one is talking about the other 32bit 3D printer boards. Everyone is talking about Smoothieware.

I know the people from Pixhawk (PX4), an open hardware controller for drones. First a university project, then copied by the chinese, now officially working together with Broadcom as the standard firmware for there drone solutions.

wolfmanjm, 2018-01-12:
Ahh ok it was unclear this was a panel issue. I thought I had fixed that a while back, if not I'll fix it now. Thanks

nuess0r, 2018-01-12:
Thanks for your answers.

I tried the CNC build right from the beginning (Because in the wiki it is noted that it has a nicer panel for non-3D printer machines).
I used the CNC version from 22. Dec 17

Took me some hours to figure out, that it seems that the laser module is excluded in this CNC build and therefor the board does not boot due to unknown module that I was trying to enable in the config.

wolfmanjm, 2018-01-12:
try the firmware-latest.bin it includes the laser module

arthurwolf, 2018-01-12:
I'm pretty sure there is laser in cnc, and if enabling laser crashes, that
wouldn't happen if laser wasn't there, so it's there and crashing possibly
due to an error config. Can we see your config file ?

nuess0r, 2018-01-12:
Now with my working configuration I tried again with the CNC build and it works!

Yes, the CNC build includes the laser module. 
Yes, the CNC build does home when I select "Home All Axes" from the panel

Here are my configuration files:
https://gist.github.com/nuess0r/1e19fe7ac8723345e31a096a602a407b

Yesterday I rebuilt my configuration from scratch (downloaded the current default) because I had very strange stepper movements when jogging (sometimes fast as it should, sometimes slow or extremly slow).  I tried the default one, and it worked as expected. Then I changed one parameter at a time to find my bug. In the end, I had a fully working configuration and I don't know the difference.

(Yes, I know that I should not use the external SD Card. I try with short cables and when I have doubts I disable it again).

Thanks for your help