Smoothieware/Smoothieware #1259: Enhancement: G2/G3 should support the Radius Format

nuess0r, 2018-01-12:
#### What were you doing?
Playing around with bCNC and a laser cutter.

In bCNC there is a CAM plugin to generate jigsaws and I wanted to try that on my machine.

Example code:
```
g0 x10 y0
m4
f1200
m4
g2 x9.941176 y3.921569 r8.150827
g3 x9.235294 y5.490196 r2.716942
g2 x7.823529 y4.941176 r1.766013
g2 x7.823529 y8.392157 r1.698089
g2 x9.235294 y7.843137 r1.766013
g3 x9.941176 y9.411765 r2.716942
g2 x10 y13.333333 r8.150827
g3 x9.97906 y17.218596 r8.184513
g2 x10.83301 y18.800594 r2.675149
g3 x12.233143 y18.358325 r1.740505
g3 x12.142308 y21.613801 r1.71772
g3 x10.725415 y21.087819 r1.762637
g2 x10.063096 y22.601302 r2.693963
g3 x10 y26.666667 r8.150827
g3 x10.128359 y30.529112 r8.170386
g2 x10.880742 y32.203865 r2.732872
g3 x12.237911 y31.498861 r1.747096
g3 x12.225349 y34.987229 r1.696604
g3 x10.868607 y34.535223 r1.748246
g2 x10.020347 y36.159376 r2.739191
g3 x10 y40 r8.150827
m5
g0 x20 y40
```
#### What did you expect to happen?

That a nice jigsaw is cut out.

#### What happened instead?

Only the G0 moves and the G1 moves for the outline where executed. All the G2/G3 lines where skipped.

#### Branch & Commit or Version of Smoothieware
Build version: edge-e9942d1, Build date: Jan  1 2018 22:24:20, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)

#### Host and Operating System running the Host

bCNC github head version, linux x64

#### Link to and console logs

    
#### Link to running config and M503 results

wolfmanjm, 2018-01-12:
If you want to add this feature feel free to supply a pull request. Otherwise it is likely to not get done.

cilynx, 2018-02-17:
@wolfmanjm -- what do you think of this approach?  I figured it made more sense to find the center offset from the radius and use the pre-existing arc computation rather than reinventing the wheel as it were.

@nuess0r -- can you build off of my branch linked above and see if it meets your needs?  Right now, it only works in the XY plane, but that looks like your use case.

wolfmanjm, 2018-02-17:
any implementation needs to fully support the spec and any errors...
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3

As it states there R is not recommended due to the precision errors, so it is not totally a bad thing it is not currently supported.

cilynx, 2018-02-17:
Agreed that this needs to be cleaner and more robust before inclusion, thus the commentary here instead of sending a PR right away.

For the time being, I'm going to wait and see if @nuess0r or any other users comment on this thread.  I have no need for this functionality myself -- just doing this for fun and to give back a bit.  If folks come back saying this is something they really want, I'll flesh it out and send a PR.

nuess0r, 2018-02-20:
Hi cilynx

Thanks for your work. I was thinking myself of a quick (and dirty) solution like yours and dug a bit into the code.

I have to set up first the toolchain to build your version but hopefully I can give it a try today evening. My laser cutter is also ready again, I finished yesterday the cooling circuit based on gaming-PC water cooling parts. Works great!

nuess0r, 2018-02-20:
Hi cilynx

Thanks for your work. I was thinking myself of a quick (and dirty) solution like yours and dug a bit into the code.

I have to set up first the toolchain to build your version but hopefully I can give it a try today evening. My laser cutter is also ready again, I finished yesterday the cooling system based on gaming-PC water cooling parts. Works great!

wolfmanjm, 2018-02-20:
FWIW here is how GRBL does it, I suggest you use that code.
https://github.com/gnea/grbl/blob/master/grbl/gcode.c#L696

nuess0r, 2018-02-20:
It works nicely. Did two puzzles today.

cilynx, 2018-02-21:
Right on. I’ll check out the GRBL implementation and see about putting together a robust implementation here that Jim can pull into mainline edge.

wolfmanjm, 2019-03-19:
This seems like a dead issue being over a year old. I'd be happy to reopen if someone is going to work on it.

closed 2019-03-19 by wolfmanjm