Smoothieware/Smoothieware #1260: current_control mapping bug?

kantlivelong, 2018-01-17:
#### What were you doing?

Adjusting current_control settings for each motor.

#### What did you expect to happen?

Smoothieware to set the appropriate current per motor.

#### What happened instead?

It appears that current_control is setting the current for the wrong motor but everything else is working as expected ( e.g *_steps_per_mm, *_max_rate, *_[step|dir|en]_pin ).

(X)alpha    -> left Z
(Y)beta     -> right Z
(Z)gamma    -> left XY motor
(A)delta    -> right XY motor
(B)epsilon  -> extruder

As you can see in the config I have the following defined the following pinouts:
gamma = M1 (Primary Z)
epsilon = M2 (Secondary Z *Current Only*)
alpha = M3 (X/Y Left Motor)
beta = M4 (X/Y Right Motor)
delta = M5 (Extruder)

#### Branch & Commit or Version of Smoothieware

Upstream SHA1: 318863377daacedc97229e667ce7fee0ded4bcbf
Version: edge-8e4a64c

#### Hardware model you are using (Make, and version)

SBC-5xM v1.0b

#### Host and Operating System running the Host

OctoPrint 1.3.6

#### Link to and console logs

N/A
    
#### Link to running config and M503 results

[config](https://pastebin.com/g9eJuLx7)
[M503](https://pastebin.com/7f3KyH7q)

kantlivelong, 2018-01-17:
After looking at the list again it seems as if Smoothieware is hard-coding:

alpha = M1
beta = M2
gamma = M3
delta = M4
epsilon = M5

triffid, 2018-01-17:
The step/dir/en control pins and the current control channels are completely independent.

Smoothie will not and cannot automagically remap current control channels if you have altered the default pin mappings since it has _no built-in awareness of how the circuit board is laid out_ - which is arguably a very good thing for anyone running smoothieware on something other than the smoothieboard.

Instead, the "alpha" current control channel is hard-wired to the onboard M1 driver, regardless of what you're using that driver for or any step/dir/en pin remappings you have performed in your config.

This happens at the PCB layout level and has nothing to do with firmware. Feel free to reroute your board if it bugs you.

This is Not A Bug, as there could conceivably be a board in existence which is wired as per your config, and trying to "fix" this would really mess with that board.

If *you* choose to remap the default pins or otherwise operate far outside the normal configuration variation, *you* are responsible for understanding and dealing with the side effects that that may bring. Be happy that smoothie lets you choose, rather than getting in your way ;)

kantlivelong, 2018-01-17:
Makes sense. Is this in any docs out there that I could have missed? Are there any other settings that behave like this?

sbts, 2018-01-17:
Seems there should be comments in the same place as the settings them selves clarifying this

kantlivelong, 2018-01-17:
or would an option like  `alpha_current_channel 1` be feasible?

closed 2018-01-17 by wolfmanjm

kantlivelong, 2018-01-17:
I fail to see how this is not an issue at all. What is the point of allowing pin definition per axis but having current control hard coded? Are you saying PR wouldn't be accepted either?

wolfmanjm, 2018-01-17:
I think the point is being misunderstood here, nothing is being hard coded at all. it is totally configurable via both the config and the M commands. 
You just need to understand the terminology. alpha_current is always the M1 driver, beta_current is always the M2 driver etc. There is no bug, and there is nothing to be fixed here. @triffid explained it perfectly well.

kantlivelong, 2018-01-17:
Seems so. Based on the documentation I was under the impression that each could be defined as the user saw fit ( e.g alpha(X)=M4 ). Being that I am able to do it for everything other than current would it not make sense to make it configurable? I would be willing to make a PR but don't want to go through the effort if it will just be rejected.

sbts, 2018-01-17:
@wolfmanjm I was just looking around to get a handle on how smoothie assigns functionality to M* outputs, especially in the case of a corexy which @kantlivelong is running

I found [corexy](http://smoothieware.org/hbot) which explains the 3 options that should be set to make the printer a corexy.
However, I can't find info on M* => stepper function mapping (for corexy)
Is someone able to point me at appropriate docs for that

triffid, 2018-01-18:
@sbts please don't hijack issue-specific threads for general unrelated questions. There are other channels (G+, mailing list, IRC) for that type of discussion.

@kantlivelong the GPIOs can be mapped to match the current control, if you un-map the GPIO from the current control, that's on you.

Feel free to implement reconfigurable current control channels, but since you're literally the only person we've heard from who wants this feature, it will likely remain in your personal branch as I suspect @wolfmanjm won't merge it.

I'm curious though, *why* have you remapped your stepper driver GPIOs in such a fashion?

kantlivelong, 2018-01-18:
@triffid It started out as thinking I would need more than 1A per Z motor so I went the route of setting up a slave stepper. In my mind it made sense to wire them close to each-other (M1 and M2 in my case) instead of doing M3 & M5. When I was testing them in this configuration I was using current control and the values seemed to justify the need as it took nearly 2A to drive each motor but in reality I wasn't changing the right driver.

I've now since rewired to meet the standard config and am able to drive both Z motors with with 2A and a Y splitter.