Smoothieware/Smoothieware #1264: Max Limit switches causing crash?

FabCreator, 2018-02-01:
#### What were you doing?

During a job, smoothie will disconnect from the host, and move into a hold state.
This occurs randomly and getting it to trigger reliably during a job has not possible.
However with some experimenting I believe there may be an issue with the Max Limit switches. 
I can replicate the fault by pressing a max end stop during the job. 

#### What did you expect to happen?
Machine to halt

#### What happened instead?
First time it may act as it should machine halts. second or third time a mass of disconnect errors followed by Com3 corrupting the USB driver and smoothie locking up with all leds solid green.

As triggering a Max end stop will cause the smoothie to freeze up.
More information already posted on the laser web github, I thought perhaps it was a coms issue to being with but same error can be reproduced with pronterface, on both of our laser cutters.
https://github.com/LaserWeb/LaserWeb4/issues/455 

#### Branch & Commit or Version of Smoothieware

 Firmware is smoothie edge-4f622ed

#### Hardware model you are using (Make, and version)

    Smoothieboard 5XC 1.1

#### Host and Operating System running the Host

   Pipo 9x Windows 10 x86
#### Link to and console logs

    [On gist.github.com or pastebin.com. Always include and never truncate.]
    
#### Link to running config and M503 results

    [On gist.github.com or pastebin.com. Always include and never truncate.]
[config.txt](https://github.com/Smoothieware/Smoothieware/files/1685987/config.txt)

Currently I have disabled the abort when limit switch is pressed and so far i have not had a crash.

M503 after disabling the abort when limit pressed.
 M503
 ; No config override
 ;Steps per unit:
 M92 X266.66599 Y266.66599 Z1280.00000
 ;Acceleration mm/sec^2:
 M204 S1500.00000 X1500.00000 Y250.00000 Z50.00000
 ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
 M205 X0.10000 Z-1.00000 S0.00000
 ;Max cartesian feedrates in mm/sec:
 M203 X300.00000 Y50.00000 Z10.00000
 ;Max actuator feedrates in mm/sec:
 M203.1 X300.00000 Y50.00000 Z10.00000
 ;WCS settings
 G54
 G10 L2 P1 X20.000000 Y160.000000 Z0.000000 ; G54
 ;Digipot Motor currents:
 M907 X1.80000 Y1.80000 Z1.50000 A0.80000
 ;Home offset (mm):
 M206 X0.00 Y0.00 Z0.00

All the best,
Bonne

wolfmanjm, 2018-02-01:
If you have limits enabled and you trigger one it puts smoothie into a ALARM (or HALT) state it does not crash. you can release this with M999. This is documented in the wiki. If you get random triggers then you have noise on your endstops.

This is NOT a bug.

wolfmanjm, 2018-02-01:
It sounds a lot more like LW is not handling the ALARM condition correctly. This would be a LW bug not a smoothie issue. 
Try with a console or bCNC where it works perfectly well. This has been working for years, and any bugs would have been noticed years ago.

closed 2018-02-01 by wolfmanjm

wolfmanjm, 2018-02-01:
You also need to be in GRBL mode which you appear to NOT be in. Which may explain why LW is not working properly. Try using the firmware-cnc-latest.bin which is in grbl mode by default.

newt80ns, 2018-02-01:
I agree on noisy signal, but the coms lockup and board freeze are nothing, to be expected, and not worth a look?

wolfmanjm, 2018-02-01:
@newt80ns the board does not freeze up or lose comms I have been using limits for years. This appears to be a host issue not a smoothie issue.

FabCreator, 2018-02-01:
@wolfmanjm  Perhaps i did not explain very well,
A normal response of the end stop being pressed should be to go into a halt state, as you said.

This does happen every time with the Min end stops. However with the Max end stops when pressed during a job it crashes.
By crashes I mean does not enter a halt mode but instead disconnects from the host, in the process corrupting the com port.

Smoothie is now non responsive. all LEDs are solid. I can reset the PC/repair the driver but i cannot reconnect to the smoothie unless i power cycle it. As i said this is also true for pronterface.

I had been using the lasted CNC bin, but reverted to the older one to see if this would remove the issue. 

[Smoothie board error.txt](https://github.com/Smoothieware/Smoothieware/files/1686098/Smoothie.board.error.txt)

FabCreator, 2018-02-01:
As for the random triggers, they are not random triggers of the endstops, This end stop issue is the only way i can replicate it reliably currently.

When it disconnects / alarms randomly i get the following error below.

[image]

If i reconnect when this error occurs smoothie appears to have moved into a hold state, which I cannot exit. 

[image]

wolfmanjm, 2018-02-01:
These are problems in your host not smoothie. As I said try using bCNC  it works well with that.

LW obviously does not handle ALARM mode well and/or you are not in grbl mode as I said before.

wolfmanjm, 2018-02-01:
If you believe this to be a crash in smoothie then please provide a crash dump as explained in the wiki, as I cannot reproduce any crashes when a hard limit switch is activated, it does exactly what I expect it to do, goes into ALARM mode which you can clear with $X or M999.

wolfmanjm, 2018-02-01:
Why are you continuing to send gcode to the system that is in ALARM state? that seems to be what is happening.

Please provide a valid crash dump, if it is indeed crashing in smoothie.

wolfmanjm, 2018-02-01:
FYI the code for min limits is identical for the max limits. if one works the other must work. unless you have a config error.

FabCreator, 2018-02-01:
I will try other hosts, Like i said the same error would occur using pronterface.
It only occurs when running a job. I can trigger the end stop while jogging with no errors.

$X and M999 do work in a normal abort. but in this case i cannot reconnect to the smoothie. 
Laser web shows a hold instead of abort, I cannot clear it with ~ 

I will read up on how to produce a crash report. 

With the end stops hard limits disabled I have yet to run into a random disconnect/pause. It may well be a laserweb issue but when the same would occur with pronterface i thought i would open an issue here and attack the problem from both sides. 

Which host are you currently testing with bCNC?

wolfmanjm, 2018-02-01:
pronterface will  not work with  grbl_mode so you need to turn off grbl-mode to test pronterface.

the log you show has it continuing to send after it halts, and the halt seems correct after Max X limit is triggered (no crash) the "crash" appears after your host continues to ignore the ALARM/HALT state and send more gcodes. This should not really crash smoothie, but it is also incorrect behavior of the host. Pronterface has been tested (in 3d mode) with limits and it stops sending as soon as it gets the ALARM.
the Latest bCNC also seems to work correctly.
Try running the job from sdcard, then trigger the max endstop, see if it still "crashes".
If the host continues to ignore the HALT state then it could trigger some odd issue in smoothie I am not aware of.
Also the HOLD state must be a LW thing as that can only be entered with a !, so I remember LW had a bug where it could not get out of HOLD when it was also in ALARM. (it is in the issues). We do not have good cooperation with LW developers so it is sometimes hard to get them to fix smoothie incompatibilities.

I recommend you create the gcode file with LW but do not use LW as a host to smoothie just copy the gcode to sdcard and play it, or send the gcode from bCNC. (which is what I do).

Without a crash dump I cannot tell you if this is a crash or not. and my testing works fine.

wolfmanjm, 2018-02-01:
Also after clearing an Alarm you need to either reset or at least home, which you appear to not be doing.

FabCreator, 2018-02-01:
Thanks, I will try what you have suggested, 
After an alarm we do normally home. before restarting the job, however when jogging etc you can hit the end stops accidentally and then I do not always home as i am still jogging around to find the origin for my work piece. 

What do you mean by reset after an alarm? should you be resting the smoothie board?

wolfmanjm, 2020-05-10:
This may have well been a bug that has now been fixed in PR #1464 It was harder to see it here as LW was hiding the issue by continuing to ignore the HALT state.