Smoothieware/Smoothieware #1268: Check preset current

Magic-Pi, 2018-02-14:
**What were you doing?**
I want to change to preset current for the extruder stepping motor. The default value of my printer is 1.5 A but I bought a new extruder and have to update the setting. I checked the firmware and found the delta_current parameter. I changed the parameter to a new value of 0.1 A.

**What did you expect to happen?**
Perfect run of the new stepper motor.

**What happened instead?**
The new motor still getting warm/hot during operation (that's ok) but getting very hot during the dwell time.

**Branch & Commit or Version of Smoothieware**

**Hardware model you are using (Make, and version)**
MKS-Base1.3 with DRV8825 driver. New stepper motor is an unipolar 6-pin motor.

**Host and Operating System running the Host**
Windows 10, Repetier

wolfmanjm, 2018-02-14:
try checking the override settings. M503. Delte the override settings, or set with the M code for current.

Try reading the wiki to understand how config and overrides work.

closed 2018-02-14 by wolfmanjm

wolfmanjm, 2018-02-14:
Also please get support from MKS, we do not support MKS products. AFAIK MKS does not have S/W settable currents.

pixel3design-hub, 2018-02-15:
AFAIK smoothieware firmware is opensource and can be used in other boards than smoothieboards.
The answer is legit, because it's a firmware config and YES, MKS-SBASE is a smoothieboard knockoff with software configurable drivers currents.

I think he asked for the holding current, not the working current for the stepper.
DRV8825 is NOT an SPI programmable driver like the Trinamic TMC2130/26xx or the DRV8711 and the holding current is identical to the working current.
If you want to TURN OFF the extruder motor during inactivities, you can do "M18 E0" and disable the driver until you print.