Smoothieware/Smoothieware #1271: Do not interprete an empty newline over serial as an empty command.

slundell, 2018-02-21:
Smoothieware and Cura 3.x are incompatible when printing over USB/Serial (streaming G-CODE). The printer hangs mid-print, but seems to be alive, but not responding to new commands. e.g. https://github.com/Ultimaker/Cura/issues/1777

This seems to be due to that Cura sends a newline  *before and after* each G-CODE. Smoothieware treats the first newline as an empty command, happily answering  `ok\r\n`. Cura takes this as a confirmation that the actual G-CODE has been accepted, and sends an new `\nGCODE\n`, while the old G-CODE is still in the hosts in-buffer. This causes buffer overflow and/or communication congestion.

The bug has been reported to Cura as well, as I do not know what is the correct behaviour. https://github.com/Ultimaker/Cura/pull/3346

wolfmanjm, 2018-02-21:
This is required behavior for smoothie, ALL lnes recieved MUST be acknowledged with an OK, this is the protocol most other systems use.

wolfmanjm, 2018-02-21:
There is no buffer overflow in Smoothie as the USB serial has its own  flow control (you can actually cat a gcode file if you want)

closed 2018-02-21 by wolfmanjm

slundell, 2018-02-21:
OK. I will make a proper PR to cura once I get their toolchain up and running.