Smoothieware/Smoothieware #1273: TevoLittleMonster SmoothieBoard - Play Sound/Beep/Buzzer

Thisismydigitalself, 2018-02-28:
#### What were you doing?
I'm here asking if it is possible to send Play sound / Buzz(er) Gcode commands to Smoothie's TFT screen's buzzer, or to add a buzzer to Smoothieboard.
  
#### What did you expect to happen?
Sending M300 (I know M300 is not supported by Smoothie - looking for a solution/workaround/substitute) to play a beep like with my Marlin based printer.
#### What happened instead?
Nothing.

#### Branch & Commit or Version of Smoothieware
Recv: Build version: upstreamedge-6e4c7fe, Build date: Apr 12 2017 16:50:09, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)
The one that comes with Tevo Little Monster. do no the version/model.

#### Host and Operating System running the Host
Octoprint
 
#### Link to and console logs
..
    [On gist.github.com or pastebin.com. Always include and never truncate.]
    ..
#### Link to running config and M503 results
..
    [On gist.github.com or pastebin.com. Always include and never truncate.]
..
I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 
..
Copy-paste this template **completely**. Do not skip any lines!
..
It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)
..

wolfmanjm, 2018-02-28:
If you want this you would need to add it yourself. There is a buzz() call in the panels that support a buzzer. The developers currently have no plans to add this.

wolfmanjm, 2018-02-28:
A work around is to add a PWM switch and a buzzer to a spare pin.

closed 2018-02-28 by wolfmanjm

openbuilds-engineer, 2018-03-23:
I wrote a module that supports M300: https://github.com/OpenBuilds/FW-Smoothieware/tree/apex/src/modules/utils/beeper

wolfmanjm, 2018-03-23:
@openbuilds-engineer that code is not legal in Smoothie, you are not allowed to call wait_us() as it basically stalls the entire system. This is the main reason I suggest using a PWM pin instead.

openbuilds-engineer, 2018-03-27:
@wolfmanjm sorry only got a little time to check in on this again. 

wait_us was borrowed from the existing panel code: 

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/panel/panels/ReprapDiscountGLCD.cpp#L67-L80 - the buzz function here is just used with m300 as well.   Does this mean one should actually remove that from the other functions too? 

PWM = do you have an example somewhere of changing the PWM freq (as to replace the current Buzz function)

wolfmanjm, 2018-03-27:
Yes that needs to be replaced, and was on my todo list :)

Your example code is just as bad as wait_us() it would completely stall all other code from executing.

There are two ways to do this correctly...
1. use a timer interrupt to pulse the pin. (There are not many timer interrupts available ATM) you could use the slow ticker if it is fast enough (goes upto about 2KHz).
2. Use the H/W PWM pin (or any S/W PWM pin)

The frequency for PWM pins can be set as per the temperature control code for S/W PWM or the laser control code for H/W PWM. NOTE that H/W PWM all use the same frequency so if you change the frequency and use H/W PWM for anything else (laser and servos) then you would be SoL.

To be honest my take on this is Smoothie is for motor control, if you want to play tunes use an arduino :) I doubt I would accept any code that allows tunes to be played, it is frivolous and not really practical.