Smoothieware/Smoothieware #1283: Soft Endstop don't work

martind-n, 2018-03-09:
I've found you guys now support Soft(ware) endstops in smoothie and I think it's great.

The only thing is I'm not getting this working, this is my config file

# Software Endstops
soft_endstop.enable         true         # Enable soft endstops
soft_endstop.x_min          0            # Minimum X position
soft_endstop.x_max          295          # Maximum X position
soft_endstop.y_min          0            # Minimum Y position
soft_endstop.y_max          315          # Maximum Y position
soft_endstop.z_min          0            # Minimum Z position
soft_endstop.z_max          290          # Maximum Z position
#soft_endstop.halt          true         # Whether to issue a HALT state when hitting a soft endstop (if false, will just ignore commands that would exceed the limit)

But after homing I'm still able to move to a position <0 or >Maximum position or each axis.

Is there anything I'm doing wrong here? Also, after sending M311 on serial interface nothing happens, pronterface stays waiting for an answer that never arrrives.

I'm using the latest firmware downloaded just a few hours earlier today.

Thanks in advance!
Martin.-

wolfmanjm, 2018-03-09:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2018-03-09:
as you did not providei that actual information required we are closing this. you are NOT using the latest code as it would be the firmware-latest.bin, or you need to build it.

closed 2018-03-09 by wolfmanjm