Smoothieware/Smoothieware #1284: Feature Request:  G92.1 applied to individual axis

Zaaphod, 2018-03-09:
#### What were you doing?
G92.1Z

#### What did you expect to happen?
Clear the G92 offset to the Z axis but leave X and Y alone

#### What happened instead?
All Axis were Cleared

#### Branch & Commit or Version of Smoothieware
Current Edge as of 3/8/2018

#### Hardware model you are using (Make, and version)
Smoothieboard 0x 

#### Host and Operating System running the Host
Production Automation Gcode processor

#### Terminal Output
G92X1Y2Z3
ok
?
<Idle|MPos:0.0000,0.0000,0.0000|WPos:1.0000,2.0000,3.0000|F:600.0000>
ok
G92.1Z
ok
?
<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:600.0000>
ok

wolfmanjm, 2018-03-09:
This would violate the NIST gcode standard... G92.1 takes no parameters.

closed 2018-03-09 by wolfmanjm

wolfmanjm, 2018-03-09:
However G92.3 Z0 will do what you want.

Zaaphod, 2018-03-09:
Thank you very much for the advice, I did not know there was a G92.3, It's not listed here: http://smoothieware.org/supported-g-codes which is where I always go.   I'm guessing G92.3 sets the coordinates relative to Mpos?  so I could do G92.3Z1 which would be the same as doing a G92Z1 if Mpos Z was 0?    However it looks like it always sets it to mm

?
<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:600.0000>
ok
G92X1Y2Z3
ok
?
<Idle|MPos:0.0000,0.0000,0.0000|WPos:1.0000,2.0000,3.0000|F:600.0000>
ok
G0X2Y4Z6
ok
?
<Idle|MPos:1.0000,2.0000,3.0000|WPos:2.0000,4.0000,6.0000|F:600.0000>
ok
G92.3X0
ok
?
<Idle|MPos:1.0000,2.0000,3.0000|WPos:1.0000,2.0000,3.0000|F:600.0000>
ok
G92.3Y0.5
ok
?
<Idle|MPos:1.0000,2.0000,3.0000|WPos:1.0000,2.0197,3.0000|F:600.0000>
ok
G0X0Y0Z0
ok
?
<Idle|MPos:0.0000,-0.0197,0.0000|WPos:0.0000,0.0000,0.0000|F:600.0000>
ok

As you can see Y has been adjusted by 0.0197 which is 0.5/25.4 

ahh me and my silly inches... I don't actually need this right now, but it might as well be correct.
I would be happy to go in there and try to fix this, or I can leave it for you to fix

wolfmanjm, 2018-03-09:
no this is a special gcode for loading g92 from the config override and must not be changed, or documented. it is an internal code. sorry there is no way to do what you want if you want inches as it is not supported by the nist spec.

Zaaphod, 2018-03-09:
Ah, well don't want to break anything... Thank you again for offering the solution with G92.3  Can G92.3 be added to the page at: http://forum.smoothieware.org/supported-g-codes ? or is it better to be an undocumented feature?