Smoothieware/Smoothieware #1285: G32 fails when setting Soft Endstops

martind-n, 2018-03-13:
#### What were you doing?
Running G32 command in order to get the plane for auto bed leveling/compensation

#### What did you expect to happen?
Get the plane built from the three point set in config file

#### What happened instead?
Machine stops before first meassurement with the message
 
Error: Soft Endstop Z was exceeded - entire move ignored
[ERROR] Error: Soft Endstop Z was exceeded - entire move ignored

#### Branch & Commit or Version of Smoothieware
Build version: edge-a773658, Build date: Mar  2 2018 11:21:40, MCU: LPC1768, System Clock: 100MHz
5 axis

#### Hardware model you are using (Make, and version)
MKS Sbase 1.3

#### Host and Operating System running the Host
Pronterface

#### Link to and console logs
SENDING:G32
Error: Soft Endstop Z was exceeded - entire move ignored
[ERROR] Error: Soft Endstop Z was exceeded - entire move ignored
    
#### Link to running config and M503 results
configtxt: https://gist.github.com/martind-n/3975df3c06fef6f2d86e8db29258f87c    
M503 results: https://gist.github.com/martind-n/e5624bf0f2ecc9929f05a3a9596d8dc8

wolfmanjm, 2018-03-13:
Then do not set your z where it is violated, or use the M code to disable soft endstops before running G32

closed 2018-03-13 by wolfmanjm

martind-n, 2018-03-13:
OK so I should do this?

M211 S0
G32
M211 S1

Why is that enablig soft endstop for Zmin G32 command doesn't work?

Thanks,
Martin.-