Smoothieware/Smoothieware #1287: Release firmware.bin MASTER branch bug on MKS SBASE boards.

pixel3design-hub, 2018-03-15:
I found that latest version of precompiled smoothie firmware.bin in master branch is buggy on MKS SBASE boards.
Printing results in vertically skewed objects.

I've teared down all the machine, rebuild it from ground up, changed motors, belts, bushings, rods, ball bearings and plastic parts.

Same result. Swapped 3 boards. No luck.
Then I roll back to an earlier fw version and BANG. It works.
So then I swapped the MKS board with a Smoothieboard and it worked. So I get back to the new firmware release on the Smoothieware board and it worked. Back again on the MKS with new firmware and it failed. Back on the old firmare and it worked.
No changes made on the configuration file whatsoever.

Seems that the BETA is losing steps or ALPHA does not have enough holding retention power.
Maybe a bug in the current control drivers code?

My machine is a CoreXY mechanism.

wolfmanjm, 2018-03-15:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2018-03-15:
I suggest you contact MKS for support.

closed 2018-03-15 by wolfmanjm

pixel3design-hub, 2018-03-16:
No. That's not an MKS issue. It's about firmware.
You made the firmware, you have to support firmware users, no matter what board they use.

Are you pissed off by MKS? Indeed they don't behave the right way. Think you're better than they? Show me and give support to users.
Or should I intend you deliberately make firmware buggy just for MKS boards? Is that the case?
Or most simply you're not focused anymore on fw development and debugging? Because if I wrote a brilliant project and someone tells me that a bug affect the functionality of it, I'll make al least a test to wipe away the bug. Indeed you are able to avoid the support for electronics related bugs, but not for the firmware.
Your behavior is not correct the way you intend to.
How much people use Wikipedia and not spent a second or a dollar to support it? Did Wikipedia ban their ip? Not. They ask people to support them. Did you want money for your support? I can afford it. Others can't.
But community can't be treated like garbage because of MKS behaviors. I didn't do anything bad to you neither the community. And until you are on the GPL side, you are in charge to give support to peoples who are using your piece of software.
I can understand you. Really. But if you know how this dev/chineese war works, you should bear the fact that chineese companies will always make money on your work when you publish it opensourced.
It's a matter of choices.
I will be always on the community side, giving support to EVERYONE who ask for it. No matter what. Makers do it.
You said to ask MKS because you know they doesn't give any. If you check on their Repo you'll see me answering the most of the questions about boards problems. I'm not an MKS employee.
As long as I know, you made a commercial board and an opensourced firmware that can run on ANY LPC17xx MCU. Am I wrong?
So... Give us support on your firmware Sir.

Cheers

arthurwolf, 2018-03-16:
http://smoothieware.org/troubleshooting#somebody-refused-to-help-me-because-my-board-is-a-mks-what-s-that-all-about

also, not a firmware issue.

arthurwolf, 2018-03-16:
> refused-to-help-me-because-my-board-is-a-mks-what-s-that-all-about

Actually this is even more relevant :
http://smoothieware.org/troubleshooting#i-m-very-upset-at-something-everyone-is-mean-and-nobody-listens-to-me

wolfmanjm, 2018-03-16:
1. we are under no obligation to support MKS boards
2. we don't have any MKS boards to test with
3. if it works with a smoothieboard then it can't be a firmware bug
4. You have given no information that could be used to debug
5. The information you have given indicates it must be a board H/W issue
6. The MKS boards are known to be very badly designed and cause all sorts of issues, google should bring up those issues for you.