Smoothieware/Smoothieware #1288: $G does not return actual spindle state (M3/M5)

evgs, 2018-03-20:
Current implementation of $G (View gcode parser state) always returns spindle state M5 (hardcoded here: 
https://github.com/Smoothieware/Smoothieware/blob/573ed7c0c719008075acb1d36850e718a2707372/src/modules/utils/simpleshell/SimpleShell.cpp#L833 )

Some software may be confused with this behaviour :)
For example, bCNC 0.9.10 periodically polls machine and resets GUI spindle control to 'OFF' state. So I can't jog spindle speed by this slider during milling.

wolfmanjm, 2018-03-20:
I don't think this is fixable. What do you suggest? can I omit that entirely, as it is not possible to get actual spindle state.

wolfmanjm, 2018-03-20:
This would require some changes to the spindle, which would need to be done by the spindle author. You could ask him to add a public request so we can check if spindle is running or not etc.
The core developers are unlikely to do this though.

cilynx, 2018-11-16:
Hey Jim -- could you take a look at https://github.com/cilynx/Smoothieware/commit/8f336f24e6c0e6ab045ef914241cb7c9af2beded please?  As the saying goes, "it works on my machine", but I'm not sure if my implementation here is following the standards you're going for.  

```
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F4000.0000 S0.8000]
Recv: ok
Send: M957
Recv: Current RPM:     0  Target RPM:  1000  PWM value: 0.000
Recv: ok
Send: M3
Recv: ok
Send: M957
Recv: Current RPM:     0  Target RPM:  1000  PWM value: 0.654
Recv: ok
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M3 M9 T0 F4000.0000 S0.8000]
Recv: ok
Send: M5
Recv: ok
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F4000.0000 S0.8000]
```

wolfmanjm, 2018-11-16:
it looks ok, it is not how I would do it though, returning the instance of the spindle is bad form IMO. I would have had a public request to get spindle state and just returned a bool true/false for whether the spindle was running or not. I think that is safer and maintains the class integrity. But that is just my opinion.

cilynx, 2018-11-17:
Right on -- I wasn't sure which way to go between returning the instance (like the Laser module) or putting together a struct (like most of the rest of the modules).  Refactored to align with the struct methodology and tested again on my lathe.  Let me know what you think.  If you're good with it, I'll send the PR.

https://github.com/Smoothieware/Smoothieware/compare/edge...cilynx:feature/parser_state_spindle_state

```
Send: M115
Recv: FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:feature/parser_state_spindle_state-9f5c87c, X-FIRMWARE_BUILD_DATE:Nov 16 2018 16:12:07, X-SYSTEM_CLOCK:100MHz, X-AXES:3, X-GRBL_MODE:1, X-CNC:1, X-MSD:1
Recv: ok
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F4000.0000 S0.8000]
Recv: ok
Send: M3
Recv: ok
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M3 M9 T0 F4000.0000 S0.8000]
Recv: ok
Send: M5
Recv: ok
Send: $G
Recv: [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F4000.0000 S0.8000]
Recv: ok
```

wolfmanjm, 2019-03-19:
I have implemented this myself for a switch named spindle. I therefore cannot merge this, however if it is OK with you I'll manually merge the spindle part of this, and update mine to check spindle first then check switch next if spindle is not enabled. It is easier to have a switch for simple on/off of spindles than use the spindle module.

closed 2019-03-19 by wolfmanjm