Smoothieware/Smoothieware #1290: XY positions are weird when two corners leveling activated

gtopba, 2018-03-24:
#### What were you doing?
I am building a mini CNC for PCB milling. I want to use the two corners levling strategy.

#### What did you expect to happen?
The two corners leveling do Z compensation currectly but X,Y position are weird.
#### What happened instead?
Running Gcode from sdcard or sending Gcode from serial terminal to move the spindle to the position inside the leveling area causes the spindle to stop at wrong position. The xyz coordinates display on the LCD screen currectly but the physical position of the spindle is not. It is ofset by around 3-5mm in both X and Y directions. There is an abnormal high pitch noise from the (X and Y ) steper motors right before the spindle stop moving. I tried slowly changing target position in the Gcode, in some point, the problem dissapeared and the spindle landed at currect position. The same Gcode runs currectly if the leveling is deactivated.

#### Branch & Commit or Version of Smoothieware
    Branch : edge
    firmware-cnc-latest.bin

#### Hardware model you are using (Make, and version)

   Smoothieboard 5X

#### Host and Operating System running the Host

    Windows 10 , bCNC

#### My Leveling command 
 <script src="https://gist.github.com/gtopba/a2c45ee9299a6e7803b99fd02d34bcae.js"></script>

#### My G54
<script src="https://gist.github.com/gtopba/91189b7a7d20b981d8d3eda6645fcec1.js"></script>

#### My Gcode
   The problem at line 6
<script src="https://gist.github.com/gtopba/fa81d343725eebf2b147f34a85e7b743.js"></script>

#### My config
<script src="https://gist.github.com/gtopba/d283238820088202a547ba03838b7dec.js"></script>

#### M503 result
<script src="https://gist.github.com/gtopba/0b40d7a857af9a53abd01928438fe735.js"></script>

wolfmanjm, 2018-03-24:
This sounds more like you are trying to move too fast, as levelling would not affect XY positioning at all. Please try running at a much slower speed and see if that helps (try at half or quarter the feedrate).

wolfmanjm, 2018-03-24:
Also your G54 looks very suspicious. seems to be way off.

wolfmanjm, 2018-03-24:
If you are using tr8x8 leadscrews then your max feedrates are way too high. Generally these leadscrews can do 30mm/sec maximum, so set the M203 to X30 Y30 Z30.

gtopba, 2018-03-24:
My ballscrew is 4mm/rev. My XY table is 300x300mm. The problem still occurs even I change the G0 feedrate to 500mm/min. The stepper motors still produce high pitch noise ("beep") for short period (1 sec) right before reaching the target pos. The same Gcode works find repeatably while the leveling is deactivated. Tomorrow, I will try reducing the max feedrate.

wolfmanjm, 2018-03-24:
I am sorry I cannot reproduce this, it does seem to work fine. The high pitch shriek of a stepper is when it stalls which is usually due to running it too fast. Please keep this issue updated with your findings, but as this cannot be reproduced we cannot fix it.

wolfmanjm, 2018-03-24:
additionally at line 9 of your g code you are moving the z axis at F3000 which IMO is also way too fast for any Z.

wolfmanjm, 2018-03-24:
Another thing to look at is the grid itself, look for any anomalous Z values where it maybe trying to move a long Z. Also remember that the grid must only operate within the scanned region if you try to move even a little outside that region the Z offset will be undefined and could cause undefined results.

wolfmanjm, 2018-04-18:
As no more information is forthcoming I am closing this issue.

closed 2018-04-18 by wolfmanjm

gtopba, 2018-05-25:
I don't think it is about feedrate stuff. Seems like I can avoid this problem by enlarge the grid area. And I have checked carefully that my working area is inside the grid area. However, working area outside grid area will work just fine as I tested. For example, my grid area was 2x1 cm which had problem then I changed grid to 5x5 cm, now it works perfectly. I suspect the problem only occur with very small grid area.  But now I'm happy with this feature. Yes, I have read the code in CartGridStrategy.cpp and everything look good, it does not modify x, y at all. I want to study all the code in Smoothieware project. May I ask you stupid question, what is toolchain do you use to develop code? Could you recommend me development tools? I have seen you use Travis CI to build code which seems interesting.

wolfmanjm, 2018-05-25:
Please do not use issues to ask questions. Especially when the answers are clearly in the wiki.
http://smoothieware.org/compiling-smoothie