Smoothieware/Smoothieware #1291: G2/G3 do not always perform full circles

Zaaphod, 2018-03-26:
#### What were you doing?
Running the following Gcodes:
G0X7.86Y13.96M400 
M400
G1Z0
G2X7.86Y13.96I-0.11J-0.06     

#### What did you expect to happen?
A full circle to be performed

#### What happened instead?
no circle was performed.

#### Branch & Commit or Version of Smoothieware

edge-e9942d1, Build date: Jan 1 2018 22:24:20, 
CNC build

#### Hardware model you are using (Make, and version)

Smoothieboard 0x

#### Host and Operating System running the Host

Production Automation Gcode host

#### Link to and console logs

2018/03/25 07:13pm COM3: --->N15G0X7.86Y13.96M400*113
2018/03/25 07:13pm COM3: <---ok
2018/03/25 07:13pm COM3: --->N16M400*48
2018/03/25 07:13pm COM3: <---ok
2018/03/25 07:13pm COM3: --->N17G1Z0*84
2018/03/25 07:13pm COM3: <---ok
2018/03/25 07:13pm COM3: --->N18G2X7.86Y13.96I-0.11J-0.06*2
2018/03/25 07:13pm COM3: <---ok
    
#### Link to running config and M503 results

Config - https://pastebin.com/F4U55Yuu
M503 - https://pastebin.com/ktuYRg57

#### Additional information:
This is a known bug and has been happing a very long time.  The quick fix has been to break full circles into 2 arcs, however trying to interface with a new cam program is making breaking the circles into arcs time consuming so I'm now trying to solve this issue.

here are more examples:
G0X18.038253Y9.539614M400
G1Z0
G2X18.038253Y9.539614I-0.108253J-0.0625
Does not perform the circle

G0X18.0383Y9.54M400
G1Z0
G2X18.0383Y9.54I-0.1083J-0.062886
Does not perform the circle

G0X18.0383Y9.54M400
G1Z0
G2X18.0383Y9.54I-0.1083J-0.063
DOES perform the circle

I suspect that what is happening is that for some reason, the G2 commands on the circles that do not cut are being processed as microscopic arc segments.   I have not been able to figure out why this occasionally happens.  I have a feeling like it's something along the lines of that the G0X18.0383Y9.54 line cannot land on an even step exactly of course, between the steps per mm being 41.58464566929134 and the fact that I am running in G20 inches so always converting to mm, even if I do G0X1Y1 I do not land on an even step... so if it's really ending up at something like X=18.0382Y9.5339 and then I give it my G2X18.038253Y9.539614I-0.108253J-0.0625  It is calculating that it needs to do a tiny little arc segment.   

This is just a possible explanation for this behavior.  I'm fairly certain that the issue is that it's calculating a tiny arc segment instead of a full circle because something is off... but the reason it is off, I do not know.  I have been trying to determine where this might be happening, but I'm not too familiar with the way the buffer works, and I'm not sure if it's happening before the buffer or after it, and I'm not sure exactly what's causing it, it could be floating point weirdness, inches conversion, steps per mm conversion... something somewhere is making it think the G2 end point is not exactly the same as the pervious G0 used for the start point, and so it is doing a microscopic arc segment, and not a full circle.   It's most likely that when it does appear to do a full circle, it's really a full circle minus a tiny arc segment... the error just happens on the opposite side so it has to go almost all the way around to get to the end point.

wolfmanjm, 2018-03-26:
Did you try the latest edge source? I fixed some issue with tiny segments causing huge circles to be drawn, maybe the same issue.

Zaaphod, 2018-03-26:
I will try it today... it does sound like the same issue on the other side... if the program has a what is supposed to be a tiny little arc segment in it and the start and end point are so close together it calculates it to be a full circle instead of a tiny segment.

Zaaphod, 2018-03-26:
I've just tested with the Current Edge and have the same issue.

I re-tested with previous version which I made sure was before PR#1272 and also have the same issue.

This confirms that PR#1272 is not the cause of this, nor does it fix it.

wolfmanjm, 2018-03-26:
did you rebuild? because the pre built edge does not have it.

wolfmanjm, 2018-03-26:
you can also try to tweak that ARC_ANGULAR_TRAVEL_EPSILON make it smaller see if it helps, I am not sure what it is for anyway.

Zaaphod, 2018-03-26:
Yes I did re-build it myself, and it did have the changes from PR#1282 in it.

I'm not sure what ARC_ANGULAR_TRAVEL_EPSILON is for either, but I will try to figure it out.

I thought I would also try to output the actual numbers being used for the start point and end point... they should be identical at max precision, but if not, then we may be dealing with a conversion or floating point issue

wolfmanjm, 2018-03-26:
It is used here...

`float millimeters_of_travel = hypotf(angular_travel * radius, fabsf(linear_travel));`

the `angular_travel` is used to make sure it goes the correct way. My issue was millimeters of travel was too big for very small arcs, causing it to travel way too far.

Zaaphod, 2018-03-27:
I put some output a bunch of variables to the console just to try to get a handle on what's going on here.  I've put quite a bit of time into trying to figure this out, but I am at a point where I really don't know enough about what happens before this part of the code to understand what is going on here.  so I thought I would share my efforts so far in the hopes that it will be helpful.

Code I used to generate reports of variables is at:
https://github.com/Zaaphod/Smoothieware/pull/10/commits/a5bd9b6172fbf28584e306c429ff91a34799e9a5

The Results of some tests are at: 
https://pastebin.com/msGGYuuh

First thing I notice is that millimeters_of_travel for the circles that fail are tiny,  so indeed it is calculating a tiny microscopic arc segment instead of a full circle.... I have not tested yet, but I suspect that getting a full circle when you expect a tiny segment could also happen.  I have had that problem in the past, but do not have an example handy.

I don't really understand what's going on with ARC_ANGULAR_TRAVEL_EPSILON and why it is needed... it's not needed to figure out if it's a full circle or an arc segment,  all that is needed to do that is to compare the start coordinates to the end coordinates, if they are exactly equal, it's a full circle.. absolutely no exceptions.  If the user send G0 X1Y1, and then G2X1Y1I-0.1J-0.4  then it is a full circle...  ALWAYS.  If X=prevX and Y=prevY then it's a full circle.   If the user sends G0X1Y1 then G2X1.000000001Y1,I-0.1J-0.4 then it's NOT a full circle. It may be 99.999999999% of a circle or 0.00000000001% of a tiny ridiculous arc segment, but that's all the fault of the Gcode not sending what should be sent.... the results will always calculate to be less than 2pi and greater than 0 and should not need to be fixed.  by definition however, if X=PrevX and Y=PrevY of the actual Gcode sent, then it must be a full circle always with no exceptions.... Then after that, there can be a filter to try to reject micro segments or to try to prevent circles from hacking through the part both of which can be cause by rounding errors when the gcode files is produced and generally exporting more decimal digits would fix it.....  There is a known issue with microscopic arc segments on ALL Gcode systems just by the nature of Gcode, if the beginning and end point are less than the resolution you are exporting in your gcode, they are going to look like a full circle... if you have a tiny arc segment that starts at X5.432Y1, and ends at X5.4321Y1 and you only export 3 decimal places, you are going to get 5.432 for both the start and end and you will and should get a full circle.  In my host I can filter these out with a disable_full_circles command, meaning that part of the code is expected to not have a full circle, so any apparent full circles are ignored... Unfortunately there is no gcode standard to perform this function, so it's pretty much up to the host.  I also discard any arc segments smaller than a defined tolerance and replace them with lines... this is something that can always happen automatically... if you determine the angular travel is less than say 0.025mm, then you might as well just move to the end point in a straight line and not bother with further arc calculations, although such tiny segments should fall through all the arc calculations and end up with only one segment anyway, so it's probably not worth the extra code.

from the results I have here,  for example on the first test, the start machine position and target are both identical, so it should have been forced to do a full circle,  no real calculations are needed, if X=XPrev and Y=YPrev then Angular_Travel = 2Pi always ... this takes care of any possibility that Anglular_Travel comes out to 0, and it forces it to be 2Pi which is what it must be.   After this is sorted out, anything where  X<>XPrev or Y<>YPrev must be less than 2Pi and greater than 0 Angular_Travel, now,  go calculating the Arctan and try to figure it out...  doing the compare first saves you bothering with the arctan calculation and it also should save you from fixing the arctan calculation after, and I think we would find that using the exact result of the arctan calculation would be usable without ARC_ANGULAR_TRAVEL_EPSILON

So while I really don't have this nailed down in any way, there are two different ways this issue is coming up,  one always fails:
G20
G0X7.86Y13.96
G2X7.86Y13.96I-0.11J-0.06
this circle fails even if I move it around,  Any time I use I-0.11J-0.06, it fails... except at X0Y0

and one intermittently fails.  
G21
G0X199.644Y354.584
G2X199.644Y354.584I-2.794J-1.524
I did not try moving this one yet... but it's strange that in several sets of tests, I've observed this code passing and failing, more than once of each... I have not determined a pattern to force it to fail or pass yet.

Both the G20 and G21 circles above describe exactly the same circle with no rounding error... the G21 values are exact, not approximations of the inch values

Something is odd with the intermittently failing example.  If you look at the successful run that starts at line 62, Mpos of the Y Axis is 354.5839843750000000   but it SHOULD have been 354.584014892578125... I do not have an explanation for why this number is incorrect.  The previous move was G0X199.644Y354.584.   I used the floating point converter here: http://www.exploringbinary.com/floating-point-converter/ and found out it should be 354.584014892578125

[... cut ...]  2Pi... but then we still need to find out why mpos was off.

I need a bigger piece of paper to fill up with circles with right triangles inside them and formulas and figure this out.  I'm wondering if Fixed point would be more accurate, at least with fixed point the numbers being represented would match the Gcode input exactly and be much easier to follow.  We don't even need to change the precision, 4 digits before the decimal and 5 after should be sufficient for everything.... just a thought.

wolfmanjm, 2018-03-27:
I think this smacks of float imprecision. Have you tried using doubles instead of floats where it matters? Most of the numbers you are showing are way out of the epsilon range for floats. This is probably what the ARC_ANGULAR_TRAVEL_EPSILON is supposed to do, when you compare two floats you actually have to compare them withing an epsilon value. If it works with doubles then we can consider if we need to stick with doubles or use fixed point instead.

FYI [float epsilon](https://en.wikipedia.org/wiki/Machine_epsilon)
and [comparing floats](http://floating-point-gui.de/errors/comparison/)

Zaaphod, 2018-03-27:
Update:  I think I've figured out why there is a difference in machine position... it's the inch conversion being floating point that is causing it.  If I am in G20 and go to Y13.96, then Y = 354.58398437500000  but if I am in G21 and go to Y354.584 now Y = 354.58401489257812500   but this only happens if Y was somewhere else first.   If I do a G20 and move to Y=13.96, then immediately do G21 and then move to Y354.584, Y remains 354.58398437500000.  I suppose it doesn't need to move so it doesn't change anything... but it should be changing it to the proper floating point value, even if it doesn't move otherwise the math for the arcs is off.... however in all practicality this should never be an issue, because it would be extremely rare that someone did a G20 then a G0XY move, then a G21 and a G2 move

I feel that doing the mm to inch conversion in floating point is  introducing an error that would not exist if it was done with fixed point. because we are always multiplying and dividing by 25.3999996185302734375 not 25.4, and that is the reason when in G20 mode our Y is not identical to the Y we have in G21.   Y should always be 354.58401489257812500  whether I do G20 and a G0 Y13.96 or a G21 and a G0Y 354.584, and changing just the mm to inch conversions to fixed point would fix that.  It would be better to truncate a number with ridiculous precision occasionally than to always be multiplying by the wrong constant.  doing a G20 then G0Y 13,96 in fixed point would be exactly the same as doing a G21 G0 Y354.584, because with fixed point math, the conversion would come out to exactly 354.584 not 354.5839946746826171875.   That's the whole problem with floating point, you start multiplying and dividing and taking square roots of things and it gets off real fast, the more multiplies you have the worse it is.   Fixed point math doesn't have as much precision, but the digits it does have are exact.  

The mm to inch conversion is not the problem with arcs, it's just an annoyance, because if you are always in G20 (which I am normally except for this testing) then it's always doing it the same way, and Mpos and target are both identical so the compare would work.... and the math should work... but the math is producing angular_travel2:-0.0000046134859985613729804754257202 when it should be 2Pi.  It would be nice if it were calculated with Fixed Point, but not necessary to fix this issue.

Now that I know the reason for the difference with Mpos is caused by switching between G20 and G21 without actually moving anywhere,  we can discount that as a fluke of not moving anywhere after switching.  Which means that if we do a G0XY and then a G2XY to identical XY coordinates, we would always detect a full circle by doing if MposX=TargetX AND MposY=TargetY then just set angular_travel to 2Pi else do the arctan calculation and figure out what the angular travel is for arc segments.  I think we may need 2 arctan calcualtions, but would only run one Or the other... if G2 run one, if G3 run the other, then we don't need any Arc_Angular_Travel_Epsilon at all, the result of the correct Arctan calculation will be precise and give the exact correct answer and we don't need to figure which way it was going.... the direction was given to us at the beginning, by defining G2 or G3, why should we have to compensate for doing the wrong Arctan calculation... just do the right one, then it's impossible to get a tiny arc segment when you expect a large arc, or full circle, and it's also impossible to get a large arc or full circle when you expect a tiny segment.

My plan is to test this by temporarily doing both procedures back to back and generating a report every time an arc runs.

wolfmanjm, 2018-03-27:
As I said it is not correct to expect float == float to work. unless it is zero. Doing all calcs in fixed point is not an option either.

The arc calcs need to accommodate the fact floats are imprecise, there is no way to make them more precise.

If the problem is inch to mm conversion then for practical purposes it is not an issue for Smoothie as you are the only one using inch mode :) (sorry)

wolfmanjm, 2018-03-27:
> the direction was given to us at the beginning, by defining G2 or G3, why should we have to compensate for doing the wrong Arctan calculation... just do the right one, then it's impossible to get a [...]

I like this approach :) The code was copied from GRBL, so could probably be improved.

Zaaphod, 2018-03-27:
Thank you for the links, they are most helpful!

I see what you mean about float == float...   I think the a proper float compare would still be the first thing to do though. that would weed out the full circles, then anything left is an arc segment... just do the correct arctan calculation and we're done.  we would still use the epsilon value, but at the beginning for the initial compare, not the end after we have introduced a lot of floating point weirdness to everything

I'll try a quick test as you suggest and change the variables that matter to doubles just to see if that alone changes the results. 

I'm not too concerned about inch to mm conversion, as I said it's just an annoyance, and it only ever shows up if you switch and don't move.

I don't think all calcs need to be in fixed point, I think that if only variables that were involved with Gcode coordinates were stored as fixed point it would make everything a lot easier.  the calculations can all be floating point and the results stored in floating point variables, but storing the Gcode coordinates as fixed point would mean that we are always retaining the actual Gcode Value, not a floating point representation of it.

wolfmanjm, 2018-03-27:
that maybe true, however there is no way i am changing the coordinates to fixed point. for what smoothie is used for float is sufficient.

Zaaphod, 2018-03-27:
well they don't need to be 64bit fixed point numbers, they can be 4 digit's before the decimal, and 5 digits after, that gives us a range of -9999.99999mm to 9999.99999mm which is more than sufficient and uses the same memory as floats, but we can do == instead of something like this:

	public static boolean nearlyEqual(float a, float b, float epsilon) {
		final float absA = Math.abs(a);
		final float absB = Math.abs(b);
		final float diff = Math.abs(a - b);

		if (a == b) { // shortcut, handles infinities
			return true;
		} else if (a == 0 || b == 0 || diff < Float.MIN_NORMAL) {
			// a or b is zero or both are extremely close to it
			// relative error is less meaningful here
			return diff < (epsilon * Float.MIN_NORMAL);
		} else { // use relative error
			return diff / Math.min((absA + absB), Float.MAX_VALUE) < epsilon;
		}
	}

I don't know... the above function is not really all that bad, probably better than trying to convert coordinates for some fixed point format.   I'm wondering if there a float comparison function somewhere already?

If you want a headache, try to read this:
https://randomascii.wordpress.com/2012/02/25/comparing-floating-point-numbers-2012-edition/
ugh floating point math, I just want to know if I have a circle or not

Zaaphod, 2018-03-27:
actually... looking at the data here....  we can safely ignore the anomaly caused by switching between G20 and G21 and not moving... and if we do that, then we can do if Mpos==target and it would work to detect full circles.   Because Mpos is the result of storing the value of the pervious Gcode that moved an axis, and target is the result of storing the current gcode that is trying to make the arc.   they are always on the same float... if they end up on different floats the only way that should be possible is if the Gcode did not pass in identical numbers, and if the Gcode did not pass in identical numbers, it wasn't trying to make a circle.   Whatever math is applied to the storage of the Gcode number, it happens the same way for the previous gcode that got it to the start point as it does to the current arc gcode.   So this is an extremely rare case when you could use == with floats

Zaaphod, 2018-03-27:
I can put a test for this into my host, then I can use the host to look for errors,  I can then run a program with hundreds or thousands of different arcs and generate 2 reports, one of all the results, and one of only any exceptions detected.    on the host I can do string compares between the positioning move and the arc, and determine if it should be full circle or not, no possibility of error with string compares... then have smoothie do the compare and output if it thought it was a full circle or not, and see if there is ever an error where the 2 results do not match.    I can actually generate millions of different arcs with my host because I can run a program with nested loops and do math on Gcodes in the loops, and I don't need to confine the results to the size of any of my machines, because I can run this on a smoothieboard on my desk with an insane federate, because we are just looking to verify whatever method of compare we are testing.

wolfmanjm, 2018-03-27:
You cannot use MPos as the actual coordinates go through the Wpos processing, so I don't think you can use == as mpos is NOT what the coordinates are in.

Zaaphod, 2018-03-27:
ooops , by bad, I think I am referring to the wrong thing... When I say MPos, I'm referring to my abbreviation of machine_position[this->plane_axis_0] for machine position that I used in my report, not MPos you get from ?.   The machine_position[] variable should be the result of the previous Gcode stored because this is all happening before things go into the queue and has nothing to do with the physical machine position yet.

Zaaphod, 2018-03-28:
I'm making some progress on this.  I have been so far successful with detecting full circles and eliminating them from the Arctan calculation.  

I've been looking at the math and it's finally making some sense.  The ArcTan calculation can only ever produce a result between pi and -pi (excluding -pi) therefore it would only be able to give results of + 1/2 of a circle or -1/2 of a circle.  The purpose of the correction is that if it is going counterclockwise and we ended up with a negative angular_travel, we add 2Pi to it to make it the positive Angular_travel which is greater than pi... and do the revers to clockwise arcs.   the thing is, we need to do this correction for ALL results that are <=0  I don't see why we would exclude numbers less than the arc_angular_travel_epsilon.  There is no problem in floaing point with >= or <=  expecially >=0 and <=0 because in floating point 0 is the only number you can count on as being absolutely correct.  The intention of the correction is to correct for Atan2 having the limitation of only being able to give a result from pi to -pi, but what we need is a result from 0 to 2pi  for counterclockwise arcs and from 0 to -2pi for clockwise arcs... that is all... leaving out a narrow band of results close to zero is just doesn't seem right to me..  Lets just look at counterclockwise arcs, because the numbers are positive:

`if (angular_travel <= ARC_ANGULAR_TRAVEL_EPSILON) { angular_travel += (2 * PI); }`

with this statement, if angular_travel is positive and less than ARC_ANGULAR_TRAVEL_EPSILON then we are adding 2Pi to it... but now we end up with a result greater than 2pi,  which is more than 360degrees of a circle,  and that's just not right.   the idea is to shift angular_travel from a -pi to +pi to 0 to 2pi..... not to shift it to 0 to 2pi plus a little bit that would make the result greater than 2pi.   It's not needed to correct for any floating point and it's not an == comparison,  so I just can't understand why we don't just use:
`if (angular_travel <= 0) { angular_travel += (2 * PI); }`

I think the problem is that the math has difficulties when you have a full circle and the start and end point are identical... all the floating point math in the Atan2 formula is supposed to all cancel out, but it doesn't quite due to floating point... but the solution or pre-filtering out the full circles seems to be working so far,  so if we pre-filter out the full circles, then I think we don't need the ARC_ANGULAR_TRAVEL_EPSILON anymore.    If we needed an epsilon comparison anywhere it would have been in the full circle filter, but so far I have tested over 100 circles with:

```
if((this->machine_position[this->plane_axis_0]==target[this->plane_axis_0]) and(this->machine_position[this->plane_axis_1]==target[this->plane_axis_1])) {
        gcode->stream->printf("Full Circle: True\r\n");
        if (is_clockwise) {
           angular_travel = (-2 * PI);
        } else {
           angular_travel = (2 * PI);
        }
    } else {
        gcode->stream->printf("Full Circle: False\r\n");
    }
```

and I have had 100% success rate at determining it was a full circle or not.... and I've been trying to trip it up, by sending it ridiculous commands like G0X1Y1 then G2X1.0001Y1I-0.2J0  
I also notice that the original Atan2 formula seems to be very reliable without needing to deal with the full circles.  I want to do more testing though to make sure.  I've been testing  by hand, but I want to do my automated testing which will be much more thorough and try to throw some real edge cases at it like an arc segment with a radius of 1000 inches that a very small segment would fit on a machine table... or micro arc segments.   It's probably going to take me a few days to do all the testing.   for testing purposes I am doing the angular_travel calculation both with and without ARC_ANGULAR_TRAVEL_EPSILON and I will be specifically checking to see if angular_travel is ever greater than 2pi or less than -2pi, as well as checking to see if it ever makes any difference.

wolfmanjm, 2018-03-28:
I think setting ARC_ANGULAR_TRAVEL_EPSILON to 0.0F is worth a try, i set it to a tiny number as the original setting was relatively big and was reversing some small arcs making them do full circles instead

wolfmanjm, 2018-03-28:
in think  `if((this->machine_position[this->plane_axis_0]==target[this->plane_axis_0]) and(this->machine_position[this->plane_axis_1]==target[this->plane_axis_1])) ` should have same wiggle room, so maybe best to use an epsilon there which is smaller than any likely cnc resolution, maybe 0.00001mm ?

Zaaphod, 2018-03-28:
I've tried setting ARC_ANGULAR_TRAVEL_EPSILON to 0.0F and managed to find an extreme edge case that failed.   I was able to correct it by doing:

```
        if (is_clockwise) { // Correct atan2 output per direction
           if (angular_travel > 0) { angular_travel -= (2 * PI); }
        } else {
           if (angular_travel < 0) { angular_travel += (2 * PI); }
        }
```

We really only want to apply the correction if the result is >0 , not >=0.  If it is =0, it needs to stay 0 or you end up with a massive circle.  Here's the test case:

ok
G0X1Y1
ok
G2X1.00005Y1I-2000J0
Full Circle: False
Mpos Plane_Axis_0: 25.3999996185302734375000000000000000
Mpos Plane_Axis_1: 25.3999996185302734375000000000000000
Offset Plane_Axis_0: -50800.0000000000000000000000000000000000
Offset Plane_Axis_1: 0.0000000000000000000000000000000000
Target Plane_Axis_0: 25.4012680053710937500000000000000000
Target Plane_Axis_1: 25.3999996185302734375000000000000000
center_axis0: -50774.6015625000000000000000000000000000
center_axis1: 25.3999996185302734375000000000000000
Radius: 50800.0000000000000000000000000000000000
r_axis0: 50800.0000000000000000000000000000000000
rt_axis0: 50800.0039062500000000000000000000000000
r_axis1: -0.0000000000000000000000000000000000
rt_axis1: 0.0000000000000000000000000000000000
ARC_ANGULAR_TRAVEL_EPSILON: 0.0000000049999999696126451453892514109611511230468750000000000000
initial angular_travel1: 0.0000000000000000000000000000000000
old angular_travel2: -6.2831854820251464843750000000000000
new angular_travel2: 0.0000000000000000000000000000000000
millimeters_of_travel:0.0000000000000000000000000000000000
ok

New_angular_travel2 is from the above code using >0  Old_angular_travel2 is the original code with epsilon.  The reason it came out to exactly 0 is because my start X and end X were only 1 float away from each other, and the math just can't do anything but come up with zero using single precision floating point...  It's ok for this to happen, but if something tries to do a ridiculous operation like this, then we are better off with 0 than a massive incorrect circle... using >0 and <0 basically forces anything that caused the math to fail to also fail.

I don't know that we need the wiggle room, but if we put it in we have to be very careful about the full circles.   The problem is if someone tries to just type a true type font and generate gcodes from it, many of those are terrible with the microscopic arc segments that have technically different start and end points, but they are way way less than the machine resolution.  by making an exact compare, we will filter those out and they will always be ignored,  but if we give it wiggle room we may end up including them as a full arc, which we defiantly do not want to happen.  It think it's better to have an occasional edge case non-full circle than an occasional edge case huge circle that hacked through the entire part.  Also if the start and end point are not exactly equal and fail the full circle test,  the Arctan formula is still working and you will get 99.9999% of a full circle,  which is basically what you asked for... it's only startpoint == endpoint that the arctan formula fails, unless I have a radius of 2000 inches (50800mm) and 1 float away on an axis.  

Here's an extreme example: of being one float away on X and a radius of 81 inches (2057mm) which does not fail the Arctan math.  It fails the full circle test, as it should because X is off by one float, this is a super micro arc segment. but this proves the Atan2 formula works in this extreme case.

G0X1Y1
ok
G2X1.00005Y1I-40J-70
Mpos Plane_Axis_0: 25.3999996185302734375000000000000000
Mpos Plane_Axis_1: 25.3999996185302734375000000000000000
Offset Plane_Axis_0: -1016.0000000000000000000000000000000000
Offset Plane_Axis_1: -1778.0000000000000000000000000000000000
Target Plane_Axis_0: 25.4012680053710937500000000000000000
Target Plane_Axis_1: 25.3999996185302734375000000000000000
center_axis0: -990.5999755859375000000000000000000000
center_axis1: -1752.5999755859375000000000000000000000
Radius: 2047.8134765625000000000000000000000000
r_axis0: 1016.0000000000000000000000000000000000
rt_axis0: 1016.0012207031250000000000000000000000
r_axis1: 1778.0000000000000000000000000000000000
rt_axis1: 1778.0000000000000000000000000000000000
ARC_ANGULAR_TRAVEL_EPSILON: 0.0000000049999999696126451453892514109611511230468750000000000000
Full Circle: False
initial angular_travel1: -0.0000005067316237727936822921037674
old angular_travel2: -0.0000005067316237727936822921037674
new angular_travel2: -0.0000005067316237727936822921037674
millimeters_of_travel:0.0010376918362453579902648925781250

the G2 is still a tiny little segment 0.001mm in circumference.

here is the G3 Version, this is almost but not quite a full circle, so it also fails the full circle test, because it's not a full circle, it's very close to on.  but it proves the ATan2 formula is still just fine.

[... 25 lines cut ...]
Here's the full circle version:
G0X1Y1
ok
G3X1Y1I-40J-70
Mpos Plane_Axis_0: 25.3999996185302734375000000000000000
Mpos Plane_Axis_1: 25.3999996185302734375000000000000000
Offset Plane_Axis_0: -1016.0000000000000000000000000000000000
Offset Plane_Axis_1: -1778.0000000000000000000000000000000000
Target Plane_Axis_0: 25.3999996185302734375000000000000000
Target Plane_Axis_1: 25.3999996185302734375000000000000000
center_axis0: -990.5999755859375000000000000000000000
[... cut ...]
rt_axis1: 1778.0000000000000000000000000000000000
ARC_ANGULAR_TRAVEL_EPSILON: 0.0000000049999999696126451453892514109611511230468750000000000000
Full Circle: True
Full Circle angular_travel: 6.2831854820251464843750000000000000
millimeters_of_travel:12866.7919921875000000000000000000000000

The only reason for the Start == End comparison is to prevent the ATan2 formula from breaking down... due to everything canceling out... as long as Start <> End even by one float, the Atan2 formula seems to be just fine.

Still testing more....

wolfmanjm, 2018-03-28:
ok sounds ok to me then. i'll go with whatever you come up with.

Zaaphod, 2018-03-28:
Everything has been testing well with angular_travel > 0 or < 0 modification.  The good news is that everything is perfect as is using the existing floats and floating point math, and the precision is also just fine.   

I have run into only one small issue.... and I fixed it!   but I have a feeling the fix may need some more work.   

Here's the issue:
It could happen that the host sends commands that cause no motion because they are less than the step resolution... as in my previous example of trying to engrave true type fonts... It would be nice if the host did not bother to send them at all, but most hosts just transmit what they are given.  That in itself is not a problem, they just waste a little time and fall though, whether they are arc or lines... BUT here's what is causing me the issue...  If they do not move, they do not update the machine_postion variable, and now when I process the next gcode, I am not using the previous position specified by Gcode as my start position anymore, it's the last position that caused motion... and as small as the error could be,  I have encountered errors because of it.

So my temporary solution was simple:  
change 
```
    if(moved) { 
        // set machine_position to the calculated target 
        memcpy(machine_position, target, n_motors*sizeof(float)); 
    } 

```
to not have the condition, and always do the update to machine_position whether it moved or not:

```
    // always set machine_position to the calculated target even if it was not large enough to cause a move so arc calculations will be accurate 
    memcpy(machine_position, target, n_motors*sizeof(float)); 
``` 

so far I have not had an issue from doing this... yes, I waste time setting the machine_position when I didn't get any motion, but now machine_position is always exact... problem solved.... well,  not really...  
well it DOES solve MY problem completely, BUT I can't help thinking there are some other conditions that need to prevent machine_position from updating.. not just segments too small to bother making a step.
First of all I notice the 'moved' variable has no other use other than to determine whether machine_position is updated or not. (because I get a compiler warning that the variable is never used) I got to thinking about this, and it seems like a lot of effort to bother reporting back if everything  moved or not just to save updating one variable,  so maybe there is some other reason for it... a reason that is not happening during the testing of things that run normally... other things besides just entities smaller than the machine resolution that can cause an axis not to move, like trying to move when you are on a limit switch..... or specifying an federate of 0...  so we don't want to update machine_position if we didn't move due to some other error, but we do want to update it if the entity was just too small to bother causing motion....  but if we hit a limitswitch after we started moving, then the variable got updated already, so what's the difference if we update it anyway, even if it didn't move?

So the question is,  can we somehow detect that there was nothing else wrong and the only issue for no move was the entity was too small and go ahead and update machine_position... because the new target could always have been the machine position... it just wasn't far enough away to cause a step.  I'm thinking if we detect an error we could set a move_error flag, and when we get done, if move_error is false we update machine_postion, otherwise we don't.... whether it moved or not.   

Then again, anything I can think of that would cause no motion, would also trigger a halt condition. so maybe we don't need to care so much about what happens to machine_position because the recovery from halt will reset it anyway.  We defiantly need to care about it being precise during normal operation, because then everything is working perfectly and we don't skip circles and we don't hack giant circles through parts,  so If I had to choose between everything being perfect when running and having an issue recovering from an exception... an issue that home would fix anyway, then I would prefer the issue to be in the recovery of the exception rather than in normal operation... the machine should be homed after any halt anyway.... but I'm bringing all this up because I'm not confident I have all the facts about why there is a 'moved' variable and what undesirable conditions would cause us to not want to update machine_position... so I'm asking and hoping we can work around whatever the issues may be.   luckily the 'moved' variable is only used in a handful of places, so it's not like it's in hundreds of places.

I have an alternative thought on a possible solution... I do not think it would be too difficult to pre-determine if a G0, G1, G2, or G3 would cause motion... just check whatever axis are specified to see if they are less than 1/steps_per_mm away from machine_position, and if so, just update machine_position and skip all other processing... it doesn't matter if it's a G0,G1,G2, or G3, it's not going to produce a step anyway... well then again... maybe... because the previous move could be 80% of the way to the next step, and the current move could be only 30% of the distance it takes to make a step, so it would cross the threshold and give us a step... so I guess there must be a variable containing the step remainder in it that we need to include in the comparison.. ah, but then [... cut ...]
If we really look at the other conditions that cause 'moved' to be false, is machine_position going to end up being reset anyway due to recovery from a halt condition? Do they really 'Need' to prevent machine_positon from being updated?

if we update machine_position without the machine taking a step, do we need to update whatever keeps track of the remainder of a step to go before the next step, or will this happen automatically?  (I suspect it is taken care of automatically because I never see a correction for anything at the step level in the code I have looked at)

Zaaphod, 2018-03-29:
ah, the step rounding isn't updated if the machine_position is updated without taking at least one step....   and that's also why a move that is 1/10th of a step could sometimes take a step... if the step remainder has accumulated to being close to a step, so yes, I can see why they need to accumulate. 

I suspect this will not be as much of an issue on microstep machines as it is on my half-step machines, and even with half steps it's a rare occurrence, even when I'm trying to make it happen, it's splitting hairs between making it happen when it could have been detected and operating below the resolution of float where it would be impossible to detect.   however, I'll try to think of a way around it.. it would be nice to solve the problem in a foolproof way.... but of course without breaking anything.

Zaaphod, 2018-03-29:
I can think of three ways to deal with this.... 
1. Make a new Variable called Gcode_Position which would always be updated and contain the current position as sent by the Gcode program, whether it caused motion or not.  Then this position can be used to properly calculate the arc defined by the Gcode as the start point would be accurate... otherwise we have the possibility to lose the actual start point and could be slightly off what was sent, as we are actually using some other previous point as start, not the actual previous point. This would have to have the same structure as Machine_Position.

2. Incorporate a command in smoothie that would 'disable full circles' or basically define that the gcode which is to follow is a mess of tiny line and arc segments that will never have a complete circle in them.  If this mode is active, we can apply a filter to get rid of micro arc segments that calculate to be a large arc and replace it with a straight line to the endpoint... one would need to 'enable full circles' to turn this filter off before cutting a real full circle... this could end up using more resources than the extra variable to make work correctly. but this would also filter out even the issues caused by arc segments with end points less than floating point precision away... two drawbacks are the resources needed, and the requirement that the user turn the feature on and off.... if they do not turn it on, it will not work, and it would be very difficult to turn it on or off automatically.

3. Leave it the way it is and see if it's even a real issue...  Even on my half-step configuration I have to try pretty hard to make it happen, and it involves moves less than 0.00005", on a microstep machine it would be much less.... so even though it could happen, chances are it won't.... in order to have the issue, you need to have a non-move entity immediately before a very specifically defined arc.  same goes for the possibility that someone could send an arc with an endpoint less than floating point precision away from the start point resulting in a full circle... it's possible, but it's so far out there that it's very unlikely.  Just tell everyone who wants to engrave true type fonts that they need to convert them to arc-splines, or interpolate it into lines first :)  
The way it is, it solves the issues for any programs that do not contain any micro segments in them... and correcting for microsegments could be considered another issue that may or may not justify the resources needed to detect and correct them.
Personally, I like option 1, but thought I would present all of them.

wolfmanjm, 2018-03-29:
neither 1 or 2 appeals to me, so we will go with option 3, leave it be.

Zaaphod, 2018-03-29:
Ok, fair enough, as I mentioned it's an extreme edge possibility anyway,  I will clean it up a bit and take out my reports and test it some more with some real data... perhaps I will even try to run a non-cleaned up true type font just to see what happens.

triffid, 2018-03-29:
What happens if you convert the start and end positions into machine
coordinates first, and compare those to find the edge case? They're
integers...

Zaaphod, 2018-03-29:
I didn't know there were integer machine coordinates, how can I do this conversion?  Do the machine coordinates represent steps from 0?   If I converted to machine coordinates as you suggest, I believe it would find the edge case of missing a tiny move that would not have made a step, because now my start and end are both adjusted to step resolution.  I don't think I need to use the converted numbers for anything else... just the detection, so I don't think I even need any other variables, just the function.   I'm guessing there is a function somewhere to convert a mm value into machine step coordinates.  Can you point me in the right direction?

wolfmanjm, 2018-03-29:
@Zaaphod there aren't any triffid is mistaken

Zaaphod, 2018-03-29:
It wouldn't have worked anyway... it would make the problem worse... as if I reduce everything to step resolution first, some small arcs will be mistaken for circles due to rounding error... the float comparison I am currently using is the most accurate way to catch these.

Zaaphod, 2018-03-30:
I've tested this with everything I can think of and have not been able to produce an error using normal means (generating a test programs with CAD/CAM and running them as opposed to forcing edge commands with extreme coordinates)  
My previous programs that had the original issue now work perfectly, as well as numerous other tests, including 2 tests engraving true type fonts that had tons of micro arc segments in them.  I've tested on a half step CNC router and a microstep CNC Mill.   Everything seems good so I've submitted PR #1293 for this.

wolfmanjm, 2018-03-30:
Your PR breaks my test case.... it is attached please fix ASAP.

[bad-curve.zip](https://github.com/Smoothieware/Smoothieware/files/1863175/bad-curve.zip)

If you run this you will see that within the first few seconds the head moves in a large circle on the very first piece.

wolfmanjm, 2018-03-30:
here is a shorter version that demonstrates the bad curve, it does a full circle instead of a tiny curve now, whereas I had fixed that with my previous PR, it is now broken again.
[bad-curve-2.zip](https://github.com/Smoothieware/Smoothieware/files/1863266/bad-curve-2.zip)

Zaaphod, 2018-03-30:
I've done testing with my report turned on.  here is the full results:
https://pastebin.com/SMx7ZtNv

This is the issue I was trying to avoid... I guess it's not as much of an edge case as I thought.  the conditions that must exist for this to happen are you must have a gcode command that produces no motion followed by a very very small arc segment where it's critical that the start point be precise... and that is what you have here.

Here is your issue:
G2 X406.1211 Y172.6116 I45.5591 J-15.5129   <---  This works fine

G1 X406.1208 Y172.6107 <--- this does not produce any motion, it's not one step away, machine_position is not updated.

G2 X406.121 Y172.6111 I49.9647 J-17.0159 <--- this produces a huge circle because it is using  X406.1211 Y172.6116 for the start point, not X406.1208 Y172.6107  and by definition a circle with that start point and the given endpoint is a huge circle.... 

So I can fix this (ignoring consequences for now....)   by changing to my firmware that always updates machine_position regardless of motion.  now....

G2 X406.1211 Y172.6116 I45.5591 J-15.5129   <---  This works fine

G1 X406.1208 Y172.6107 <--- this does not produce any motion, it's not one step away but machine position is updated

G2 X406.121 Y172.6111 I49.9647 J-17.0159 <--- this produces a small arc segment because it is correctly using  X406.1208 Y172.6107 as the start point, and the definition of this is a tiny arc segment.

Here is an abbreviated version of the report:

-------------------- results of using incorrect start point ------------------

G1 X406.1208 Y172.6107
ok
G2 X406.121 Y172.6111 I49.9647 J-17.0159
Full Circle angular_travel: -6.2831854820251464843750000000000000
millimeters_of_travel:331.6434326171875000000000000000000000

-------------------- results of using correct start point ------------------

G1 X406.1208 Y172.6107
ok
G2 X406.121 Y172.6111 I49.9647 J-17.0159
new angular_travel2: -0.0000084563916971092112362384796143
millimeters_of_travel:0.0004463510995265096426010131835938

Since we cannot update machine_position when there is no motion, the only 100% reliable way to fix this is to keep a second gcode_position that is always updated whether motion happens or not.   It would take too much code to try to pre-filter it out these cases any other way,  but it's relatively simple to always update gcode_position when machine_position is updated, but then also update gcode_position when the previous command produces no motion, leaving machine_position alone.   With gcode position being used for the start point we will have 100% reliable arc computations.

wolfmanjm, 2018-03-30:
I think this whole approach is wrong. All the gcode viewers I have are a lot simpler and get these edge cases correct. bCNC displays the arcs correctly.

I also now think that your comparison of start and end is wrong too. I think there may be alternative ways of doing this. The GCode spec says full circles should not specify X Y if they do it is an arc not a circle. So maybe the gcode is wrong?

Also why does this edge case work in the current edge?

wolfmanjm, 2018-03-30:
Here is the bCNC arc code...
```
	elif self.gcode in (2,3):	# CW=2,CCW=3 circle
			xyz.append((self.x,self.y,self.z))
			uc,vc = self.motionCenter()

			gcode = self.gcode
			if self.plane == XY:
				u0 = self.x
				v0 = self.y
				w0 = self.z
				u1 = self.xval
				v1 = self.yval
				w1 = self.zval
			elif self.plane == XZ:
				u0 = self.x
				v0 = self.z
				w0 = self.y
				u1 = self.xval
				v1 = self.zval
				w1 = self.yval
				gcode = 5-gcode	# flip 2-3 when XZ plane is used
			else:
				u0 = self.y
				v0 = self.z
				w0 = self.x
				u1 = self.yval
				v1 = self.zval
				w1 = self.xval
			phi0 = math.atan2(v0-vc, u0-uc)
			phi1 = math.atan2(v1-vc, u1-uc)
			try:
				sagitta = 1.0-CNC.accuracy/self.rval
			except ZeroDivisionError:
				sagitta = 0.0
			if sagitta>0.0:
				df = 2.0*math.acos(sagitta)
				df = min(df, math.pi/4.0)
			else:
				df = math.pi/4.0

			if gcode==2:
				if phi1>=phi0-1e-10: phi1 -= 2.0*math.pi
				ws  = (w1-w0)/(phi1-phi0)
				phi = phi0 - df
				while phi>phi1:
					u = uc + self.rval*math.cos(phi)
					v = vc + self.rval*math.sin(phi)
					w = w0 + (phi-phi0)*ws
					phi -= df
					if self.plane == XY:
						xyz.append((u,v,w))
					elif self.plane == XZ:
						xyz.append((u,w,v))
					else:
						xyz.append((w,u,v))
			else:
				if phi1<=phi0+1e-10: phi1 += 2.0*math.pi
				ws  = (w1-w0)/(phi1-phi0)
				phi = phi0 + df
				while phi<phi1:
					u = uc + self.rval*math.cos(phi)
					v = vc + self.rval*math.sin(phi)
					w = w0 + (phi-phi0)*ws
					phi += df
					if self.plane == XY:
						xyz.append((u,v,w))
					elif self.plane == XZ:
						xyz.append((u,w,v))
					else:
						xyz.append((w,u,v))

			xyz.append((self.xval,self.yval,self.zval))
```

Zaaphod, 2018-03-30:
The gcode viewers are using the actual gcodes and do not care if they take a step or not

wolfmanjm, 2018-03-30:
Then I think your equality test is wrong. It needs to be a float test with an epsilon where the epsilon is the substep (or machine resolution), so it is equal if the sart and end are withing the machine resolution which would generate a step.

wolfmanjm, 2018-03-30:
I won't accept keeping a separate last position around. That is  a hack to make the start and stop test work which I think maybe flawed anyway.

wolfmanjm, 2018-03-30:
You may want to look at how GRBL does it, as it does not seem to have these issues. (you may want to test grbl to see if your edge cases work on it).
Also maybe worth looking at how linuxcnc does it.

Zaaphod, 2018-03-30:
the epsilon is to fix floating point precision errors, not machine resolution errors.  Something must be done to account for the missing move.   If you allow for error in the comparison, then you end up sometimes making full circles out of arc segments that do not have a tiny non-move entity in front of them

wolfmanjm, 2018-03-30:
Another alternative is to round the floats to 4 decimal places (or 6 depending on the accuracy of the machine) and then compare those instead of the raw floats. Most machines are only good to 4 dp anyway.

Zaaphod, 2018-03-30:
if you round them then you still make circles out of what would have been arc segments, or arc segments out of what would have been circles.. there would be no way to distinguish between them anymore

wolfmanjm, 2018-03-30:
FYI http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3

Zaaphod, 2018-03-30:
the definition of the Gcode is that you use the previous position in specified in gcode for the start point... we are not doing that, we are using the last gcode that produced motion, and in this case, that is 1 gcode previous to the one we are supposed to be using.

Zaaphod, 2018-03-30:
Thanks for the link.   Start must be the previous Gcode position, not the previous machine position.  A Gcode program of just G2X1Y2I1J0 is invalid because who knows what you will get,  there was no move before it to specify the start position.     It's very clear that it's the previous Gcode move that defines the start positon, not where the machine happens to be at.    BCnc is always keeping the previous Gcode coordinates, they never throw one away.  We need to keep the start position accurate to the codes or we are doing it wrong.   I don't see what's wrong with a Gcode_positon variable, it's only 3 floats we need to keep

Zaaphod, 2018-03-30:
G2 X406.1211 Y172.6116 I45.5591 J-15.5129 
G1 X406.1208 Y172.6107 <---  We can't ignore these coordinates and reliably detect full circles or not it's impossible
G2 X406.121 Y172.6111 I49.9647 J-17.0159 

look how close this is....  
G2 X406.1211 Y172.6116 I45.5591 J-15.5129 
G2 X406.121 Y172.6111 I49.9647 J-17.0159 

is completely different than 
G2 X406.1211 Y172.6116 I45.5591 J-15.5129 
G1 X406.1208 Y172.6107
G2 X406.121 Y172.6111 I49.9647 J-17.0159

wolfmanjm, 2018-03-30:
bCNC is not checking if the start == target anyway so it is irrelevant what it stores. I think that check is just wrong.

wolfmanjm, 2018-03-30:
you keep insisting that you test for a full circle by testing start == target, however none of the other code I have looked at (including GRBL) do that. All your workarounds are to try to make that test work, which also indicates it is the wrong path to take.

Zaaphod, 2018-03-30:
Perhaps we can skip the == test if we modify the math being done by the atan2 formula... but I guarantee the results of above will calculate the same if the start point is not accurate   my == test is to make the atan2 formula work reliably...    so if we fix the atan2 formula (possible to do, I need to analyze some alternatives)  then we don't need ==   but in any case, to be reliable we sill can't, do any rounding, which introduces errors, and we still can't throw away any Gcodes.

I will work on an alternative atan2 formula that will work without the == test... it will take me some time, but I still think skipping over a crucial set of coordinates is going to always be a problem... if all the other math is perfect... then 

G2 X406.1211 Y172.6116 I45.5591 J-15.5129
G2 X406.121 Y172.6111 I49.9647 J-17.0159
will always produce a nearly full circle.. because 
G1 X406.1211 Y172.6116
G2 X406.121 Y172.6111 I49.9647 J-17.0159
is a nearly full circle with 
angular_travel of  -6.283175945281982421875
and millimeters_of_travel of 8423.72949218750 

Leaving out the 
G1 X406.1208 Y172.6107
completely changes it.... go into bcnc or anything else and remove this G1 line and see what happens.

It just so happens with your example here, that the G1 line moves away far enough from the previous coordiantes that now the G2 is a small arc.

Zaaphod, 2018-03-30:
Here is a simulation,  there is no smoothie processing here,  just strictly off the Gcodes the original program segment produces:
[image]

if I remove the G1 line by commenting it out, we get: 
[image]

Anything that allows this to work without the G1 coordinate included, will inevitably cause what are supposed to be full circles to be tiny arc segments instead

The G1 move here is critical, and cannot be thrown away... even though it causes no motion, it is essential to things processing correctly.  

if you set your output on your CAD/CAM package to round to 3 decimal places, you would have gotten this:

G2 X406.121 Y172.612 I45.561 J-15.513
G1 X406.121 Y172.611
G2 X406.121 Y172.611 I49.965 J-17.016
then it's impossible to never get the full circle with any means... because the gcodes actually define a full circle as the start and end points are identical.  we need to keep track of actual Gcodes issued with the precision given.  that extra precision is necessary and must be taken into consideration.

Zaaphod, 2018-03-30:
Here is the simulation with the entire program....
Full program as is.... 
[image]

program with G1 X406.1208 Y172.6107 commented out... (effectively what we do when we do not update the maching_position because it is too small to take a step)

[image]

wolfmanjm, 2018-03-30:
Ok I think I understand. So if the original code were using a machine_position which had not been skipped it would have worked with all test cases right? including yours, so no testing of start == target required.

The trouble is it is not really feasible to maintain a machine_position that did not move, as it is set in so many places, you would need to find every place it is modified and modify the mirror copy as well, which will be very ugly and simply not practical IMO.

wolfmanjm, 2018-03-30:
Ok try this with all your test cases....

Comment out this line from the unchanged current edge... (do not use your new version)

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/Robot.cpp#L1044

(Or change the moved to true).

See if your test cases all work then.

I am not entirely sure what side effect this will have on lots of tiny non moving segments, but I'll worry about that later.

Zaaphod, 2018-03-30:
yes. it's true that if the machine_position had not been skipped then your test case and several (not all) of mine would have worked...   there is a separate issue with atan2 not working correctly if machine_position and Target are ==   I think I can get around it without doing the == comparison... we will need to avoid atan2, and do the math another way.   I believe the issue with atan2 it can only produce a result from pi to -pi (excluding -pi)  but sometimes we need -pi and atan2 can't produce it.  you can only expect to get -pi from atan2 if all inputs are == 
I believe that the math using atan2 works mostly for G3 but fails occasionally on G2,  and it's because it cannot produce -pi...  so I'm thinking of reversing the input to it on G2 moves and maybe that alone will be enough to solve the issue... I need to work out exactly how to do that though... but it is also possibly better to just not use atan2 and use other trig functions that do not have the limitation...  We will probably need one calculation for G3 and another one for G2, but then we can avoid the == comparison.

I am not certain there is an issue with updating non-moving segments, but I am not familiar with how the step rounding happens.  if it uses absolute reference to 0,0 then there would never be a problem as every position is evaluate with absolute math to 0,0  if it uses some kind of relative to previous positon method, then there could be a cumulative effect of a 1 step drift when steps are not taken but machine position is updated... I can specifically test for this... you need to do a lot of moves that move with moves that don't move in the same direction... like 1000 of them,  then do one large move back to 0 and see if your dial indicator comes back to 0

The line you have commented out is the one I commented out to test my version, and with that commented out then all cases pass, even yours passed....    but I will do the tests you recommend with original edge.   I will also devise a test to see if there really is any negative consequences to always updating machine_position.

Just to clarify, for our purposes, if there were a GCode_position, it would not need to be updated everywhere machine _position is.... only in robot.cpp where there is a G0,G1,G2, or G3... it's purpose it strictly to track Gcode positions for those 4 commands... it doesn't need to nor should it follow machine_position.   If something does a G92 or homes the machine or whatever else might affect Machine_position, it isn't valid to just run G2 or 3 immediately after, G2 & G3 require a Gcode positioning move before they are executed, nothing else is valid.... even positioning with G20 then doing a G21 followed by a G2 or G3 is invalid.   G2 and G3 are only valid with a G0, G1, G2, or G3 immediately preceding them.

I am out smoothifying a CNC router.  I'll test when I get home.... back to soldering wires

wolfmanjm, 2018-03-30:
Oh there is an issue, I know there is, this is NOT a work around, I am just curious if it is the only issue.

The trouble is if I update the machine position then future speed and distance calculations will be wrong causing acceleration issues and speed issues. The steps moved will not correlate to the calculated distance so speed and acceleration will be wrong, potentially causing skipped steps.

I am not sure there is a solution to this, and as very few people use G2/G3 I am afraid we may need to let this go, and simply document the issue.

I will continue to consider if there is a way around this, but it is quite complex and involves the entire motion calculation paths.

wolfmanjm, 2018-03-30:
I don't see where in the arc calcs you would use this mirrored position. As I said the size of the segments and moves needs to match the steps that are issued. I can see things getting out of sync if you have a mirrored position that does not get set when machine position gets set.

[7 more replies]

closed 2018-03-31 by wolfmanjm