Smoothieware/Smoothieware #1292: Feature Request: Probe while traversing a circle.

Zaaphod, 2018-03-29:
The reason one would want to do this is because it allows a method to auto-probe a 2D part and easily find changes in geometry like sharp corners... the concept is:  You probe one point, then set a centerpoint some fixed distance away, then try to do a full circle while probing around that center point.  at some point, you must probe the object again, because you will be ending up where you started and there was a probe hit there... this method can detect accurately even the sharpest inside or outside corners.   It would then be helpful if the probe report would contain the existing coordinates of the linear probe, but add an additional absolute XY angle that the probe was detected, this would be the angle from the CenterPoint to the probe contact point.

wolfmanjm, 2018-03-29:
this is usually done in the host, for instance bCNC had a probe circle command. they use the standard G38.2 commands.
I also have a script that finds the center of an object and also finds the dimensions of an object again using g38.2 commands.

as this would be a non standard g code we won't be adding it.

closed 2018-03-29 by wolfmanjm

Zaaphod, 2018-03-29:
I have center finding, edge finding, and dimensioning scripts too, but how do they do a probe circle without smoothie probing having a probe circle command?   Perhaps I should install bCNC and see what smoothie codes it generates when I try their probe circle.

wolfmanjm, 2018-03-29:
it is easy you just move in a circle and probe each time at each segment, you at least need to know where the object is relative to the circle. if you mean you want probe active while moving in a circle then that is not something we would add. 
You could always add that to your own branch though, but it is unrealistic to have it in the main branch. 3d printers can't even use G2/G3 at the moment anyway.

wolfmanjm, 2018-03-29:
Actually now that you cando a G32.2 in any direction, just have you host segment the circle and issue a G32.2 probe with each segment and notice if the probe was triggered or not.

Zaaphod, 2018-03-29:
Yes, I see... I just need to calculate maximum arc error for the radius of the circle I want to use and come up with a segment length, divide that by the circumference and figure out how many divisions I will have, and then I know my angle change for each segment.. then just stack them change angle, probe the segment length, change angle probe segment length until I get a probe read.    This gives me some motivation to parse the probe report so I will get the probe detected bit at the end... I've been just doing it by comparing coordinates but just looking at the one bit would be faster.   I'll probably need G38.3 instead of G38.2 though so I don't get the error for every segment that doesn't hit.

Thanks for the suggestion.. and thanks for adding the probe in any direction, that is going to be necessary for this to work.

wolfmanjm, 2018-03-29:
Actually it can be much simpler.. here is a one line perl script (based on one triffid gave me)...

`perl -MMath::Trig -Mstrict -e 'my $radius = 80; my $segments = 200; for my $a (0..$segments) { printf "G38.3 X%f Y%f\n", sin($a * (360 / $segments) * pi / 180) * $radius, cos($a * (360 / $segments) * pi / 180) * $radius; }'`

Zaaphod, 2018-03-29:
Thank you very much for that script!  saves a quite a bit of work!  I'm curious,  there doesn't seem to be any way to stop this once the probe is hit.... does it just always stream all 201 of the G38.3s  even after the probe hits, they all just really do nothing because the probe is active before any motion can happen?

Zaaphod, 2018-03-29:
well the first probe wouldn't move because it would send G83.3X0Y0 since $a would be 0.... I think it should be 

`for my $a (1..$Segments)`

wolfmanjm, 2018-03-29:
No you would not actually use that exact script, you would issue each G38.3 then check to see if it hit or not then issue the next one, so each segment needs to check if it hit or not. Then when it does hit based on the returned coordinates you can see exactly where on the arc it hit. you can do quite big segments depending on the accuracy you want.